espressif/esp-idf #12015: Cannot compile ESP BLE MESH with CPP/C++

heculess, 2023-08-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

compile complete

### What is the actual behavior?

compile failed

### Steps to reproduce.

cd ble_mesh_sensor_model\sensor_server
change main.c to main.cpp
change main.cpp to compile with C++
get error 

### Build or installation Logs.

```plain
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:268:16: error: expected identifier before numeric constant
  268 |     .keys = { [0 ... (CONFIG_BLE_MESH_MODEL_KEY_COUNT - 1)] =       \
      |                ^
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/models/include/esp_ble_mesh_config_model_api.h:27:9: note: in expansion of macro 'ESP_BLE_MESH_SIG_MODEL'
   27 |         ESP_BLE_MESH_SIG_MODEL(ESP_BLE_MESH_MODEL_ID_CONFIG_SRV,          \
      |         ^~~~~~~~~~~~~~~~~~~~~~
E:/ble_mesh_sensor_model/sensor_client/main/ble_mesh_gateway.cpp:71:5: note: in expansion of macro 'ESP_BLE_MESH_MODEL_CFG_SRV'
   71 |     ESP_BLE_MESH_MODEL_CFG_SRV(&config_server),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
E:/ble_mesh_sensor_model/sensor_client/main/ble_mesh_gateway.cpp: In lambda function:
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:268:61: error: expected '{' before '=' token
  268 |     .keys = { [0 ... (CONFIG_BLE_MESH_MODEL_KEY_COUNT - 1)] =       \
      |                                                             ^
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/models/include/esp_ble_mesh_config_model_api.h:27:9: note: in expansion of macro 'ESP_BLE_MESH_SIG_MODEL'
   27 |         ESP_BLE_MESH_SIG_MODEL(ESP_BLE_MESH_MODEL_ID_CONFIG_SRV,          \
      |         ^~~~~~~~~~~~~~~~~~~~~~
E:/ble_mesh_sensor_model/sensor_client/main/ble_mesh_gateway.cpp:71:5: note: in expansion of macro 'ESP_BLE_MESH_MODEL_CFG_SRV'
   71 |     ESP_BLE_MESH_MODEL_CFG_SRV(&config_server),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
E:/ble_mesh_sensor_model/sensor_client/main/ble_mesh_gateway.cpp: At global scope:
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:65:51: error: no match for 'operator=' (operand types are '<lambda()>' and 'int')
   65 | #define ESP_BLE_MESH_KEY_UNUSED                   0xFFFF
      |                                                   ^~~~~~
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:269:13: note: in expansion of macro 'ESP_BLE_MESH_KEY_UNUSED'
  269 |             ESP_BLE_MESH_KEY_UNUSED },                              \
      |             ^~~~~~~~~~~~~~~~~~~~~~~
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/models/include/esp_ble_mesh_config_model_api.h:27:9: note: in expansion of macro 'ESP_BLE_MESH_SIG_MODEL'
   27 |         ESP_BLE_MESH_SIG_MODEL(ESP_BLE_MESH_MODEL_ID_CONFIG_SRV,          \
      |         ^~~~~~~~~~~~~~~~~~~~~~
E:/ble_mesh_sensor_model/sensor_client/main/ble_mesh_gateway.cpp:71:5: note: in expansion of macro 'ESP_BLE_MESH_MODEL_CFG_SRV'
[... 246 lines cut ...]
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/models/include/esp_ble_mesh_sensor_model_api.h:33:9: note: in expansion of macro 'ESP_BLE_MESH_SIG_MODEL'
   33 |         ESP_BLE_MESH_SIG_MODEL(ESP_BLE_MESH_MODEL_ID_SENSOR_CLI,    \
      |         ^~~~~~~~~~~~~~~~~~~~~~
E:/ble_mesh_sensor_model/sensor_client/main/ble_mesh_gateway.cpp:73:5: note: in expansion of macro 'ESP_BLE_MESH_MODEL_SENSOR_CLI'
   73 |     ESP_BLE_MESH_MODEL_SENSOR_CLI(NULL, &sensor_client),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:271:63: note: candidate: 'constexpr<lambda()>&<lambda()>::operator=(const<lambda()>&)'
  271 |     .groups = { [0 ... (CONFIG_BLE_MESH_MODEL_GROUP_COUNT - 1)] =   \
      |                                                               ^
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/models/include/esp_ble_mesh_sensor_model_api.h:33:9: note: in expansion of macro 'ESP_BLE_MESH_SIG_MODEL'
   33 |         ESP_BLE_MESH_SIG_MODEL(ESP_BLE_MESH_MODEL_ID_SENSOR_CLI,    \
      |         ^~~~~~~~~~~~~~~~~~~~~~
E:/ble_mesh_sensor_model/sensor_client/main/ble_mesh_gateway.cpp:73:5: note: in expansion of macro 'ESP_BLE_MESH_MODEL_SENSOR_CLI'
   73 |     ESP_BLE_MESH_MODEL_SENSOR_CLI(NULL, &sensor_client),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:271:63: note:   no known conversion for argument 1 from 'int' to 'const<lambda()>&'
  271 |     .groups = { [0 ... (CONFIG_BLE_MESH_MODEL_GROUP_COUNT - 1)] =   \
      |                                                               ^
E:/esp-idf/frameworks/esp-idf-v5.1/components/bt/esp_ble_mesh/api/models/include/esp_ble_mesh_sensor_model_api.h:33:9: note: in expansion of macro 'ESP_BLE_MESH_SIG_MODEL'
   33 |         ESP_BLE_MESH_SIG_MODEL(ESP_BLE_MESH_MODEL_ID_SENSOR_CLI,    \
      |         ^~~~~~~~~~~~~~~~~~~~~~
E:/ble_mesh_sensor_model/sensor_client/main/ble_mesh_gateway.cpp:73:5: note: in expansion of macro 'ESP_BLE_MESH_MODEL_SENSOR_CLI'
   73 |     ESP_BLE_MESH_MODEL_SENSOR_CLI(NULL, &sensor_client),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

nopnop2002, 2023-09-14:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/cplusplus.html

chegewara, 2024-03-27:
The link is not helpful in this case at all.
I have the same issues with 5.1 and 5.2, but its has been solved/fixed in v5.3 (master), hopefully for good.

closed 2025-11-19 by Alvin1Zhang