espressif/esp-idf #12016: Obstruct to esp_wifi_connect() function slow down the communication of ESP working as AP in WIFI_MODE_APSTA mode.

serdartoren, 2023-08-04:
### Is your feature request related to a problem?

While ESP is running in WIFI_MODE_APSTA mode, functions such as esp_wifi_connect(), esp_wifi_scan_start() slow down ESP's wifi traffic running in AP mode.
While this is not a problem for most applications, it is a problem in our application.

Because different devices than STA or AP can be connected to ESP32.
These devices can message 50 times with esp32 per second.
If the STA mode is connected to a hotspot there is no problem and everything works fine.

But if the hotspot connection is lost, the code will try to connect to the hotspot again every 5s.
And this negatively affects the communication at the AP point.

### Describe the solution you'd like.

What I want is that the esp_wifi_connect() function does not affect AP communication.
If this is not possible, the existence of a function like check_AP_point(char* SSID) would also be good.

This function checks the required SSID without wasting much time and can inform me if that point is available.
And then I can use esp_wifi_connect() when is necessary.

In other words, the hotspot can be closed for hours or open at any time.
I think that the disappearance of this slowdown effect will improve the quality of ESP-IDF.

serdartoren, 2023-08-04:
https://github.com/espressif/esp-idf/issues/3583#:~:text=The%20AP%20will%20be%20temporarily%20impacted%20when%20the%20ESP32%20STA%20is%20connecting%20or%20scanning.

This is not just my experience.
It is normal behavior of ESP32.
But I think it would be in everyone's interest to improve it.

Xiehanxin, 2023-08-07:
hi @serdartoren , it seems that the STA keeps trying to reconnect after losing connection to the hotspot, which results in abnormal communication with the AP, I think you can try to increase the reconnection intetval to see if there are and difference.

serdartoren, 2023-08-08:
Hi @Xiehanxin, Thanks for the quick reply. I tried your suggestion now I will send you 2 videos.
In the first video, it starts trying 5 seconds after the connection is lost. And it repeats periodically.

https://github.com/espressif/esp-idf/assets/42543700/5aa10b40-890d-4db9-ba28-1c34b1b0778c

serdartoren, 2023-08-08:
15 seconds version:

https://github.com/espressif/esp-idf/assets/42543700/89414065-e428-4970-b36b-eeedb6226ea2

serdartoren, 2023-08-08:
These videos were taken from the computer screen.
The PC is connected to the AP point of the ESP32.
ESP32 is connected to my phone's hotspot.

This occurs when I turn off the phone's hotspot.

serdartoren, 2023-08-08:
Here I tried to explain by simplifying the codes I wrote for wifi.

My ESP event handler code:

                else if (event_id == WIFI_EVENT_STA_DISCONNECTED) 
                {
                        wifi_state.client.info.has_ap = false;
                        ip_info_sta.ip.addr = 0;
                        
                                xEventGroupSetBits(s_wifi_event_group, WIFI_FAIL_BIT);
                                //Failed but don't give up a new task runs until you connect again
                                if(re_connect_Handle == NULL)
                                        xTaskCreatePinnedToCore(re_connect, "re_connect", 3072, NULL, 1, &re_connect_Handle, 1);
                }

The task I run when the STA connection is lost:

        for(;;)
        {
                EventBits_t bits = xEventGroupWaitBits(s_wifi_event_group,
                        WIFI_CONNECTED_BIT | WIFI_FAIL_BIT,
                        pdFALSE,
                        pdFALSE,
                        portMAX_DELAY);

                /* xEventGroupWaitBits() returns the bits before the call returned, hence we can test which event actually
                * happened. */
                if (bits & WIFI_FAIL_BIT) { 
                        
                        if(sock_UDP != 0)
                                close(sock_UDP);
                        sock_UDP = 0;
                        pr_debug(LOG_PFX "Try to Reconnect AP \r\n");
                        vTaskDelay(msToTicks(15000));//We don't want to keep wifi busy when there is no connection so we have a 5s delay
                        pr_debug(LOG_PFX "Try to Reconnect AP \r\n");
                        esp_wifi_connect();
                        xEventGroupClearBits(s_wifi_event_group, WIFI_CONNECTED_BIT | WIFI_FAIL_BIT);
                }
                else if (bits & WIFI_CONNECTED_BIT) {
                        pr_debug(LOG_PFX "Reconnected to AP \r\n");
                        int addr_family = 0;
                        int ip_protocol = 0;
                        dest_addr.sin_addr.s_addr = htonl(INADDR_BROADCAST);
                        dest_addr.sin_family = AF_INET;
                        dest_addr.sin_port = htons(7223);
                        addr_family = AF_INET;
                        ip_protocol = IPPROTO_IP;

                        sock_UDP = socket(addr_family, SOCK_DGRAM, ip_protocol);

                        if (sock_UDP < 0) {
                                pr_error(LOG_PFX "Unable to create UDP socket");
                        }
                        else
                        {
                                // Set timeout
                                struct timeval timeout;
                                timeout.tv_sec = 10;
                                timeout.tv_usec = 0;
                                setsockopt (sock_UDP, SOL_SOCKET, SO_RCVTIMEO, &timeout, sizeof(timeout));
                                pr_info(LOG_PFX "UDP Socket recreated \r\n");
                        }
                        xEventGroupClearBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
                        break;
                }
                else {
                        pr_error(LOG_PFX "ELSE EVENT \n");
                }
        }
        re_connect_Handle  = NULL;
[... 78 lines cut ...]
                esp_wifi_set_ps(WIFI_PS_NONE);
                EventBits_t bits = xEventGroupWaitBits(s_wifi_event_group,
                        WIFI_CONNECTED_BIT | WIFI_FAIL_BIT,
                        pdFALSE,
                        pdFALSE,
                        portMAX_DELAY);

                /* xEventGroupWaitBits() returns the bits before the call returned, hence we can test which event actually
                * happened. */
                if (bits & WIFI_CONNECTED_BIT) {
                        pr_info(LOG_PFX "Connected to AP");
                        pr_info("\r\n");
                        pr_info_str_msg("SSID: ", cfg1->ssid);
                        pr_info("\r\n");
                        set_client_ap_cb(true);
                        xEventGroupClearBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
                } else if (bits & WIFI_FAIL_BIT) {
                        pr_info(LOG_PFX "Failed to connect AP \r\n");
                        pr_info_str_msg("SSID: ", cfg1->ssid);
                        pr_info("\r\n");
                        set_client_ap_cb(false);
                } else {
                        pr_error(LOG_PFX "UNEXPECTED EVENT \n");
                }
        }

serdartoren, 2023-08-08:
@Xiehanxin, Is the behavior you see in the video abnormal for ESP-IDF?

Xiehanxin, 2023-08-08:
hi @serdartoren  thanks for your video, we will take a look and give you feedback

Xiehanxin, 2023-08-10:
hi @serdartoren ,from your video, it seems that it keeps scaning after station disconnecting which takes the AP's resource.
I think you can do such things:
1.change the log level to verbose, it will help us know what STA is doning at that time.
2.try to do specify channel scanning at first, which can reduce the scanning time。
3.you can try to modify the logic to scan on the previous  channel after disconnecting, don't scan all the time, you can perform a scan every few seconds, and do connecting when scan successully.

serdartoren, 2023-10-15:
Hi, channel scanning worked and everything works fine now.
I search only one channel with esp_wifi_scan_get_ap_records() every 20 seconds, this works for me.
I'm closing the issue, thanks for your help ^^

closed 2023-10-15 by serdartoren