espressif/esp-idf #12018: [mesh] v4.4 Question: avoid cyclic connections in fixed root environment

KonssnoK, 2023-08-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello @zhangyanjiaoesp ,
we have a small question on the mesh.

We currently use the fixed root functionality when we need to force the LTE device to be the root.
We took the initial code from the `manual_networking` example.

The problem is that it can happen that if you have A B C devices,
A connects to B and B connects to A, because C is maybe with lower RSSI (imagine creating a mesh and then moving away/switching off the root).

Sadly, the information that we have available is not well documented (or we didn't find the documentation),
so we don't know what we can use to identify such situations and skip conecting in a cyclic way.

Do you have any idea/way of preventing that?

Where can we find a better explanation of the following struct?
Maybe the answer is here.

```
/**
 * @brief Mesh networking IE
 */
typedef struct {
    /**< mesh networking IE head */
    uint8_t eid;             /**< element ID */
    uint8_t len;             /**< element length */
    uint8_t oui[3];          /**< organization identifier */
    /**< mesh networking IE content */
    uint8_t type;            /** ESP defined IE type */
    uint8_t encrypted : 1;   /**< whether mesh networking IE is encrypted */
    uint8_t version : 7;     /**< mesh networking IE version */
    /**< content */
    uint8_t mesh_type;       /**< mesh device type */
    uint8_t mesh_id[6];      /**< mesh ID */
    uint8_t layer_cap;       /**< max layer */
    uint8_t layer;           /**< current layer */
    uint8_t assoc_cap;       /**< max connections of mesh AP */
    uint8_t assoc;           /**< current connections */
    uint8_t leaf_cap;        /**< leaf capacity */
    uint8_t leaf_assoc;      /**< the number of current connected leaf */
    uint16_t root_cap;       /**< root capacity */
    uint16_t self_cap;       /**< self capacity */
    uint16_t layer2_cap;     /**< layer2 capacity */
    uint16_t scan_ap_num;    /**< the number of scanning APs */
    int8_t rssi;             /**< RSSI of the parent */
    int8_t router_rssi;      /**< RSSI of the router */
    uint8_t flag;            /**< flag of networking */
    uint8_t rc_addr[6];      /**< root address */
    int8_t rc_rssi;          /**< root RSSI */
    uint8_t vote_addr[6];    /**< voter address */
    int8_t vote_rssi;        /**< vote RSSI of the router */
    uint8_t vote_ttl;        /**< vote ttl */
    uint16_t votes;          /**< votes */
    uint16_t my_votes;       /**< my votes */
    uint8_t reason;          /**< reason */
    uint8_t child[6];        /**< child address */
    uint8_t toDS;            /**< toDS state */
} __attribute__((packed)) mesh_assoc_t;
```

For example, in the manual_networking example `layer2_cap` it's used to check if we want to change the parent node, but we have no idea what it represent.

Also,
we saw that the RSSI reported by the record_assoc (AP) is different from the one reported by the mesh_assoc RSSI, described as `RSSI of the parent` ... we have not clear why.

Thanks!

zhangyanjiaoesp, 2023-08-07:
@KonssnoK 

> A connects to B and B connects to A

This won't happen，If B is the parent of A, and A wants to connect to B, then B will show the log `[CYCLIC] cyclic found, deauth aid : xx, child_mac `,  and the deauth is `MESH_REASON_CYCLIC(100)`.

The `layer2_cap ` in  `mesh_assoc_t` structure  represents the current capacity of the Layer 2 node, that is, the total child numbers plus 1. And it will be updated only when the device is layer2 node.

> we saw that the RSSI reported by the record_assoc (AP) is different from the one reported by the mesh_assoc RSSI, described as RSSI of the parent ... we have not clear why.

Can you explain what is `RSSI reported by the record_assoc (AP)` and what is `mesh_assoc RSSI` ? Which two APIs are you trying to express?

KonssnoK, 2023-08-07:
@zhangyanjiaoesp ,
why you say it won't happen?
with my current periodic scan selection i do see sometimes
```
I (124638) wifi:station: 7c:df:a1:e0:8b:7c leave, AID = 1, bss_flags is 658531, bss:0x3dea1834
I (124639) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (02:16:15.133) mesh_main: <MESH_EVENT_CHILD_DISCONNECTED>aid:1, 7c:df:a1:e0:8b:7c
W (02:16:15.140) mesh_main: <MESH_EVENT_ROUTING_TABLE_REMOVE>remove 2, new:1
I (124710) wifi:state: auth -> init (200)
I (124711) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (02:16:15.210) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 2 WIFI_REASON_AUTH_EXPIRE
I (124790) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (124792) wifi:state: init -> auth (b0)
I (124795) wifi:state: auth -> init (65c0)
I (124796) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (02:16:15.294) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
I (124892) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (124893) wifi:state: init -> auth (b0)
```

we follow the example of "manual_networking",
so we have the following structs 

```
        wifi_ap_record_t parent_record; // Data coming from the AP
        mesh_assoc_t parent_assoc;      // Data coming from the MESH layer
```
The RSSI reported by these 2 different structs are different,
specifically

```
best_result.parent_record.rssi
best_result.parent_assoc.rssi
```

but we have no idea why

KonssnoK, 2023-08-07:
i somehow get the feeling that without propagating its own parent we cannot avoid the cyclic 🤔 
I'll investigate a bit more

KonssnoK, 2023-08-07:
@zhangyanjiaoesp is there any way to change the values sent in the structure
`mesh_assoc_t` ? 

For example i see the value child is returned as zeros, maybe i can set it to the parent of the current node?

zhangyanjiaoesp, 2023-08-08:
> @zhangyanjiaoesp , why you say it won't happen? with my current periodic scan selection i do see sometimes
> 
> ```
> I (116731) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
> W (02:16:07.237) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 100 MESH_REASON_CYCLIC
> W (02:16:10.087) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 205 WIFI_REASON_CONNECTION_FAIL
> W (02:16:13.206) monitor: 3W3P:: rule activated
> ```
> 
I mean it will connect fail due to `MESH_REASON_CYCLIC` reason.

> we follow the example of "manual_networking", so we have the following structs
>
> ```
>         wifi_ap_record_t parent_record; // Data coming from the AP
>         mesh_assoc_t parent_assoc;      // Data [...]
The `parent_record.rssi` is rssi of the scanned AP, and the `parent_assoc.rssi` is the scanned AP's parent.  It has been spelled out in the notes.
[image]

> but we have no idea why
> 
> > The layer2_cap in mesh_assoc_t structure represents the current capacity of the Layer 2 node, that is, the total child numbers plus 1. And it will be updated only when the device is layer2 node.
> 
> But what if the device is a layer 3 node? what does it represent then? does it have a value different than 0?
> 
If the device is a layer3 node, `layer2_cap` won't be updated when ie updating, the `self_cap` will be updated.

zhangyanjiaoesp, 2023-08-08:
> @zhangyanjiaoesp is there any way to change the values sent in the structure `mesh_assoc_t` ?
> 
> For example i see the value child is returned as zeros, maybe i can set it to the parent of the current node?

@KonssnoK There is no API for customer to set the `assoc_ie` in mesh frame.  Of cause you can follow your own rules to specify the parent node for your device.

KonssnoK, 2023-08-08:
@zhangyanjiaoesp sorry but the comments in the structure are not really that explanatory.. :)

For example i saw that:
- rc_rssi is -120 for layers > 1 but populated for layer 1. The rc_addr instead is always populated.
- rssi is not populated (-120) for layer 1 but it's populated for higher layers.
- parent RSSI is not self-explanatory, because you are currently looking for a parent, so it can be confused for the parent you are looking for
- flag is populated with 68, no idea what that maps to. What is the source of such flag?
- everything after rc_rssi is set to 0, i guess that is normal in fixed root, but i'm not sure
- what is reason?
- what is child? ( a device can have multiple children, which one would be picked?)

This just to name a few improvements that could be made to the documentation of the struct.

I'll continue to try to create a way of uniquely select a parent in a fixed root environment with the data provided by the ie.
Thanks.

KonssnoK, 2023-08-08:
Adding an example:

```
W (10:30:40.554) mesh_hand: <MESH>7CDFA1FBFF05, layer:1/5, assoc:3/6, 0, ch:1 [0] rssi:-33, parent_rssi:-120 root_rssi:-28 Type 1
W (10:30:40.562) mesh_hand: Better parent found 1<6 0<0 -33>-120
W (10:30:40.568) mesh_hand: <MESH>7CDFA1E07441, layer:2/4, assoc:0/6, 1, ch:1 [1] rssi:-28, parent_rssi:-71 root_rssi:-120 Type 2
```
in this case the device is seeing the root device 7CDFA1FBFF05 which reports a record.rssi of -33, assoc.rssi-120, and a rc_rssi of -28.
in this case i have no idea what rc_rssi is, considering the device is connected to LTE 🤔 (and it's apparently different from record.rssi)

zhangyanjiaoesp, 2023-08-09:
@KonssnoK First we should be clear about the meaning of `parent_record` and `parent_assoc`:
[image]
In the `mesh/manual_networking` example, it will call `esp_mesh_scan_get_ap_record(&record, &assoc)`, so the `parent_record` is the record of the scanned AP, and the `parent_assoc` is the assoc ie of the scanned AP.

> @zhangyanjiaoesp sorry but the comments in the structure are not really that explanatory.. :)
> 
> For example i saw that:
> 
> * rc_rssi is -120 for layers > 1 but populated for layer 1. The rc_addr instead is always populated.

The `rc_rssi` is the root candidate rssi, it's used in self-organized network, so for root-fixed network, only root node will populate this variable, other nodes will use the default value -120.

> * rssi is not populated (-120) for layer 1 but it's populated for higher layers.
> * parent RSSI is not self-explanatory, because you are currently looking for a parent, so it can be confused for the parent you are looking for. how is it populated in case the device which sent the beacon is idle?

The `rssi` is the rssi of the parent, (note: the `rssi` is a variable of AP's assoc ie, so it represents the rssi of the AP's parent node). root node will not update this value, it use the default value -120. Here is my test using `mesh/manual_networking` example:

[image]
[image]
[image]

A (3e:84) connects to the router as root, B (ec:04) connects to A (34:84) as a layer2 node, and C (83:44) connects to B (ec:04) as a layer3 node. 

A (3e:84) is root, it's parent is the router, and router don't have mesh ie, so it doesn't contain related log.
B (ec:04) is layer2 node, `parent_record->rssi` indicates the rssi(-10) value of the scanned AP ESPM_403E84 (A). `parent_assoc->rssi` represents the rssi(-120) value of the parent of A (3e:84).
C (83:44) is layer3 node, `parent_record->rssi` indicates the rssi(-11) value of the scanned AP ESPM_A6EC04 (B). `parent_assoc->rssi` indicates the rssi(-11) value of the parent of B (ec:04) , that is, the rssi value when B (ec:04) is connected to A (3e:84).

> * flag is populated with 68, no idea what that maps to. What is the source of such flag?

The `flag = 68 = 0x44` means not voted and root-fixed. 

> * everything after rc_rssi is set to 0, i guess that is normal in fixed root, but i'm not sure

Yes, you are right.  The variables after `rc_rssi` are vote related, they are used in self-organized network.

> * what is reason?

The `reason` is vote reason, 1 indicates root initiated, 2 indicates child initiated.

> * what is child? ( a device can have multiple children, which one would be picked?)

I will check the mesh code and reply you.

> This just to name a few improvements that could be made to the documentation of the struct.

We will modify the mesh relevant documents for the convenience of customers. Thanks for your suggestion.

zhangyanjiaoesp, 2023-08-09:
@KonssnoK 

> what is child? ( a device can have multiple children, which one would be picked?)

The `child` variable is not used in mesh code, i will always be 0

KonssnoK, 2023-08-09:
> @KonssnoK
> 
> > what is child? ( a device can have multiple children, which one would be picked?)
> 
> The `child` variable is not used in mesh code, i will always be 0

maybe it could be changed to "parent".. when building layers it's useful to know the device parent in order to avoid cyclical connection 🤔 .

Of course you can still have a cyclic like A -> B -> C -> A but you can avoid this by forcing the current explored layer to connect to to be at distance 1 from yours
( A -> B C -> A) -> B goes offline -> A sees C with good RSSI, but looking at C parent it knows it should be avoided.

zhangyanjiaoesp, 2023-08-31:
> maybe it could be changed to "parent".. when building layers it's useful to know the device parent in order to avoid cyclical connection 🤔 .
>
> Of course you can still have a cyclic like A -> B -> C [...]

1. change the `child` to `parent` may introduce a breaking change, so we won't consider to change it.
2. I have checked the mesh code, when one mesh node receives the parent disconnect event with reason `MESH_REASON_CYCLIC(100)`, it will add this parent into the `parent_cyclic_list`, and then try to find a new parent. This mesh node will ignore the parents in the `parent_cyclic_list` in the scan phase until it connects to a new parent. After it connects to new parent, the `parent_cyclic_list` will be cleared.

closed 2023-10-20 by KonssnoK