espressif/esp-idf #12019: [ESP32S3] v4.4 - Question: Support for high performances mesh and BLE beaconing

KonssnoK, 2023-08-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
this is an explorative issue to understand if it's possible with IDF to support high performances mesh with very low performances BLE.

From what we could see,
there is no middle ground between putting the power saving mode of the WIFI to "NONE" or "MODEM_MIN".

We wanted to put a BLE beacon in our setups, where the beaconing would occur once every minute (or every 5 minutes).
Therefore the bandwidth requirements on the BLE are very small.
Meanwhile, having Modem_min as power setting degrades the performances of mesh networking (up to 20x degraded latency on layer2).

The only way we could think of to make this combination work was to periodically switch off the mesh, change the power saving, switch on the BLE, get the beacon, go back to full mesh.
It really doesn't seem a good idea in terms of network reshaping and connection traffic.

Do you have any idea if it is possible in any other way to use the BLE just to intercept a beacon at a relatively low pace (minutes to hours) without impacting the MESH so much?

Thanks!

zhangyanjiaoesp, 2023-08-09:
@KonssnoK What do you want to do with BLE beacon？

KonssnoK, 2023-08-09:
@zhangyanjiaoesp we want to read a temperature, so we don't need anything faster than one measurement every minute.
At the same time, we need the mesh wifi network to perform in a timely manner (low latency).

KonssnoK, 2023-08-09:
is it compatible with esp mesh?
specifically i don't know if `esp_now_init  `can be called after `esp_mesh_start` (or viceversa).

zhangyanjiaoesp, 2023-08-09:
Yes, ESP-NOW is compatible with esp wifi mesh. You can call `esp_now_init` after `esp_mesh_start`.

KonssnoK, 2023-08-09:
thanks, we'll look into that!
I close the issue for the time being.

closed 2023-08-09 by KonssnoK