espressif/esp-idf #12023: Adding esp_https_server dependency to project doesn't work

dannybackx, 2023-08-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

add -I.../components/esp_https_server/include to the build if I specify that component in main/CMakeLists.txt

### What is the actual behavior?

doesn't do that

### Steps to reproduce.

My CMakeLists.txt looks like this. Works except for the https server that won't get included :
```
idf_component_register(
  SRCS App.cpp Ota.cpp WebServer.cpp Lcd.cpp
  INCLUDE_DIRS "." ${BUILD_DIR}
  REQUIRES
        esp_wifi
        driver
        esp_lcd esp_lcd_touch esp_lcd_touch_gt911
        lvgl esp_lvgl_port
        wifi_connect esp_netif nvs_flash esp_event wpa_supplicant
        ftpserver
        app_update esp_http_server esp_https_server
        littlefs
        acmeclient
)

add_custom_command(OUTPUT ${BUILD_DIR}/build.h
        COMMAND ${COMPONENT_DIR}/generate_build_date ${BUILD_DIR}/build.h
        DEPENDS ${SOURCES}
        VERBATIM)

add_custom_target(buildh DEPENDS ${BUILD_DIR}/build.h)
add_dependencies(${COMPONENT_LIB} buildh)
```

### More Information.

See https://sourceforge.net/p/esp32s3-4827s043/code/HEAD/tree/trunk/https/main/CMakeLists.txt

dannybackx, 2023-08-05:
I suggest adding an "else" branch that issues a warning.
```
::::::::::::::
esp_https_server/CMakeLists.txt
::::::::::::::
if(CONFIG_ESP_HTTPS_SERVER_ENABLE)
    set(src "src/https_server.c")
    set(inc "include")
endif()

idf_component_register(SRCS ${src}
                    INCLUDE_DIRS ${inc}
                    REQUIRES esp_http_server esp-tls
                    PRIV_REQUIRES lwip)

```

mahavirj, 2023-08-07:
> I suggest adding an "else" branch that issues a warning.

That would produce a warning for all examples that do not enable `CONFIG_ESP_HTTPS_SERVER_ENABLE`. This component has been kept conditionally enabled, since it also depends on the TLS server config from ESP-TLS abstraction layer. We recommend using our examples as getting started point as they are correctly set to handle such dependencies, e.g., `examples/protocols/https_server/simple` for this case.

dannybackx, 2023-08-07:
Both the build system (you need to select a component in main/CMakeLists.txt) and the normal linker already ensure components are pulled in only when selected *and* called.

closed 2023-11-07 by espressif-bot