espressif/esp-idf #12026: A bit of failure reporting in esp-tls

dannybackx, 2023-08-06:
### Is your feature request related to a problem?

When debugging a https server, mbedtls error codes are lost (in components/esp-tls).
This is frustrating.

### Describe the solution you'd like.

```
*** components/esp_https_server/src/https_server.c~     2023-08-06 17:33:17.172638405 +0200
--- components/esp_https_server/src/https_server.c      2023-08-06 17:38:34.450871360 +0200
***************
*** 125,136 ****
      ESP_LOGI(TAG, "performing session handshake");
      int ret = esp_tls_server_session_create(global_ctx->tls_cfg, sockfd, tls);
      if (ret != 0) {
! #if 0
!         ESP_LOGE(TAG, "esp_tls_create_server_session failed");
! #else
        char buf[120];
        mbedtls_strerror(ret, buf, sizeof(buf));
          ESP_LOGE(TAG, "esp_tls_create_server_session failed, 0x%04x %s", -ret, buf);
  #endif
          goto fail;
      }
--- 125,136 ----
      ESP_LOGI(TAG, "performing session handshake");
      int ret = esp_tls_server_session_create(global_ctx->tls_cfg, sockfd, tls);
      if (ret != 0) {
! #ifdef CONFIG_ESP_TLS_USING_MBEDTLS
        char buf[120];
        mbedtls_strerror(ret, buf, sizeof(buf));
          ESP_LOGE(TAG, "esp_tls_create_server_session failed, 0x%04x %s", -ret, buf);
+ #else
+         ESP_LOGE(TAG, "esp_tls_create_server_session failed");
  #endif
          goto fail;
      }

```

AdityaHPatwardhan, 2023-11-06:
Hi @dannybackx, thanks for the issue,
We already have an API in esp-tls called `esp_tls_get_and_clear_last_error` (ref - https://github.com/espressif/esp-idf/blob/b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e/components/esp-tls/esp_tls.h#L618)
This shall provide you the last error occurred for esp-tls of respective type.
If you give the error type as `ESP_TLS_ERROR_TYPE_MBEDTLS` to this API then you shall get the last mbedtls error that has occurred in esp-tls.
Let me know if the above API fixes your issue.
Thanks,
Aditya

dannybackx, 2023-11-07:
Hmm I don't see my reply turning up here so I'll copy it.
It looks like a good start but I don't think the current IDF components export the info I need to get that call to work.

Next to the function you point to, I also find esp_tls_get_error_handle() which I appear to need to fetch an error handle. The argument to that is an esp_tls_t pointer which I only get from user_cb.

So if I create a secure web server and pass it certificate info, it'll call mbedtls functions for that. If these fail, then I don't know where to get the info from.

    Danny

Note also that my example in the original question can't be resolved with this - user_cb is never called.

dannybackx, 2023-12-26:
I don't see any work on this (in esp_tls.h) in 5.2 beta 1 ?

hmalpani, 2023-12-27:
Hello @dannybackx 
We have an internal MR in review for adding this.

closed 2024-02-18 by espressif-bot

dannybackx, 2024-02-21:
I confirmed that this works for me, thanks.

See svn+ssh://dannybackx@svn.code.sf.net/p/esp32-s3-86-box-project-base/code/trunk/alarm/main/WebServer.cpp lines 64..83 .

Example output (only the httpsErrorHandler line is relevant) : 
E (20:03:24.172) App: Failed alloc: size 16717 caps 0804 (caps 8bit internal) fn heap_caps_calloc, task httpd
E (20:03:24.173) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (20:03:24.180) esp-tls-mbedtls: create_ssl_handle failed, returned [0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)
E (20:03:24.190) esp_https_server: esp_tls_create_server_session failed, 0x0001
E (20:03:24.198) WebServer: httpsErrorHandler: error 0, 8017 UNKNOWN ERROR CODE (8000) : LMS - LMS failed to allocate space for a private key
E (20:03:24.207) httpd: httpd_accept_conn: session creation failed
W (20:03:24.218) httpd: httpd_server: error accepting new connection

dannybackx, 2024-02-24:
Looks like one of the cases doesn't work : 

In the "performing session handshake" piece (line 170), the new code adds printing the hex value of ret and then jumps to "fail" but then only zeroes get reported.
```
E (11:07:21.858) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7480
E (11:07:21.859) esp_https_server: esp_tls_create_server_session failed, 0x7480
E (11:07:21.863) WebServer: httpsErrorHandler: error 0, 0000 
E (11:07:21.870) httpd: httpd_accept_conn: session creation failed
W (11:07:21.876) httpd: httpd_server: error accepting new connection
```