espressif/esp-idf #12027: Option to limit the number of core dump writes to flash

nomis, 2023-08-06:
### Is your feature request related to a problem?

I'd like to be able to enable the `coredump` partition but during development there will be times where the application crashes immediately on startup. This results in a fast loop of repeatedly overwriting a core dump to the same area of flash.

I'm concerned about wearing out the flash when an crash involving a reboot loop occurs.

### Describe the solution you'd like.

I'd like to be able to limit the number of core dump writes to flash, so that after a core dump has been written no more writes will occur until the existing core dump is erased.

### Additional context.

Currently only one core dump to flash is supported so the only valid limit (if enabled) will be `1` until storage of multiple core dumps becomes a feature.

The only change required to implement this would be to check if there is already a core dump present in `esp_core_dump_to_flash()` and `return` instead of writing the core dump.

chipweinberger, 2023-08-20:
flash can be written to ~50,000 times! 

Don't be too concerned!

nomis, 2023-08-20:
> flash can be written to ~50,000 times!
> 
> Don't be too concerned!

You do realise that that's not a very large number?

At a rate of 1 core dump per second the flash write cycles could be exceeded in 1-2 days. For an unattended device stuck in a fast crash loop with nothing obviously wrong that could happen.

I've made a PR to make it possible to skip writing a core dump to flash if one already exists. A future improvement would be to support storing multiple core dumps until the partition (or partitions) becomes full but this will do for now.

o-marshmallow, 2023-08-21:
Hello @nomis ,

If you are worried about crash loop that would write the coredump to flash again and again, you can configure the panic behavior from `ESP_SYSTEM_PANIC_PRINT_REBOOT` to `ESP_SYSTEM_PANIC_PRINT_HALT`. As such, the board will not reboot but simply halt after writing the coredump.

chipweinberger, 2023-08-21:
you can also set the ESP_SYSTEM_PANIC_REBOOT_DELAY

rnd-ash, 2023-08-21:
As a potential idea (I have no idea how hard it would be to implement)... Would it be possible to store some kind of SHA of the current coredump, and when a new one is created, compare its SHA to the coredump on NVS, and only overwrite if they are different?

I ask this since usually if a program is crashing very early on after boot, the coredump should be 100% identical since hardly any variance has occurred in the programs execution