espressif/esp-idf #12028: 弱网环境 | rssi=70左右，esp32 开始能正常连接，自动断开后，无法重连

jiafeng-1, 2023-08-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.4

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32c3-mini-1

### Power Supply used.

USB

### What is the expected behavior?

esp断开连接后，正常应该能正常恢复。目前再网络条件不好的情况下无法正常恢复。只能重启操作才能恢复网络。

### What is the actual behavior?

目前测试，将设备放到距离较远的位置，运行一段时间后就掉线。虽然一直重连，但是无法成功。掉线后移动esp32距离wifi近一些，也是无法恢复的。只有重启才可以正常恢复网络

### Steps to reproduce.

重连wifi
```c
····
case WIFI_EVENT_STA_DISCONNECTED:

        /* This is a workaround as ESP32 WiFi libs don't currently
           auto-reassociate. */
        gl_sta_connected = false;
        memset(gl_sta_ssid, 0, 32);
        memset(gl_sta_bssid, 0, 6);
        gl_sta_ssid_len = 0;
      
        raco_BluFi_adv();

        ESP_LOGW(TAG, "free_heap_size = %d", esp_get_free_heap_size());

        wifi_event_sta_disconnected_t *sta_disconnect_evt =(wifi_event_sta_disconnected_t *) event_data;
        ESP_LOGW(TAG, "wifi disconnect reason:%d",sta_disconnect_evt->reason);

 

        // raco_retry_connect_wifi();
   
        // xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        break;
····
void raco_retry_connect_wifi()
{
    static uint8_t s_retry_num = 0;
    if(s_retry_num < 3)
[... 493 lines cut ...]
D (1316600) wifi:connect status 1 -> 3
D (1316600) wifi:disable connect timer
D (1316600) wifi:clear scan ap list
D (1316610) event: running post WIFI_EVENT:5 with handler 0x4206bce0 and context 0x3fca9cac on loop 0x3fca8a7c
0x4206bce0: wifi_default_action_sta_disconnected at /home/jiafeng/esp/esp-idf/components/esp_wifi/src/wifi_default.c:101

D (1316620) esp_netif_handlers: esp_netif action disconnected with netif0x3fca99a4 from event_id=5
D (1316620) esp_netif_lwip: check: remote, if=0x3fca99a4 fn=0x42068a34
0x42068a34: esp_netif_down_api at /home/jiafeng/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1279

D (1316630) esp_netif_lwip: esp_netif_down_api esp_netif:0x3fca99a4
D (1316640) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3fca99a4
D (1316640) esp_netif_lwip: if0x3fca99a4 start ip lost tmr: already started
D (1316650) esp_netif_lwip: check: local, if=0x3fca99a4 fn=0x420694dc
0x420694dc: esp_netif_update_default_netif_lwip at /home/jiafeng/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:188

D (1316660) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3fca99a4
D (1316670) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (1316670) event: running post WIFI_EVENT:5 with handler 0x4200bb76 and context 0x3fca9d84 on loop 0x3fca8a7c
0x4200bb76: wifi_event_handler at /home/jiafeng/jiafeng/code/Bioradar/software/radar_comm_board/Bioradar_raco_comm_for_esp32/build/../components/blu_fi_app/blu_fi_app.c:587

W (1316680) [Blu_Fi_TASK]: free_heap_size = 5208
W (1316690) [Blu_Fi_TASK]: wifi disconnect reason:201
```
```

Xiehanxin, 2023-08-07:
hi @jiafeng-1 从Log 看没连上的原因是没有扫描到 AP （201），问一下是某个路由器有这个情况还是都有，然后如果是特定路由器的话能提供下路由器的型号吗，另外问一下你这边有抓包环境吗

jiafeng-1, 2023-08-08:
> hi @jiafeng-1 从Log 看没连上的原因是没有扫描到 AP （201），问一下是某个路由器有这个情况还是都有，然后如果是特定路由器的话能提供下路由器的型号吗，另外问一下你这边有抓包环境吗

1. 对的，是没有扫描到AP，但是重启是可以扫描到的。
2. 应该是所有的都有，目前客户和我在公司测试都遇到了，弱网环境，上行带宽在0.7-1.2Mbps，下行带宽4.5-5.7Mbps。rssi 在-50-70的情况下断网后，不能正常恢复。
3. 客户的网络环境是多相同的SSID网络环境。我已经手动把漫游值调成了-90，应该不至于产生漫游。公司的网络是独立的ap
4. 目前用的Ruijie RG-EW3200GX 、TP-LINK TL-WAR6200L两个型号。
5. 我可以尝试抓包，但是不清楚怎么抓，抓那些数据。

jiafeng-1, 2023-08-08:
[Uploading esp32_wireshark.zip…]() 这是我使用wireShark抓了一些包，不知道有没有用！

Xiehanxin, 2023-08-08:
> Uploading esp32_wireshark.zip… 这是我使用wireShark抓了一些包，不知道有没有用！

不好意思，我这边下载不了你的附件，打开还是这个issue的链接，另外你如果抓了包，可以提供下设备的MAC地址，路由器的账号密码信息吗

jiafeng-1, 2023-08-10:
[归档.zip](https://github.com/espressif/esp-idf/files/12308422/default.zip)
您好，我这边抓取了，相关数据，可以看下，mac地址等相关信息在压缩包里。还有一个文件太大无法上传。

> > Uploading esp32_wireshark.zip… 这是我使用wireShark抓了一些包，不知道有没有用！
> 
> 不好意思，我这边下载不了你的附件，打开还是这个issue的链接，另外你如果抓了包，可以提供下设备的MAC地址，路由器的账号密码信息吗

notcaremath, 2023-08-19:
遇到了同样的问题，我在重连中加了尝试次数，超次数重启设备然后可以连接，关注后续解决

Jacques-Zhao, 2023-08-30:
@jiafeng-1 你这个连不上的原因是这个打印导致的 “wifi:m f probe req l=0”，这个代表WiFi的 buffer满了，所以收发包会有异常，考虑到你们当前剩余内存还比较多，你可以尝试先把WiFi动态buffer调大一些

01hello, 2023-08-31:
@jiafeng-1  您好我的情况和您的是一样的，请问有解决吗

maenkai, 2024-01-04:
我也遇到了同样的情况，希望能得到解决

Hard-bitten, 2024-06-26:
遇到了同样的问题，有解决方案么？

MaxwellAlan, 2024-06-26:
Hi @Hard-bitten 

关于这个问题，帮忙提供一下详细的复现log（如果能提供WIFI无线抓包更好）及复现步骤，以便我们进一步分析

1.测试的IDF commit
2.测试的路由器型号及配置；
3.复现的 demo code 及 sdkconfig；
4.所用的模组型号及芯片型号等信息

谢谢。

closed 2024-08-09 by Sherry616