espressif/esp-idf #12029: adc1, adc2 one shot read issue (when switching)

imkifel, 2023-08-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32

### Power Supply used.

USB

### What is the expected behavior?

expects the adc_raw value of both adc1 and adc2 as 10 bit raw values.

### What is the actual behavior?

if i try to define adc1 & adc2 configuration in separate functions, when switching between adc1 and adc2 one shot reed, the raw value goes above 10 bit(above 1023).

### Steps to reproduce.

[oneshot_read_main.txt](https://github.com/espressif/esp-idf/files/12274945/oneshot_read_main.txt)

### Debug Logs.

```plain
I (12436) EXAMPLE: ADC2 Channel[2] Raw Data: 4095
I (12536) EXAMPLE: ADC2 Channel[5] Raw Data: 3051

I (13536) EXAMPLE: ADC1 Channel[5] Raw Data: 4095
I (13636) EXAMPLE: ADC1 Channel[7] Raw Data: 4095

I (14636) EXAMPLE: ADC2 Channel[2] Raw Data: 4095
I (14736) EXAMPLE: ADC2 Channel[5] Raw Data: 2885

I (15736) EXAMPLE: ADC1 Channel[5] Raw Data: 4095
I (15836) EXAMPLE: ADC1 Channel[7] Raw Data: 4095

I (16836) EXAMPLE: ADC2 Channel[2] Raw Data: 4095
I (16936) EXAMPLE: ADC2 Channel[5] Raw Data: 2583

I (17936) EXAMPLE: ADC1 Channel[5] Raw Data: 4095
I (18036) EXAMPLE: ADC1 Channel[7] Raw Data: 4095
```

### More Information.

when trying the example code it works when both adc1 and adc2 are configured from app main, but if i try to configure from different place, the value jumps from 10 bit adc to higher.

v-3n0m, 2023-08-07:
I am also facing the same issue. Unable to toggle the ADC1 and ADC2. In my case the ADC1 is configured by default and the ADC2 is been configured dynamicaly inside an anthrt function and when at first it works fine but after that wher i try to configure the adc1 agin the value shoots up to more than 10 bit. same as shown above by @imkifel

Icarus113, 2023-08-08:
@imkifel  Hi, I think it's because the variable `config` passed to the `adc_oneshot_config_channel` is the .bss one in L59 which is not initialised. If you 
- just move the stack variable `config` in L72 to your `app_main` function, and remove the .bss `config` in L59
- just remove the stack variable `config` in L72, and instead, do initialisation to the .bss config here

Then you can get 10-bit-width result.

Icarus113, 2023-08-08:
> In my case the ADC1 is configured by default and the ADC2 is been configured dynamicaly inside an anthrt function and when at first it works fine but after that wher i try to configure the adc1 agin the value shoots up to more than 10 bit

The `adc_oneshot_config_channel` can save the newest configurations to each channel. If in your case ADC1 won't be configured dynamically, then you just configure it once. If you configure it multiple times, you'll need to make sure each time the `adc_oneshot_chan_cfg_t` set to the channel is correct.

@v-3n0m

imkifel, 2023-08-09:
@Icarus113 hii, thanks for the reply, i have tried that way as you mentioned but the resuts are same, when i toggle from adc2 to adc1 the adc readings jump from 10 bit to 12 bit value.

Icarus113, 2023-08-10:
Hi @imkifel , here's my result.
```
I (116948) EXAMPLE: ADC1 Channel[5] Raw Data: 1023
I (117048) EXAMPLE: ADC1 Channel[7] Raw Data: 1023

I (118048) EXAMPLE: ADC2 Channel[2] Raw Data: 1023
I (118148) EXAMPLE: ADC2 Channel[5] Raw Data: 1023

I (119148) EXAMPLE: ADC1 Channel[5] Raw Data: 1023
I (119248) EXAMPLE: ADC1 Channel[7] Raw Data: 1023

I (120248) EXAMPLE: ADC2 Channel[2] Raw Data: 1023
I (120348) EXAMPLE: ADC2 Channel[5] Raw Data: 1023

I (121348) EXAMPLE: ADC1 Channel[5] Raw Data: 1023
I (121448) EXAMPLE: ADC1 Channel[7] Raw Data: 1023

I (122448) EXAMPLE: ADC2 Channel[2] Raw Data: 1023
I (122548) EXAMPLE: ADC2 Channel[5] Raw Data: 1023

I (123548) EXAMPLE: ADC1 Channel[5] Raw Data: 1023
I (123648) EXAMPLE: ADC1 Channel[7] Raw Data: 1023
```
Maybe you can share you code snippet and we can check

imkifel, 2023-08-14:
[oneshot_read_main.txt](https://github.com/espressif/esp-idf/files/12335700/oneshot_read_main.txt)
Hi @Icarus113, here is the code snippet. "adc_oneshot_chan_cfg_t config;" this line i have to declare globally to access these from other functions. as same as the example code it works but when i split the adc1 and 2 configurations its not reading as your result. can you please share me the code where configuring adc1and adc2 from two different functions as in my sample code.  thanks in advance.

closed 2023-08-25 by imkifel