espressif/esp-idf #12030: Cannot access some members of wifi_pkt_rx_ctrl_t

aldardou, 2023-08-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1962-g53ff7d43db

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected it to be able to access the value of all public members of wifi_pkt_rx_ctrl_t and I can access many (e.g. rssi, timestamp, mac, rate) and I expect to be able to access cwb and sig_mode the same way. They are defined in the same struct.

### What is the actual behavior?

Instead when building the project so far only for the fields cwb and sig_mode the compiler claims error: 'wifi_pkt_rx_ctrl_t' {aka 'esp_wifi_rxctrl_t'} has no member named 'cwb'.

### Steps to reproduce.

In my csi_callback function I try to read out as much information about my read CSI data as possible (all of them except cwb and sig_mode actually work). Here is the relevant code snippet causing the issue:

    uint8_t* csi_buffer = (uint8_t*)csi_info->buf;
    uint16_t csi_length = csi_info->len;
    wifi_pkt_rx_ctrl_t csi_meta = csi_info->rx_ctrl;
    unsigned csi_chan = csi_meta.channel;
    signed csi_rssi = csi_meta.rssi;
    unsigned csi_timestamp = csi_meta.timestamp;
    uint8_t* smac = csi_info->mac;
    unsigned csi_rate = csi_meta.rate;
    //After here is where it starts being fishy
    unsigned channel_bandwidth = csi_meta.cwb;
    unsigned csi_sig_mode = csi_meta.sig_mode;

### Build or installation Logs.

```plain
Executing action: all (aliases: build)
Running ninja in directory /home/alexander/hardwareesp32/csiextractor/build
Executing "ninja all"...
[1/1] cd /home/alexander/hardwareesp32/csiextractor/build/bootloader/esp-idf/esptool_py && /home/alexander/.espressif/python_env/i...ion_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/alexander/hardwareesp32/csiextractor/build/bootloader/bootloader.bin
Bootloader binary size 0x53f0 bytes. 0x2c10 bytes (34%) free.
[3/9] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/csiextractor.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/csiextractor.c.obj 
/home/alexander/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.2-dev-1962-g53ff7d43db\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DUNITY_INCLUDE_CONFIG_H -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/alexander/hardwareesp32/csiextractor/build/config -I/home/alexander/hardwareesp32/csiextractor/main -I/home/alexander/esp/esp-idf/components/newlib/platform_include -I/home/alexander/esp/esp-idf/components/freertos/config/include -I/home/alexander/esp/esp-idf/components/freertos/config/include/freertos -I/home/alexander/esp/esp-idf/components/freertos/config/riscv/include -I/home/alexander/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/alexander/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/home/alexander/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -I/home/alexander/esp/esp-idf/components/freertos/esp_additions/include -I/home/alexander/esp/esp-idf/components/esp_hw_support/include -I/home/alexander/esp/esp-idf/components/esp_hw_support/include/soc -I/home/alexander/esp/esp-idf/components/esp_hw_support/include/soc/esp32c6 -I/home/alexander/esp/esp-idf/components/esp_hw_support/port/esp32c6/. -I/home/alexander/esp/esp-idf/components/esp_hw_support/port/esp32c6/private_include -I/home/alexander/esp/esp-idf/components/heap/include -I/home/alexander/esp/esp-idf/components/log/include -I/home/alexander/esp/esp-idf/components/soc/include -I/home/alexander/esp/esp-idf/components/soc/esp32c6 -I/home/alexander/esp/esp-idf/components/soc/esp32c6/include -I/home/alexander/esp/esp-idf/components/hal/esp32c6/include -I/home/alexander/esp/esp-idf/components/hal/include -I/home/alexander/esp/esp-idf/components/hal/platform_port/include -I/home/alexander/esp/esp-idf/components/esp_rom/include -I/home/alexander/esp/esp-idf/components/esp_rom/include/esp32c6 -I/home/alexander/esp/esp-idf/components/esp_rom/esp32c6 -I/home/alexander/esp/esp-idf/components/esp_common/include -I/home/alexander/esp/esp-idf/components/esp_system/include -I/home/alexander/esp/esp-idf/components/esp_system/port/soc -I/home/alexander/esp/esp-idf/components/esp_system/port/include/riscv -I/home/alexander/esp/esp-idf/components/esp_system/port/include/private -I/home/alexander/esp/esp-idf/components/riscv/include -I/home/alexander/esp/esp-idf/components/lwip/include -I/home/alexander/esp/esp-idf/components/lwip/include/apps -I/home/alexander/esp/esp-idf/components/lwip/include/apps/sntp -I/home/alexander/esp/esp-idf/components/lwip/lwip/src/include -I/home/alexander/esp/esp-idf/components/lwip/port/include -I/home/alexander/esp/esp-idf/components/lwip/port/freertos/include -I/home/alexander/esp/esp-idf/components/lwip/port/esp32xx/include -I/home/alexander/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/alexander/esp/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/alexander/esp/esp-idf/components/esp_ringbuf/include -I/home/alexander/esp/esp-idf/components/efuse/include -I/home/alexander/esp/esp-idf/components/efuse/esp32c6/include -I/home/alexander/esp/esp-idf/components/driver/include -I/home/alexander/esp/esp-idf/components/driver/deprecated -I/home/alexander/esp/esp-idf/components/driver/analog_comparator/include -I/home/alexander/esp/esp-idf/components/driver/dac/include -I/home/alexander/esp/esp-idf/components/driver/gpio/include -I/home/alexander/esp/esp-idf/components/driver/gptimer/include -I/home/alexander/esp/esp-idf/components/driver/i2c/include -I/home/alexander/esp/esp-idf/components/driver/i2s/include -I/home/alexander/esp/esp-idf/components/driver/ledc/include -I/home/alexander/esp/esp-idf/components/driver/mcpwm/include -I/home/alexander/esp/esp-idf/components/driver/parlio/include -I/home/alexander/esp/esp-idf/components/driver/pcnt/include -I/home/alexander/esp/esp-idf/components/driver/rmt/include -I/home/alexander/esp/esp-idf/components/driver/sdio_slave/include -I/home/alexander/esp/esp-idf/components/driver/sdmmc/include -I/home/alexander/esp/esp-idf/components/driver/sigma_delta/include -I/home/alexander/esp/esp-idf/components/driver/spi/include -I/home/alexander/esp/esp-idf/components/driver/temperature_sensor/include -I/home/alexander/esp/esp-idf/components/driver/touch_sensor/include -I/home/alexander/esp/esp-idf/components/driver/twai/include -I/home/alexander/esp/esp-idf/components/driver/uart/include -I/home/alexander/esp/esp-idf/components/driver/usb_serial_jtag/include -I/home/alexander/esp/esp-idf/components/esp_pm/include -I/home/alexander/esp/esp-idf/components/mbedtls/port/include -I/home/alexander/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/alexander/esp/esp-idf/components/mbedtls/mbedtls/library -I/home/alexander/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/alexander/esp/esp-idf/components/esp_bootloader_format/include -I/home/alexander/esp/esp-idf/components/esp_app_format/include -I/home/alexander/esp/esp-idf/components/bootloader_support/include -I/home/alexander/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/alexander/esp/esp-idf/components/esp_partition/include -I/home/alexander/esp/esp-idf/components/app_update/include -I/home/alexander/esp/esp-idf/components/esp_mm/include -I/home/alexander/esp/esp-idf/components/spi_flash/include -I/home/alexander/esp/esp-idf/components/pthread/include -I/home/alexander/esp/esp-idf/components/esp_timer/include -I/home/alexander/esp/esp-idf/components/app_trace/include -I/home/alexander/esp/esp-idf/components/esp_event/include -I/home/alexander/esp/esp-idf/components/nvs_flash/include -I/home/alexander/esp/esp-idf/components/esp_phy/include -I/home/alexander/esp/esp-idf/components/esp_phy/esp32c6/include -I/home/alexander/esp/esp-idf/components/vfs/include -I/home/alexander/esp/esp-idf/co [... cut ...]
/home/alexander/hardwareesp32/csiextractor/main/csiextractor.c:164:14: warning: unused variable 'channel_bandwidth' [-Wunused-variable]
  164 |     unsigned channel_bandwidth = csi_meta.cwb;
      |              ^~~~~~~~~~~~~~~~~
[4/9] Completed 'bootloader'
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/alexander/hardwareesp32/csiextractor/build/log/idf_py_stderr_output_23863 and /home/alexander/hardwareesp32/csiextractor/build/log/idf_py_stdout_output_23863
```

igrr, 2023-08-08:
Please note that for C6 there is a different definition of the hardware packet header, due to
https://github.com/espressif/esp-idf/blob/53ff7d43dbff642d831a937b066ea0735a6aca24/components/esp_wifi/include/esp_wifi_types.h#L442-L443
and the definition for C6 is here:
https://github.com/espressif/esp-idf/blob/53ff7d43dbff642d831a937b066ea0735a6aca24/components/esp_wifi/include/esp_wifi_he_types.h#L204

aldardou, 2023-08-08:
> Please note that for C6 there is a different definition of the hardware packet header, due to [...]

Thank you igrr. :) I was not aware of there being a different definition. This makes it clear why I was not able to access the meta data I was looking for, therefore I will close this issue.