espressif/esp-idf #12031: GATT CLIENT & SERVER NOT WORKING WITH 128 BIT UUID

aswathreddy, 2023-08-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When using 2 bytes uuid for gatt client and gatt server working properly when using 128 uuid it is not working 
I have made all changes with respective to 128 uuid 
Service uuid Little endian for client and server with big endian I am able to connect not sharing data. Anyone has clue for it ?

zhp0406, 2023-08-08:
It seems that the format of the UUID you have set is incorrect.

If the UUID in the BLE server is: 0000-A00C-0000-0000-0123-4567-89AB-CDEF

The BLE client should use the following code:

```
#define ESP_UUID_LEN_128    16

esp_bt_uuid_t service_id = {
    .len = ESP_UUID_LEN_128,
    .uuid.uuid128 = { 0xEF, 0xCD, 0xAB, 0x89, 0x67, 0x45, 0x23, 0x01, 0x00, 0x00, 0x00, 0x00, 0x0C, 0xA0, 0x00, 0x00 },
};
```

If there are still issues, please provide more log information.Thanks.

aswathreddy, 2023-08-08:
[image]
 Thanks for resolving my previous issues please assist how to handle this error

zhp0406, 2023-08-09:
It seems that you used API "esp_ble_gattc_get_descr_by_char_handle" and  "esp_ble_gattc_write_char_descr" incorrectly，

However, the provided logs are insufficient to determine the cause of the issue. Please increase the log level and provide more information.

Additionally, for the usage of "esp_ble_gattc_get_descr_by_char_handle" and "esp_ble_gattc_write_char_descr," you can refer to the example "esp-idf/examples/bluetooth/bluedroid/ble/gatt_client" for the sample usage.

zhp0406, 2023-08-10:
@aswathreddy 
I've written a test. You can use "git am 0001-test-bt-Test-128-bit-uuid-both-in-server-and-client.patch" to apply this patch and then proceed with the testing.

[0001-test-bt-Test-128-bit-uuid-both-in-server-and-client.zip](https://github.com/espressif/esp-idf/files/12310666/0001-test-bt-Test-128-bit-uuid-both-in-server-and-client.zip)

zhp0406, 2023-08-10:
@aswathreddy 
You need to pay attention that The client-side has converted a 128-bit UUID to a 16-bit UUID.
If you still have questions, you can provide more information.
Thanks.

aswathreddy, 2023-08-10:
Thanks for your help. Its working now great

closed 2023-08-10 by aswathreddy