espressif/esp-idf #12033: Can flash my code to the ESP32-S3-WROOM-1?

onurdof, 2023-08-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I have an ESP32-S3-WROOM-1-N16 and I am not able to flash my code . It gives error like
A fatal error occurred: Failed to connect to ESP32-S3: No serial data received. I have boot and reset button When I pressed the boot button while flashing code it still gives error. I tried to using lower baud rate but it did not work
How can I flash my code to the chip?

radimkarnis, 2023-08-08:
Hello @onurdof,

I need some more information to debug your issue:
1) How are you connecting to the module? Are you using the TX/RX pins or pins 19/20?
2) What OS are you on? Do you see your module connected when you list the serial ports?
3) If you open any serial terminal program (IDF monitor, miniterm or others), connect to the chip and press the reset button, do you see any output?

`No serial data received` means there is no data coming out of the chip. This can be caused by many reasons, like HW issues, wrong connections, insufficient power supply, and more...

radimkarnis, 2023-08-10:
> I tried this and did not see any output

That means there is definitely a HW issue. The cp2102 / FTDI chips work, that's why you see a port, but the ESP chip doesn't.
Common reasons:
- the connections between the USB-to-UART chip and the ESP are wrong (reversed or disconnected TX/RX?)
- the ESP is not getting enough power or none at all
- the ESP is being held in RESET (can be caused by shorting EN pin to GND. Bad soldering can cause this
- the ESP is fried

Please try to verify these.

radimkarnis, 2023-08-21:
Closing this issue. Feel free to reopen it if necessary, thank you.

closed 2023-08-21 by radimkarnis