espressif/esp-idf #12037: esp.emac: emac_esp32_transmit(231): insufficient TX buffer size, sent_len:0, length:94

Sky-Soo-Ap, 2023-08-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1709-g4fc2e5cb95

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

This issue is related to the descriptor.
Please help to fix, thanks so much.

### What is the actual behavior?

emac_hal_transmit_frame: desc_iter->TDES0.Own:0, EMAC_LL_DMADESC_OWNER_CPU:0
esp.emac: emac_esp32_transmit(231): insufficient TX buffer size, sent_len:0, length:94

uint32_t emac_hal_transmit_frame(emac_hal_context_t *hal, uint8_t *buf, uint32_t length)
/* Check if the descriptor is owned by the Ethernet DMA (when 1) or CPU (when 0) */
        if (desc_iter->TDES0.Own != EMAC_LL_DMADESC_OWNER_CPU) {
            ESP_LOGE("emac_hal_transmit_frame", "desc_iter->TDES0.Own:%d, EMAC_LL_DMADESC_OWNER_CPU:%d", desc_iter->TDES0.Own, EMAC_LL_DMADESC_OWNER_CPU);
            goto err;
        }

### Steps to reproduce.

1.  Init ESP32 ethernet;
2.  ESP32 is TCP client and connect to a TCP server via ethernet interface;
3.  Keep sending data and receiving response packet from TCP server;

kostaond, 2023-08-11:
@Sky-Soo-Ap, `esp.emac: emac_esp32_transmit(231): insufficient TX buffer size, sent_len:0, length:94` usually happens when you try to transmit more data that the EMAC is able to process, i.e. theoretically, you are trying to transmit more than 100 Mbps (it's a bit less in the real world though). Anyway, it is kind of expected behavior. You have basically to options how to mitigate:
1) implement maximum throughput limit for your application as we did in `iperf`
2) try to increase Ethernet DMA Tx buffers (Component config -> Ethernet -> Support ESP32 internal EMAC controller -> Amount of Ethernet DMA Tx buffers). Note that this only helps to cover intermediate peaks in throughput. If the throughput is constantly high, you sooner or later end up with `insufficient TX buffer size` anyway.

Sky-Soo-Ap, 2023-08-14:
Hi [kostaond](https://github.com/kostaond), 
Transmit more than 100 Mbps is not possible, because this issue is base on sta2eth example:
- Wi-Fi throughput is up 20 MBit/s TCP throughput and 30 MBit/s UDP throughput over the air;
- There is a socket to send 50 bytes / 50ms via ethernet = 1000 bytes/second;

And, I already set ethernet TX/RX buffer to:
CONFIG_ETH_DMA_BUFFER_SIZE=1600
CONFIG_ETH_DMA_RX_BUFFER_NUM=30
CONFIG_ETH_DMA_TX_BUFFER_NUM=30

It is more easy to cause this issue when the CPU frequency is set to 240MHz.

Sky-Soo-Ap, 2023-08-14:
Hi [kostaond](https://github.com/kostaond), 
If this issue occurs, it will not recover back.
Following is the log file and sdkconfig.
[COM12_2023_08_14.09.30.44.827.txt](https://github.com/espressif/esp-idf/files/12330978/COM12_2023_08_14.09.30.44.827.txt)
[sdkconfig.log](https://github.com/espressif/esp-idf/files/12330979/sdkconfig.log)

kostaond, 2023-08-14:
Hi @Sky-Soo-Ap, since you send very small packets, it would be more appropriate to configure `CONFIG_ETH_DMA_BUFFER_SIZE` to smaller value. The default is 512B but if you know that your app mostly sends small packets, you can configure it even smaller. The thing is the whole DMA buffer is used to pass just 50B to EMAC.

Sky-Soo-Ap, 2023-08-14:
Hi [kostaond](https://github.com/kostaond), 
The point is this issue is not caused by ethernet TX buffer transmit more than 100Mbps;
And, it is not caused by the TX buffer being full, because I added log as below but did not see any.
That's why I think it's a timing issue.
[image]

kostaond, 2023-08-14:
Did you try to decrease size of `CONFIG_ETH_DMA_BUFFER_SIZE`? As I said, you use the full 1600 B for transmission of even very small frame. Therefore, throughput at the Ethernet output is definitely not 100Mbps as you pointed out but it might be falsely higher between CPU and EMAC.

The circled code snipped tells you that you have enough buffers with enough size allocated to transmit a packet. However, it does not tell that enough DMA buffers is available.

Please try to decrease size of `CONFIG_ETH_DMA_BUFFER_SIZE`. If it does not help. Could you please provide me more detailed steps how to reproduce?

Sky-Soo-Ap, 2023-08-21:
Hi [kostaond](https://github.com/kostaond), 
Enable this configuration CONFIG_ETH_TRANSMIT_MUTEX then this issue does not happen, so close it.
The configuration  CONFIG_ETH_TRANSMIT_MUTEX put at there (as the following screenshot) cause confuse;
Suggests put it to the position (as the following screenshot), thanks.
[image]

closed 2023-08-21 by Sky-Soo-Ap

kostaond, 2023-08-21:
@Sky-Soo-Ap thanks for your report. What is your application? Enabling the Tx Mutex is used when an application may access the Ethernet from multiple tasks. LwIP and L2TAP is example. Does your application do something similar?

Regarding the "TX mutex" option position, it is really placed strangely. However, it seems to be placed correctly when using `idf.py menuconfig`
[image]

Please create issue in [VS Code IDE](https://github.com/espressif/vscode-esp-idf-extension/issues/new/choose) repo to address it.