espressif/esp-idf #12038: Struggling with memory allocation and wifi:m f null

mbastida123, 2023-08-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, having read this thread https://github.com/espressif/esp-idf/issues/2184

I'm pretty sure that I have exactly the same problem. However I have decided to open a new thread since the version is totally different.

My setup:
ESP32-S3 with 2MB SPIRAM
Running IDF v4.4.4

The devices stores data on flash. For testing we now have a .csv file with 1000 lines and 20 bytes/line.

The device periodically connects the Wifi to download the data.
The .csv file is loaded into memory and from there a certain number of lines (we are testing with 300) are converted into a json. The json is aproximately 57000 bytes.

So, in order:
1. Load the csv in memory
2. Create the json char
3. Send the data

With 10 lines (instead of 300) everything works. However, with 300 lines there's no way to avoid the `wifi:m f null` error. And from there everything goes downhill.

We use SPIRAM for everything. I have set mbedTLS to force SPIRAM memory allocation with `mbedtls_platform_set_calloc_free`.
All the CSV and json are allocted in external SPIRAM.

According to heap_size I have more than enough space always (I have never seen anything lower than 1M).

I have `CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST` from the start also.

I'm running out of idea, so I would apreciate any help :)

igrr, 2023-08-08:
As a troubleshooting step, I would recommend using `heap_caps_print_heap_info` (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-information) separately for MALLOC_CAP_INTERNAL and MALLOC_CAP_SPIRAM, to make sure you have enough space in the internal memory.

mbastida123, 2023-08-08:
> As a troubleshooting step, I would recommend using `heap_caps_print_heap_info` (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-information) [...]
Hi @igrr 

`heap_caps_print_heap_info` before doing any heavy work:
```
  At 0x3de00000 len 2097151 free 2035135 allocated 59752 min_free 1989143
    largest_free_block 2031616 alloc_blocks 57 free_blocks 3 total_blocks 60
  Totals:
    free 2035135 allocated 59752 min_free 1989143 largest_free_block 2031616
Heap summary for capabilities 0x00000800:
  At 0x3fcb41b8 len 32767 free 119 allocated 31796 min_free 59
    largest_free_block 76 alloc_blocks 29 free_blocks 2 total_blocks 31
  At 0x3fcb3770 len 221088 free 31132 allocated 187284 min_free 13144
    largest_free_block 22528 alloc_blocks 433 free_blocks 2 total_blocks 435
  At 0x3fce9710 len 22308 free 4 allocated 21140 min_free 4
    largest_free_block 0 alloc_blocks 107 free_blocks 0 total_blocks 107
  At 0x3fcf0000 len 32768 free 4 allocated 31780 min_free 4
    largest_free_block 0 alloc_blocks 62 free_blocks 0 total_blocks 62
  At 0x600fe02c len 8132 free 7752 allocated 0 min_free 7752
    largest_free_block 7680 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 38883 allocated 272124 min_free 20963 largest_free_block 22528
```

After the heavy work is done and before the esp_http_client_perform() function is called. This means that at this point I already have a char* containing the json that I want to send:
```
Heap summary for capabilities 0x00000400:
  At 0x3de00000 len 2097151 free 1483295 allocated 556768 min_free 1477483
    largest_free_block 1474560 alloc_blocks 13763 free_blocks 2 total_blocks 13765
  Totals:
    free 1483295 allocated 556768 min_free 1477483 largest_free_block 1474560
Heap summary for capabilities 0x00000800:
  At 0x3fcb41b8 len 32767 free 119 allocated 31796 min_free 7
    largest_free_block 76 alloc_blocks 29 free_blocks 2 total_blocks 31
  At 0x3fcb3770 len 221088 free 84 allocated 209920 min_free 4
    largest_free_block 76 alloc_blocks 2536 free_blocks 1 total_blocks 2537
  At 0x3fce9710 len 22308 free 4 allocated 21140 min_free 4
    largest_free_block 0 alloc_blocks 107 free_blocks 0 total_blocks 107
  At 0x3fcf0000 len 32768 free 4 allocated 31780 min_free 4
    largest_free_block 0 alloc_blocks 62 free_blocks 0 total_blocks 62
  At 0x600fe02c len 8132 free 4 allocated 6128 min_free 4
    largest_free_block 0 alloc_blocks 405 free_blocks 0 total_blocks 405
  Totals:
    free 215 allocated 300764 min_free 23 largest_free_block 76
```

Immediately after this line I get:
`W (24574) wifi:m f null`

Seems like I have ran out of internal memory? 

So as I see it there's something that is allocating internal memory. Maybe instead of using malloc I have to use heap_caps_malloc(size, MALLOC_CAP_SPIRAM) everywhere? I was using malloc because I had this configured:
[image]

igrr, 2023-08-08:
>   Totals:
>    free 215 allocated 300764 min_free 23 largest_free_block 76

Yes, looks like you only have 215 bytes of internal RAM left, and the largest continuous block is just 76 bytes.

Please try to lower the size set in [CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-malloc-alwaysinternal). Please also read the description of allocation logic [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#provide-external-ram-via-malloc). You can also try to raise the value of [CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-malloc-reserve-internal) if you have enough of RAM to start with.

To debug which code ends up allocating from internal RAM, you can use [heap tracing](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-tracing).

mbastida123, 2023-08-08:
> > Totals:
> > free 215 allocated 300764 min_free 23 largest_free_block 76
> 
> Yes, looks like you only have 215 bytes of internal RAM left, and the largest continuous block is just 76 bytes.
> 
> Please try to lower the size set in [CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-malloc-alwaysinternal), this can move more allocations into external RAM. Please also read the description of allocation logic [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#provide-external-ram-via-malloc). You can also try to raise the value of [CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-malloc-reserve-internal) if you have enough of RAM to start with.
> 
> To debug which code ends up allocating from internal RAM, you can use [heap tracing](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-tracing).

Thanks @igrr 

Chaning [CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-malloc-alwaysinternal) to 4096 and [CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-malloc-reserve-internal) to 131072 seemed to fix the issue.

So it's clear that we were running out of internal memory. We will try to force allocation (via `heap_caps_malloc()`) to external RAM insted of using `malloc()`. 

Again thanks for helping me resolve an issue in less than 2h! :)

closed 2023-08-08 by mbastida123