espressif/esp-idf #12041: ot_sleepy_device: idf.py menuconfig does not show Component config > OpenThread > Thread Operational Dataset option

stonecan, 2023-08-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

in ot_sleepy_device README.md it states to do the following to configure the Thread network dataset obtained from the BR:

"Configure the Openthread dataset using idf.py menuconfig in Component config ---> Openthread ---> Thread Operation Dataset, ensuring that the openthread sleepy device's dataset matches the dataset of the leader."

After this, build and flash the code.

### What is the actual behavior?

However, the "Thread Operation Dataset" option is not available when running idf.py menuconfig under Component config ---> Openthread

When building without setting the dataset, compiling stops with:

esp/ot_sleepy_device/main/esp_ot_sleepy_device.c:58:21: error: implicit declaration of function 'esp_openthread_auto_start'; did you mean 'esp_openthread_get_netif'? [-Werror=implicit-function-declaration]
   58 |     ESP_ERROR_CHECK(esp_openthread_auto_start(NULL));
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
/esp/esp-idf/components/esp_common/include/esp_err.h:117:30: note: in definition of macro 'ESP_ERROR_CHECK'
  117 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^

This may not be related - but not sure at this time.

### Steps to reproduce.

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/12292660/sdkconfig.zip)

chshu, 2023-08-09:
@stonecan Seems you were using an old IDF version, could you fetch the latest IDF master or release/v5.1 branch and try again?

stonecan, 2023-08-09:
@chshu I'm using release/v5.1 branch

chshu, 2023-08-09:
Could you double check the IDF version:
```
cd $IDF_PATH
git log
```
The latest release/v5.1 branch supports the Thread Operation Dataset configuration and autostart.

stonecan, 2023-08-09:
I ran this initially to clone the v5.1 release of IDF:
`$ git clone -b v5.1 --recursive https://github.com/espressif/esp-idf.git`
This did not include the ot_sleepy_device example source code which I separately downloaded (latest).
```
$ cd ~/esp/ot_sleepy_device
$ idf.py menuconfig
```
produces this when navigating to Component config > OpenThread
[image]
`$ git log` output attached below (zipped):
[output_git_log.zip](https://github.com/espressif/esp-idf/files/12298769/output_git_log.zip)

chshu, 2023-08-09:
So you downloaded the v5.1 release tag, but not the branch. Please download the latest release/v5.1 branch with the cmd:
```
git clone -b release/v5.1 --recursive https://github.com/espressif/esp-idf.git
```

stonecan, 2023-08-09:
Thank you. Now I see where the confusion came from: on https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32h2/get-started/linux-macos-setup.html I selected "stable (v5.1)" instead of "release (v5.1)"...

closed 2023-08-09 by stonecan