espressif/esp-idf #12042: Can't get Nimble's bleprph to work with security features

zzzh, 2023-08-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I'm using esp32c3, with esp-idf v4.4.2 in ubuntu 20.04.
I selected bonding, MITM security and use secure connection in menuconfig.
When I connect it with my phone, I see this:
NimBLE:  conn_itvl=36 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0

Seems none of my settings take effect?
I also checked ble_hs_cfg in the code, seems it's not used.

I have no idea what's going on, please help.

By the way, I tried it with esp-idf v5.1, it have this error: 
NimBLE: error setting advertisement data; rc=8
I can't continue without get this fixed...

ESPAbhinav, 2023-08-09:
Hi @zzzh ,
We tried to reproduce this issue but it working fine at our end. Mentioned log
` NimBLE: conn_itvl=36 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0`  is correct.
Once you pair your device you will see encrypted =1 and bonded =1 in the log. Also, bond,mitm, and sc flags of `ble_hs_cfg` are set to 1 (after doing changes from menuconfig/sdkconfig). Below are the references.

[image]

[image]
[nRf-Log.txt](https://github.com/espressif/esp-idf/files/12298640/nRf-Log.txt)
To get more about your issue can you share the following things :

1. Complete Console logs.
2. What exact failure you are observing or not working.
3. sdkconfig

zzzh, 2023-08-09:
Hi, @ESPAbhinav ,

Thanks for your quick respond.

I thought it would require the phone side to bond, and I don't understand why ble_hs_cfg is not used.
So, how to require bond from nimble side?

ESPAbhinav, 2023-08-09:
Hi , @zzzh,
The bonding process will only occur if you have enabled the flag `CONFIG_EXAMPLE_BONDING` in sdkconfig/menuconfig.
Enabling the `CONFIG_EXAMPLE_BONDING`  flag ensures that the nimble stack will automatically save the generated link key and bonding occurs.

zzzh, 2023-08-10:
Hi, @ESPAbhinav,
I did enable the flag CONFIG_EXAMPLE_BONDING in sdkconfig/menuconfig, I just hope to require bond from nimble side.
That is: at phone side, when connect to the nimble device, it's reminded to bond to it.

ESPAbhinav, 2023-08-11:
Hi, @zzzh,

Just to clarify, bonding and pairing are two different operations in BLE. Pairing usually implies the process of generating keys and bonding implies storage and distributing the keys. My previous answers are based on the above difference. 

But I believe you are referring to initiating pairing. So there are two ways to do this.

1. We are in slave role and master ( nrf ) is initiating the connection. So , usually pairing is initiated by central device.  To perform this operation from Phone side.
    - Connect with bleprph from the nRF app.
    - There you will get the option of "bond" in the 3 dots menu. Below is a reference.
    - If you click on bond then you will be reminded of the pairing request. Accept it and pairing will occur.
    - 
[image]

[image]
8e1
[image]
9)

2. You want to perform pairing which is initiated from nimble device side using the bleprph demo example. You can achieve this by below steps  
    - Connect with bleprph from the nRF app.
    - Read LE encrypted characteristic and you will be reminded for pairing request. Accept it and pairing will occur.
    - Below is the reference.
[image]

zzzh, 2023-08-11:
Hi, @ESPAbhinav,

As I know, slave can require to  bond,  I remember blueroid have a command which can do this, seems nimble just don't have such command.
The second way you recommended looks good, I'll try it.
Thank you so much for the help and detailed explanation.