espressif/esp-idf #12043: esp32-C6无法对ADC电压进行校准

woshiwstt50123, 2023-08-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://docs.espressif.com/projects/esp-idf/zh_CN/v5.1/esp32c6/api-reference/peripherals/adc_calibration.html这篇文章里提到的校准驱动器无法应用到esp32-C6上，esp32-C6只能获得ADC读到的数字而不能获得和数值对应的电压值。

woshiwstt50123, 2023-08-09:
、

closed 2023-08-09 by woshiwstt50123

Icarus113, 2023-08-09:
5.1 tag上还没有这个。 应该是5.1.1里会有。
现在你可以先用 https://github.com/espressif/esp-idf/blob/release/v5.1/components/esp_adc/esp32c6/include/adc_cali_schemes.h 这一版