espressif/esp-idf #12045: Add xtxtech spiflash driver support

yolee599, 2023-08-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[xtxtech](http://www.xtxtech.com)
[datasheet](http://www.xtxtech.com/download/?AId=443)
[0002-add-xtxtech-flash-driver-to-esp32s3.zip](https://github.com/espressif/esp-idf/files/12298781/0002-add-xtxtech-flash-driver-to-esp32s3.zip)

yolee599, 2023-08-09:
This patch file indicates compatibility with the Winbond spiflash driver

igrr, 2023-08-09:
Hi @yolee599, thank you for offering this patch. Could you please make a PR to https://github.com/espressif/esp-flash-drivers with this new driver? You can refer to https://github.com/espressif/esp-flash-drivers/pull/2 as an example of adding the driver.