espressif/esp-idf #12047: Corrupted variables at runtime -- is larger program file/memory configuration to blame?

SolidStateLEDLighting, 2023-08-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

NOTE: I have not updated my IDF branch in many months...

### What is the actual behavior?

Over the course of development I have seen and continue to see mysterious changes in variables which are subtle.   I have seen a boolean change in value, I have seen pointers become invalid.   And currently I am seeing a pointer to an RTOS queue change during run time - making that pointer unusable.   The system doesn't crash on this change, it just hangs operation on that thread.

This code area has been stable and running correctly for a while now.    So, I'm thinking this has something to do with the expanding size of the program file and how it is configured in memory.   I have studied all that I can and tested a variety of selections of memory configuration without any forward progress.

[image]

### Steps to reproduce.

The pointer to the destination queue changes value after it tries to xQueuePeek into it.   

PRE-xQueuePeek    systemCmdRequestsQue  1070468688
POST-xQueuePeek systemCmdRequestsQue  1070507836

Here is the actual line of code where this failure occurs:
if (xQueuePeek(camCmdRequestsQue, &(ptrCAMCmdRequest), pdMS_TO_TICKS(95)))
{
.....
}

### Debug Logs.

```plain
There are no debug logs because the system just fails to operation beyond that point.
```

### More Information.

I have been developing on the Esp32 for about 4 years in C/C++.   This project is fairy large and running on a Esp32s3 WROOM-1 N16R8.   There are 21 threads running.        heap_caps_check_integrity_all(true) is called on a regular basis and reveals nothing in this case.   The project has Wifi, AWS IOT Core, and a Camera driver.   Project is component based and builds in a predictable fashion.

name              priority    high water mark
ipc0                    9           1360
ipc1                    9           1352
wifi                     8           3980
esp_timer           7           3176
SYS_::TMR_        8           2228
SYS_::RUN_        7           3052
SYS_::GPIO         7           2332
IND_::RUN_       7           1228
WIFI::RUN_        7           2720
SNTP::RUN_      7           1044
RFID::RUN_       7           3212
PROX::RUN_      7           6236
IOT_::RUN_        7           2268
IOT_::AWS_        7           14616
CAM_::RUN_      7           15976
sys_evt              5           1564
tiT                     3           1212
main                 1           1652
IDLE0 and IDLE1 tasks are always priority 0.

Total sizes:
Used static IRAM:   80538 bytes ( 206182 remain, 28.1% used)
      .text size:   79511 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:   48905 bytes (  86263 remain, 36.2% used)
      .data size:   20841 bytes
      .bss  size:   28064 bytes
Used Flash size : 1274968 bytes
      .text     :  990123 bytes
      .rodata   :  284589 bytes
Total image size: 1376347 bytes (.bin may be padded larger)

ma89vik, 2023-08-18:
Hi @SolidStateLEDLighting,

> I just updated my IDF 4.4.1 to newest branch

By newest branch do you mean v4.4.5 or master?

Just from looking at the log here it is a bit hard to see what is wrong but here are some suggestions/questions:
1. Is the queue allocated on the heap?
2. Are you able to hook up a JTAG and watch the variable? Then you could at least find who/what is writing to it when it changes.
3. Is it possible to isolate the error, e.g. can you reproduce it by just running parts of the app (lets say without the wifi/network communication)?

ESP-Marius, 2023-08-18:
Hi @SolidStateLEDLighting,

I just updated my IDF 4.4.1 to newest branch

By newest branch do you mean v4.4.5 or master?

Just from looking at the log here it is a bit hard to see what is wrong but here are some suggestions/questions:

Is the queue allocated on the heap?
Are you able to hook up a JTAG and watch the variable? Then you could at least find who/what is writing to it when it changes.
Is it possible to isolate the error, e.g. can you reproduce it by just running parts of the app (lets say without the wifi/network communication)?

SolidStateLEDLighting, 2023-08-18:
Yes, the Queue is on the heap.    I only recently got my hardware setup for the JTAG debugger.   I don't know yet how to follow the variable, but I think that would be the next best step.     It is running on 4.4.5

SolidStateLEDLighting, 2023-08-21:
I have solved the problem.    I accidently changed the Queue size to accept the entire structure rather than just the size of the pointer.  When the queue received a message, that message over-wrote several of the pointers in that structure (as I was not expecting the entire structure).   The net result was that my pointers in that structure were strangely wrong.    My application now is running and has never been more stable.

closed 2023-08-21 by SolidStateLEDLighting