espressif/esp-idf #12048: i2c command link error

Kuralmozhi, 2023-08-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROOM32

### Power Supply used.

USB

### What is the expected behavior?

I am reading temperature and humidity from **_aht25_** sensor.  This is my main.c,
`#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_netif.h"
#include "esp_http_server.h"
#include "esp_netif.h"
#include "driver/i2c.h"

#include "aht25_library.h"

/* FreeRTOS event group to signal when we are connected */
static EventGroupHandle_t s_wifi_event_group;

static httpd_handle_t server = NULL;

#define EXAMPLE_ESP_WIFI_SSID      "Saranya's Galaxy F22"     // Replace with your WiFi SSID
#define EXAMPLE_ESP_WIFI_PASS      "saran142407" // Replace with your WiFi password
#define EXAMPLE_ESP_MAXIMUM_RETRY   5

#define I2C_PORT           I2C_NUM_0
#define I2C_SCL_PIN        22
#define I2C_SDA_PIN        21

/* The event group allows multiple bits for each event */
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

static const char *TAG = "wifi station";

static int s_retry_num = 0; // Add this line to declare s_retry_num

[... 294 lines cut ...]
esp_err_t aht25_read_temperature(i2c_port_t i2c_port, float *temperature);

// Function to read humidity from the AHT25 sensor
esp_err_t aht25_read_humidity(i2c_port_t i2c_port, float *humidity);

#ifdef __cplusplus
}
#endif

#endif /* AHT25_LIBRARY_H */
`

### What is the actual behavior?

it produce error,
[image]

I am new to ESP-IDF, Can anyone help me to solve this error?

### Steps to reproduce.

1. Step
2. Step
3. Step
...

mythbuster5, 2023-08-10:
You did not initialize the I2C peripheral, please refer to i2c example to use it.

mythbuster5, 2025-01-21:
Seems over one year no response, and this seems more like a usage issue not driver problem, I'm going to close it. Feel free to reopen if there is any more update. Thanks!

closed 2025-01-21 by mythbuster5