espressif/esp-idf #12049: docs.espressif.com certificate expired

jacob-firia, 2023-08-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The Let's Encrypt web certificate for docs.espressif.com has expired, leaving the site inaccessible in Google Chrome:

[image]
[image]

I can still access the site through Firefox, so it's not a total blocker, but it is broken.

kevinhikaruevans, 2023-08-09:
Is there an alternative docs URL for the time being?

jacob-firia, 2023-08-09:
Looks like the certificate has been updated. Thanks!