espressif/esp-idf #12050: Web Bluetooth API from Chrome on Macbook crashes on reconnect to bleprph example after successful bonding

rockys-spindance, 2023-08-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.3-270-g5eddb9e016

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

DevkitC with ESP32-WROVER-E module

### Power Supply used.

USB

### What is the expected behavior?

 A central peer should be able to successfully bond with a suitably configured `examples/bluetooth/nimble/bleprph` example running on a WROVER and then repeatedly disconnect and reconnect successfully an arbitrary number of times.

This expected behavior is working _correctly_ with all these centrals:
1. Web Bluetooth API via the  `chrome://bluetooth-internals` webpage from Chrome on Windows 11 laptop   [ Chrome Version 115.0.5790.110 (Official Build) (64-bit) ]
2. Android - Pixel 3 running Android 11
3. Android - Pixel 5 running Android 13
4. iOS- iPhone 11 Pro running iOS 14.2

### What is the actual behavior?

When testing specifically with the Web Bluetooth API via the  `chrome://bluetooth-internals` webpage in Chrome on MacBook, 
bonding and disconnection work fine, but reconnection often fails and often causes a nimBLE stack reset and crash.  

Versions:
- Chrome Version 115.0.5790.170 (Official Build) (arm64)
- macbook with Apple M1 pro chip running macOS Monterey v12.4  

The reconnect failure always shows the same pattern:

1. BLE_GAP_EVENT_CONNECT event is received and `ble_gap_security_initiate()` succeeds
2. There is ~20 second delay 
    - during this time all threads up to and including the `nimble-host` priority are apparently blocked 
4. BLE_GAP_EVENT_ENC_CHANGE is received with status 7 (BLE_HS_ENOTCONN)  and the new connection fails, often followed by a nimBLE stack reset and subsequent crash

### Steps to reproduce.

1. In the esp-idf repo, checkout the latest `release/v5.0` ( we started this testing at 5eddb9e016 ).
1. Navigate to `esp-idf/examples/bluetooth/nimble/bleprph`
1. Copy the `main.c` file from the attached *.zip to the `esp-idf/examples/bluetooth/nimble/bleprph/main` directory
1. Copy the `sdkconfig.defaults` file from the attached *.zip to the `esp-idf/examples/bluetooth/nimble/bleprph` directory
1. On ESP32, ensure that any previous bonding info has been removed.
1. On macbook, ensure that any previous bonding for the ESP32 has been removed
1. Build, flash, and run `bleprph` normally using the `idf.py` tools
1. On macbook, start Chrome browser and open a tab to `chrome://bluetooth-internals/`
1. Select "Devices" on that page and choose "Start Scan"
1. For the new entry for name "nimble-bleprph" with a live RSSI, choose "Inspect"
    - This should trigger a "connection request" dialog on the Mac - choose "connect".
    - You should see the connection logging on the esp32 console
[... 130 lines cut ...]

0x400dd73d: ble_hs_startup_reset_tx at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_startup.c:321

0x400dd8fb: ble_hs_startup_go at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_startup.c:331

0x400da7bc: ble_hs_sync at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:355

0x400da84a: ble_hs_reset at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:410

0x400da857: ble_hs_event_reset at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:562

0x40090ac5: npl_freertos_event_run at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:450

0x40083c59: ble_npl_event_run at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
 (inlined by) nimble_port_run at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:246

0x400d7296: bleprph_host_task at C:/ego/esp-idf/examples/bluetooth/nimble/bleprph/main/main.c:498

0x400957e1: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.0-2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

ELF file SHA256: 735899ba00c5049b

Rebooting...
ets Jul 29 2019 12:21:46
```

rockys-spindance, 2023-08-10:
Found [one issue here](https://github.com/espressif/esp-nimble/issues/46) with similar symptoms (in particular, the status 7) but it was on v4.4 and the fix was related to client-specific code that isn't relevant here.

rockys-spindance, 2023-08-10:
Tests with increased nimble logging (with the updated `main.c` and `sdkconfig.defaults` from the `bleprph-logging.zip` file attached here) seems to show the reconnect failure with 100% reproducibility at step 12. Still shows the status 7 encoding change, but the crash logging is slightly different than seen without the nimble logging.

The resulting console logging shows that there is NO nimble logging within the ~20 second "dead" period.

Here's a snippet of a failed reconnect with the logging code (see timestamp 95637 for the 20 second gap in logging):

> I (75517) NimBLE: connection established; status=0
> I (75517) NimBLE: handle=0 our_ota_addr_type=0 our_ota_addr=
> I (75517) NimBLE: e8:31:cd:e5:74:66
> I (75527) NimBLE:  our_id_addr_type=0 our_id_addr=
> I (75527) NimBLE: e8:31:cd:e5:74:66
> I (75537) NimBLE:  peer_ota_addr_type=0 peer_ota_addr=
> I (75537) NimBLE: bc:d0:74:54:ba:e9
> I (75547) NimBLE:  peer_id_addr_type=0 peer_id_addr=
> I (75547) NimBLE: bc:d0:74:54:ba:e9
> I (75557) NimBLE:  conn_itvl=12 conn_latency=0 supervision_timeout=36 encrypted=0 authenticated=0 bonded=0
> 
> I (75567) NimBLE: Boost nimble log levels
> D (75567) NimBLE: host tx hci data; handle=0 length=6
> 
> D (75577) NimBLE: ble_hs_hci_acl_tx():
> D (75577) NimBLE: 0x00
> D (75577) NimBLE: 0x00
> D (75587) NimBLE: 0x06
> D (75587) NimBLE: 0x00
> D (75587) NimBLE: 0x02
> D (75597) NimBLE: 0x00
> D (75597) NimBLE: 0x06
> D (75597) NimBLE: 0x00
> D (75597) NimBLE: 0x0b
> D (75607) NimBLE: 0x09
> D (75607) NimBLE:
> 
> I (75607) NimBLE: gap security initiate succeeded
> I (75617) NimBLE:
> 
> D (75617) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
> 
> D (75617) NimBLE: 0x16
> D (75627) NimBLE: 0x20
> D (75627) NimBLE: 0x02
> D (75627) NimBLE: 0x00
> D (75627) NimBLE: 0x00
> D (75637) NimBLE:
> 
> D (95637) NimBLE: looking up our sec;
> D (95637) NimBLE: peer_addr_type=0 peer_addr=
> D (95637) NimBLE: 0xe9
> D (95637) NimBLE: 0xba
> D (95637) NimBLE: 0x54
> D (95637) NimBLE: 0x74
> D (95637) NimBLE: 0xd0
> D (95647) NimBLE: 0xbc
> D (95647) NimBLE:
> D (95647) NimBLE: ediv=0x00 rand=0x0
> D (95647) NimBLE:
> 
> D (95657) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x001a len=18
> 
> D (95657) NimBLE: 0x1a
> D (95667) NimBLE: 0x20
> D (95667) NimBLE: 0x12
> D (95667) NimBLE: 0x00
> D (95667) NimBLE: 0x00
> D (95677) NimBLE: 0xe4
> D (95677) NimBLE: 0xcb
> D (95677) NimBLE: 0x67
> D (95677) NimBLE: 0x2b
> D (95687) NimBLE: 0xb2
> D (95687) NimBLE: 0x59
[... 59 lines cut ...]
> 0x400d7ddd: bleprph_advertise at C:/ego/esp-idf/examples/bluetooth/nimble/bleprph/main/main.c:186
> 
> 0x400d8037: bleprph_gap_event at C:/ego/esp-idf/examples/bluetooth/nimble/bleprph/main/main.c:290 (discriminator 15)
> 
> 0x400dfb6c: ble_gap_call_event_cb at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:756
> 
> 0x400e0cc5: ble_gap_conn_broken at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1314
> 
> 0x400e0d4a: ble_gap_reset_state at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1481
> 
> 0x400db872: ble_hs_reset at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:400
> 
> 0x400db89f: ble_hs_event_reset at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:562
> 
> 0x40090b35: npl_freertos_event_run at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:450
> 
> 0x40083c7d: ble_npl_event_run at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
>  (inlined by) nimble_port_run at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:246
> 
> 0x400d7cd6: bleprph_host_task at C:/ego/esp-idf/examples/bluetooth/nimble/bleprph/main/main.c:509
> 
> 0x40095851: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.0-2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
> 

[bleprph-logging.zip](https://github.com/espressif/esp-idf/files/12313961/bleprph-logging.zip)

rahult-github, 2023-08-11:
Hi @rockys-spindance , 

Tried with 2 macbook but issue is not seen. Reference log --> 
[macbook.txt](https://github.com/espressif/esp-idf/files/12321064/macbook.txt)

The setup has the same IDF version as mentioned in the original issue and also took the changes made in sdkconfig and main.c to replicate the same setup at our end.  The mac os though is 13.3 and 11.6 

I am not sure if the different Mac OS version would matter at this point of time. Though your logs denote crash on IDF side and that i have ensured is exactly the same.

To debug this further, is it ok if we try few binaries with different debugging comments to narrow down / root cause this issue ?  Please try attached patch ( to be applied at $IDF_PATH/components/bt/host/nimble path ) and share one normal console log.  Please do not enable Debug logging. You can use use default settings + your changes to enable some example features and the security initiation change done in bleprph example on connected event.  

[0001-Debug-prints.txt](https://github.com/espressif/esp-idf/files/12321103/0001-Debug-prints.txt)

rockys-spindance, 2023-08-11:
Thanks @rahult-github for taking a look at this.
I have captured a log for run where I applied your patch, 0001-Debug-prints.txt, from https://github.com/espressif/esp-idf/issues/12050#issuecomment-1674616148
I also moved to the latest esp-idf commit on the `release/v5.0` branch at baa90a6177 just to be testing with the very latest esp-idf.

The attached log shows

1. Successful connection and bonding at 41932
2. Successful disconnect at 50492
3. Successful reconnect starting at 56362 (followed by "encryption change event; status=0" at 56522)
4. Successful disconnect at 62572
5. Failed reconnect starting at 67412 (followed by "encryption change event; status=7" at 87472)

[issue-12050-log-0001-Debug-prints.log](https://github.com/espressif/esp-idf/files/12323614/issue-12050-log-0001-Debug-prints.log)

rockys-spindance, 2023-08-11:
I just tested on a different macbook / macOS / chrome version and was able to reproduce the reconnect failure

Macbook Pro 2.6 GHz 6-Core Intel Core i7
macOS Venture 13.1 (22C65)
Chrome Version 109.0.5414.119 (Official Build) (x86_64)

Same reconnect failure pattern with 20 second delay and then "encryption change event; status=7"

[1:18](https://spindance.slack.com/archives/D03F88NHC87/p1691774327142789)
Version 109.0.5414.119 (Official Build) (x86_64)

rahult-github, 2023-08-14:
Hi @rockys-spindance ,

Thanks for the logs. I see two issue in these logs. I have made a patch to address the assert (hopefully). Can you please try attached change and share back the logs. The patch needs to be applied on top of the previous debug patch ( in the same path) , so that if some issue is still observed, the same debug logs would be useful. In addition, you can also enable debug logging in nimble 

[0001-Increase-timeout-and-handle-null.txt](https://github.com/espressif/esp-idf/files/12331902/0001-Increase-timeout-and-handle-null.txt)

esp-cjh, 2023-08-15:
@rockys-spindance Could you please use the new controller lib to reproduce this problem and provide me the log.
Copy the `libbtdm_app.a` from the attached file `libbtdm_app.zip` and replace `esp-idf/components/bt/controller/lib_esp32/esp32/libbtdm_app.a`
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/12341950/libbtdm_app.zip)

rockys-spindance, 2023-08-15:
Thanks for the support @esp-cjh !
I have reproduced the MAC web bluetooth reconnect problem using the updated `libbtdm_app.a` you provided in https://github.com/espressif/esp-idf/issues/12050#issuecomment-1678511314.

I added your library to the code I tested for @rahult-github yesterday, so it still has his host-side print statements in place. I can see additional logging from your library in the console.

The attached log shows
1. Successful bonding connection at 24137
2. Successful disconnect at 36727
3. Reconnect attempt at 45637 showing the failure with the subsequent "encryption change event; status=7" at 70857, 25 seconds after the connect event.

[issue-12050-libbtdm_app-15-aug-23.log](https://github.com/espressif/esp-idf/files/12344677/issue-12050-libbtdm_app-15-aug-23.log)

esp-cjh, 2023-08-16:
@rockys-spindance Please use this controller lib [6be0483] to reproduce, thanks.
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/12359446/libbtdm_app.zip)

rockys-spindance, 2023-08-17:
Thanks @esp-cjh 
Here is a log showing the reconnect failure using your latest `libbtdm_app.a` file from https://github.com/espressif/esp-idf/issues/12050#issuecomment-1680521756

[issue-12050-libbtdm_app-16-aug-23.log](https://github.com/espressif/esp-idf/files/12364544/issue-12050-libbtdm_app-16-aug-23.log)

esp-cjh, 2023-08-17:
@rockys-spindance @rahult-github 
I have find out the root cause. The macbook start encryption after connected. At the same time the nimble host send read remote feature command. But at this time the controller can't process this command. So nimble host wait for the ack from controller. And then controller send LL_LTK_request event to host but host can't process this event. After HCI timeout, nimble host can process this event and controller can continue encrypt procedure.

rahult-github, 2023-08-17:
Hi @rockys-spindance , 

One suggestion to avoid the race condition for both sides starting the encryption process. Can you try without the application doing the below operation in connect complete event and see if it works ? 

`            
             int rc = ble_gap_security_initiate(event->connect.conn_handle);
             if (rc != 0) {
                 MODLOG_DFLT(ERROR, "ble_gap_security_inititate failed: %d", rc);
            } else {
                 MODLOG_DFLT(INFO, "gap security initiate succeeded");
            }
        }

`

rockys-spindance, 2023-08-17:
Thanks @esp-cjh  and @rahult-github for the latest info.
I follow that unlike the other centrals, the web bluetooth API on mac is initiating the security procedure and having both sides do so is causing the problem.

Per @rahult-github's [last request](https://github.com/espressif/esp-idf/issues/12050#issuecomment-1681915522), I tested without calling `ble_gap_security_initiate` when receiving the `BLE_GAP_EVENT_CONNECT`.  And I can confirm that without initiating encryption from the peripheral, I am able to reconnect to the Mac an unlimited number of times.

BUT I also don't get bonding with the Mac (there is never a dialog asking for permission to connect and the ESP32 is not listed in the mac's BLE devices). 

@rahult-github Were you proposing that change as just a test to verify @esp-cjh's theory?  Or were you proposing it as a fix?  I don't see how it works as a fix since it breaks operation on all _other_ centrals we're testing besides the Mac, and breaks bonding on the Mac as well.

rahult-github, 2023-08-17:
Hi @rockys-spindance ,

  > Or were you proposing it as a fix?
  
  No, the change was a suggestion to try and verify that indeed this is the case and it is the remote device behaviour resulting in this situation.

rockys-spindance, 2023-08-17:
@rahult-github Thanks for clarifying.

esp-cjh, 2023-08-17:
@rockys-spindance @rahult-github 
This is a bug of controller on ESP32. I will fixed it ASAP.

esp-cjh, 2023-08-17:
@rockys-spindance 
Please use this lib [226e394] to see if it can solve the problem, thanks.
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/12369539/libbtdm_app.zip)

rockys-spindance, 2023-08-17:
@esp-cjh 
I tested with [your latest library from here ](https://github.com/espressif/esp-idf/issues/12050#issuecomment-1682243781)and it _does_ fix the reconnect problem for web bluetooth on my macbook!

More specifically, I retested the reconnect failure case using
- Web Bluetooth API via the chrome://bluetooth-internals webpage from Chrome on macbook with Apple M1 pro chip [ Chrome Version 115.0.5790.170 (Official Build) (arm64)) ]

and that case is now working correctly.

I also did a quick regression test of the other centrals that were working originally and all of these continue to work correctly with your new library:
- Web Bluetooth API via the chrome://bluetooth-internals webpage from Chrome on Windows 11 laptop [ Chrome Version 115.0.5790.110 (Official Build) (64-bit) ]
- Android - Pixel 3 running Android 11
- Android - Pixel 5 running Android 13
- iOS- iPhone 11 Pro running iOS 14.2

esp-cjh, 2023-08-18:
@rockys-spindance 
This lib fixed the problem and without debug log. You can use it for now and I will submit the fix ASAP.
[Uploading libbtdm_app.zip…]()

rockys-spindance, 2023-08-31:
@esp-cjh Do you happen to know if your fix from a few weeks ago has been merged into the release/v5 branch yet?  And if so, do you have that commit?  Whenever I look at the git log for merges that change the bt controller the comments don't identify this issue so I'm never sure.

rockys-spindance, 2023-11-10:
Hi @esp-cjh and @rahult-github , any update on whether or when this fix will get merged into a release/v5 branch yet?

zhp0406, 2023-11-17:
@rockys-spindance 
This change is quite significant and requires rigorous testing. It is currently in the testing process. Please continue to use the provided library for now. If there are any issues, please contact me.

rockys-spindance, 2023-11-17:
@zhp0406 Thank you for the update.

rockys-spindance, 2024-01-29:
Hi @zhp0406 Just checking in to see if this fix has been merged in a a release yet. My client is getting close to production and ideally we'd like to avoid using a patch library in the final production firmware.

esp-cjh, 2024-03-07:
Hi @rockys-spindance The fix has been merged to master branch. See 7df536d3255d9c7c61b5dcae1f1019bd276da666. I will backport to release branch ASAP.

Indastri, 2024-07-24:
Closing the issue, if you will encounter any more problems feel free to reopen this one or create a new issue.

Cheers!

closed 2024-07-24 by Indastri