espressif/esp-idf #12053: adc_continuous_read returning ESP_ERR_TIMEOUT

puyoulu, 2023-08-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3-wroom-1

### Power Supply used.

External 5V

### What is the expected behavior?

adc_continuous_read is quite easy (80% possibility) to return ESP_ERR_TIMEOUT

### What is the actual behavior?

adc_continuous_read is quite easy (80% possibility) to always return ESP_ERR_TIMEOUT (conv_frame_size set to 512), after power on reset, and s_adc_dma_intr is then never being called.
This occurs when running v5.0.3, but never occurs on v5.0.2.

I checked the history, seems this bug was introduced by commit 5da5e1801b and d8ee45ce08.
Wish you guys can confirm this.

I tried below patch, and seems can fix:
```
diff --git a/components/esp_adc/adc_continuous.c b/components/esp_adc/adc_continuous.c
index 643dfef..ed1f760 100644
--- a/components/esp_adc/adc_continuous.c
+++ b/components/esp_adc/adc_continuous.c
@@ -355,6 +355,11 @@ static IRAM_ATTR bool s_adc_dma_intr(adc_continuous_ctx_t *adc_digi_ctx)
         }
     }
 
+    if (status == ADC_HAL_DMA_DESC_NULL) {
+        //start next turns of dma operation
+        adc_hal_digi_start(&adc_digi_ctx->hal, adc_digi_ctx->rx_dma_buf);
+    }
+
     return need_yield;
 }
 
diff --git a/components/hal/adc_hal.c b/components/hal/adc_hal.c
index 3633816..2051ce6 100644
--- a/components/hal/adc_hal.c
+++ b/components/hal/adc_hal.c
@@ -238,7 +238,6 @@ static void adc_hal_digi_dma_link_descriptors(dma_descriptor_t *desc, uint8_t *d
     HAL_ASSERT(((uint32_t)data_buf % 4) == 0);
     HAL_ASSERT((per_eof_size % 4) == 0);
     uint32_t n = 0;
-    dma_descriptor_t *desc_head = desc;
 
     while (eof_num--) {
         uint32_t eof_size = per_eof_size;
@@ -262,7 +261,7 @@ static void adc_hal_digi_dma_link_descriptors(dma_descriptor_t *desc, uint8_t *d
             n++;
         }
     }
-    desc[n-1].next = desc_head;
+    desc[n-1].next = NULL;
 }
 
 void adc_hal_digi_start(adc_hal_dma_ctx_t *hal, uint8_t *data_buf)
```

### Steps to reproduce.

running continuous_read_main.c

puyoulu, 2023-08-12:
I re-checked the code and did some testings, and found out it's because the dma interrupt stops generating during heavy load. I created a PR, it works fine for me, atleat now. Could you help to review it?

LHYYSY, 2023-09-25:
Hello, I have the same problem, what is the cause of this problem?

Icarus113, 2023-10-08:
I re-checked the code and did some testings, and found out it's because the dma interrupt stops generating during heavy load. I created a PR, it works fine for me, atleat now. Could you help to review it?

A [PR](https://github.com/espressif/esp-idf/pull/11411) with similar changes have been accepted before. Sorry for the late reply.

> Hello, I have the same problem, what is the cause of this problem?

You may upgrade to use a newer version.

puyoulu, 2023-11-16:
> A [PR](https://github.com/espressif/esp-idf/pull/11411) with similar changes have been accepted before. Sorry for the late reply.

You did not ever read the [PR,](https://github.com/espressif/esp-idf/pull/12065) right? This PR is based on the one you referenced.

Issue can still be reproduced by applying below patch to idf v5.0.4:
```
--- a/examples/peripherals/adc/continuous_read/main/continuous_read_main.c
+++ b/examples/peripherals/adc/continuous_read/main/continuous_read_main.c
@@ -12,6 +12,7 @@
 #include "freertos/task.h"
 #include "freertos/semphr.h"
 #include "esp_adc/adc_continuous.h"
+#include "nvs_flash.h"
 
 #define EXAMPLE_READ_LEN   256
 #define EXAMPLE_ADC_CONV_MODE           ADC_CONV_SINGLE_UNIT_1
@@ -111,6 +112,14 @@ void app_main(void)
     ESP_ERROR_CHECK(adc_continuous_register_event_callbacks(handle, &cbs, NULL));
     ESP_ERROR_CHECK(adc_continuous_start(handle));
 
+    vTaskDelay(100);
+    ret = nvs_flash_init();
+    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
+      ESP_ERROR_CHECK(nvs_flash_erase());
+      ret = nvs_flash_init();
+    }
+    nvs_flash_erase_partition("nvs_1");
+
     while(1) {
 
         /**
```

nvs_1 is something like this in partiton.csv:
`nvs_1,   data, nvs,             0x1f8000,      32k,`

closed 2023-12-07 by puyoulu

ZXTube, 2024-05-23:
If the PR was merged then why do I still have this problem?

I have esp-idf v5.1.2.
For me this problem appeared without me changing anything in my code.
I have a spiffs file and in a while loop I read an adc chunk and write it to the file.
I ran this code plenty of times, only changing code that happens after the while loop.
And one of the times I ran the code and got this error.
After that I consistently get this error every time I run the code.
And the thing that fixed it was _either_ **not writing to the file** _or_ **reformatting the spiffs partition once and then it works every time**
The partition was mostly empty when the error happened so I don't think it was caused by it being full.

Any ideas why this is happening.

Thanks in advance