espressif/esp-idf #12057: ESP32C3不支持nimble下使用ble mesh吗？

zksite, 2023-08-11:
### Is your feature request related to a problem?

我导入nimble下的blemesh项目，IDF_TARGET设置为esp32c3可以正常编译并且成功运行，而且日志也打印成功初始化ble mesh，但是没有广播配网相关的包。

### Describe the solution you'd like.

esp32c3如何才能成功使用nimble运行ble mesh，如果框架本身支持，我要如何调用nimble的api才能跑通整个ble mesh的基本配网等相关流程，谢谢。

### Describe alternatives you've considered.

之前使用esp ble mesh的蓝牙堆栈，但是当我需要和wifi结合在一起时，发现整个固件太大了，使用nimble可以缩减几百K的体积。

forx157, 2023-08-11:
@zksite hello,
You can use BLE Mesh on the Nimble host. Regarding the issue of not having a provisioning packet, please change the print level of the BLE Mesh stack to "info" level. The specific setup path is as follows: `(Top) → Component config → ESP BLE Mesh Support → BLE Mesh STACK DEBUG LOG LEVEL → BLE_MESH_STACK`. Please share your log file and sdkconfig.h file(path: build/config/) with us .

zksite, 2023-08-11:
我不是使用esp_ble_mesh，我的芯片是esp32-c3，示例里有说明nimble blemesh项目只支持esp32，是esp32-c3不支持nimble ble mesh的原因吗？
以下是启动日志：
I (24) boot: ESP-IDF v5.0.3-dirty 2nd stage bootloader
I (24) boot: compile time 12:42:53
I (25) boot: chip revision: v0.4
I (27) boot.esp32c3: SPI Speed      : 80MHz
I (32) boot.esp32c3: SPI Mode       : DIO
I (36) boot.esp32c3: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=1ca90h (117392) map
I (111) esp_image: segment 1: paddr=0002cab8 vaddr=3fc92200 size=021f4h (  8692) load
I (113) esp_image: segment 2: paddr=0002ecb4 vaddr=40380000 size=01364h (  4964) load
I (118) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=63b14h (408340) map
I (190) esp_image: segment 4: paddr=00093b3c vaddr=40381364 size=10d38h ( 68920) load
I (209) boot: Loaded app from partition at offset 0x10000
I (209) boot: Disabling RNG early entropy source...
I (220) cpu_start: Unicore app
I (220) cpu_start: Pro cpu up.
I (229) cpu_start: Pro cpu start user code
I (230) cpu_start: cpu freq: 160000000 Hz
I (230) cpu_start: Application information:
I (233) cpu_start: Project name:     blemesh
I (237) cpu_start: App version:      1
I (242) cpu_start: Compile time:     Aug 11 2023 12:42:42
I (248) cpu_start: ELF file SHA256:  a7c15b35a896c764...
I (254) cpu_start: ESP-IDF:          v5.0.3-dirty
I (259) cpu_start: Min chip rev:     v0.3
I (264) cpu_start: Max chip rev:     v0.99
I (269) cpu_start: Chip rev:         v0.4
I (274) heap_init: Initializing. RAM available for dynamic allocation:
I (281) heap_init: At 3FC98650 len 000440C0 (272 KiB): DRAM
I (287) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (294) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (301) spi_flash: detected chip: generic
I (305) spi_flash: flash io: dio
W (309) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (322) coexist: coexist rom version 9387209
I (327) cpu_start: Starting scheduler.
I (336) BLE_INIT: BT controller compile version [85b425c]
I (336) phy_init: phy_version 970,1856f88,May 10 2023,17:44:12
I (376) system_api: Base MAC address is not set
I (376) system_api: read default base MAC address from EFUSE
I (376) BLE_INIT: Bluetooth MAC: 70:04:1d:1f:30:16

I (386) NimBLE: GAP procedure initiated: stop advertising.

I (396) NimBLE_MESH: Bluetooth initialized

I (436) NimBLE_MESH: Mesh initialized

**sdkconfig:**
/*
 * Automatically generated file. DO NOT EDIT.
 * Espressif IoT Development Framework (ESP-IDF) Configuration Header
 */
#pragma once
#define CONFIG_SOC_ADC_SUPPORTED 1
#define CONFIG_SOC_DEDICATED_GPIO_SUPPORTED 1
#define CONFIG_SOC_GDMA_SUPPORTED 1
#define CONFIG_SOC_TWAI_SUPPORTED 1
#define CONFIG_SOC_BT_SUPPORTED 1
#define CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED 1
#define CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED 1
#define CONFIG_SOC_TEMP_SENSOR_SUPPORTED 1
[... 773 lines cut ...]
#define CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT CONFIG_VFS_SUPPRESS_SELECT_DEBUG_OUTPUT
#define CONFIG_SW_COEXIST_ENABLE CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE
#define CONFIG_SYSTEM_EVENT_QUEUE_SIZE CONFIG_ESP_SYSTEM_EVENT_QUEUE_SIZE
#define CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE
#define CONFIG_TASK_WDT CONFIG_ESP_TASK_WDT_INIT
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0
#define CONFIG_TASK_WDT_TIMEOUT_S CONFIG_ESP_TASK_WDT_TIMEOUT_S
#define CONFIG_TCPIP_RECVMBOX_SIZE CONFIG_LWIP_TCPIP_RECVMBOX_SIZE
#define CONFIG_TCPIP_TASK_AFFINITY CONFIG_LWIP_TCPIP_TASK_AFFINITY
#define CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY CONFIG_LWIP_TCPIP_TASK_AFFINITY_NO_AFFINITY
#define CONFIG_TCPIP_TASK_STACK_SIZE CONFIG_LWIP_TCPIP_TASK_STACK_SIZE
#define CONFIG_TCP_MAXRTX CONFIG_LWIP_TCP_MAXRTX
#define CONFIG_TCP_MSL CONFIG_LWIP_TCP_MSL
#define CONFIG_TCP_MSS CONFIG_LWIP_TCP_MSS
#define CONFIG_TCP_OVERSIZE_MSS CONFIG_LWIP_TCP_OVERSIZE_MSS
#define CONFIG_TCP_QUEUE_OOSEQ CONFIG_LWIP_TCP_QUEUE_OOSEQ
#define CONFIG_TCP_RECVMBOX_SIZE CONFIG_LWIP_TCP_RECVMBOX_SIZE
#define CONFIG_TCP_SND_BUF_DEFAULT CONFIG_LWIP_TCP_SND_BUF_DEFAULT
#define CONFIG_TCP_SYNMAXRTX CONFIG_LWIP_TCP_SYNMAXRTX
#define CONFIG_TCP_WND_DEFAULT CONFIG_LWIP_TCP_WND_DEFAULT
#define CONFIG_TIMER_QUEUE_LENGTH CONFIG_FREERTOS_TIMER_QUEUE_LENGTH
#define CONFIG_TIMER_TASK_PRIORITY CONFIG_FREERTOS_TIMER_TASK_PRIORITY
#define CONFIG_TIMER_TASK_STACK_DEPTH CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH
#define CONFIG_TIMER_TASK_STACK_SIZE CONFIG_ESP_TIMER_TASK_STACK_SIZE
#define CONFIG_UDP_RECVMBOX_SIZE CONFIG_LWIP_UDP_RECVMBOX_SIZE

zksite, 2023-08-11:
@forx157 麻烦帮忙看看什么原因，谢谢

rahult-github, 2023-08-16:
Hi @zksite 

> I am not using esp_ble_mesh

As you are aware , IDF clones upstream mynewt nimble fork and reuses it. This is the reason, the files from upstream code are all available in the IDF as it is.  Espressif doesn't maintain the upstream blemesh implementation. Instead the esp-ble-mesh is the official mesh solution in IDF. 

>  This example is part of upstream mynewt-nimble implementation and not maintained by Espressif.
>
> IDF maintains ESP-BLE-MESH as the official Mesh solution. Please refer to ESP-BLE-MESH guide at: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/esp-ble-mesh/ble-mesh-index.html

Hence, no further support is planned currently for any chip for the nimble's blemesh application.

closed 2024-04-22 by zksite