espressif/esp-idf #12058: W5500 ETH - Can't recover from `errno 113`

Flamabalistic, 2023-08-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

Reconnect after a brief outage

### What is the actual behavior?

- Permanently becomes stuck in an errored state, repeatedly joining and leaving the network (as seen with the `Ethernet link down` and `Ethernet link up` msgs)
- Unplugging & replugging the ethernet cable doesn't fix.
- Rebooting the ESP temporarily fixes it, erroring again after a similar time frame.

### Steps to reproduce.

1. Connect a W5500 up to an ESP32-S3-DevKitC-1 using the pinout as defined in w5500_eth_wrapper.cpp (We are using this w5500-EVB-Pico with no code running on the pico, however the w5500 chip should yield the same results) - W5500 is connected to an external 5V power supply
2. Run the given code
3. Wait ~10-30mins

This error occurs at around the same time regardless of if a client is connected or not.

For completeness, I've attached a zip of the MVP which still encounters the problem.
[EthIssueMVP.zip](https://github.com/espressif/esp-idf/files/12318675/EthIssueMVP.zip)

### Debug Logs.

```plain
-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (27) boot: ESP-IDF v5.1 2nd stage bootloader
I (27) boot: compile time Aug 11 2023 14:43:14
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
[... 287 lines cut ...]
D (426924) W5500 eth: Ethernet HW Addr 34:85:18:4b:1a:83
D (448814) w5500.mac: link is down
D (448814) event: running post ETH_EVENT:3 with handler 0x4202784c and context 0x3fcabf0c on loop 0x3fca35d8
D (448814) esp_eth.netif.netif_glue: eth_action_disconnected: 0x3fcabe24, 0x3c050230, 3, 0x3fcad1d8, 0x3fcabdac
D (448824) esp_netif_handlers: esp_netif action disconnected with netif0x3fca6bb4 from event_id=3
D (448834) esp_netif_lwip: check: remote, if=0x3fca6bb4 fn=0x42025d9c

D (448844) esp_netif_lwip: esp_netif_down_api esp_netif:0x3fca6bb4
D (448844) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3fca6bb4
D (448854) esp_netif_lwip: if0x3fca6bb4 start ip lost tmr: already started
D (448864) esp_netif_lwip: check: local, if=0x3fca6bb4 fn=0x42026320
```
```

### More Information.

I've also tried this code using TCP over WiFi rather than Ethernet. The same `errno 113` occurs, however it quickly recovers and resumes normal function. This makes me believe this is a problem specifically either with the eth modules, or with the w5500 module specifically.

This code was originally in a much larger code base, however even the trimmed down version is still encountering this issue at approximately the same rate.

We're still developing this device using a breadboard, so there may be some issues there with the SPI clock and signal delays etc., but I don't know enough about it to know if it's an issue.

Please let me know if you have any other questions!
Regards,
Xavier

kostaond, 2023-08-18:
Hi Xavier,

> We are using this w5500-EVB-Pico with no code running on the pico

How have you ensured there is no code running on the pico?

Flamabalistic, 2023-08-20:
I uploaded the following blank to make sure the Pico wasn't interfering (before this, I couldn't connect to the W5500 at all)
```void setup() {}

void loop() {
  delay(1000);
}```

kostaond, 2023-08-21:
I'm not able to reproduce. I ran your code for more than 6 hours without any issue. I used ESP32-S3-WROOM-1 with W5500 module like [this](https://www.aliexpress.com/item/1005004088381659.html?&_t=pvid:748960a8-cae8-4375-a804-bc8c4362d252&spm=a2g0o.ppclist.product.mainProduct). I used Netcat as a TCP client.

Are there any other specific steps to reproduce the issue? If not, I would recommend to use the W5500 module I mentioned above. It's cheap so it should not be a problem to give it a try.

Flamabalistic, 2023-08-21:
`We're still developing this device using a breadboard, so there may be some issues there with the SPI clock and signal delays etc., but I don't know enough about it to know if it's an issue.`

After some more tinkering, I think this (or another breadboard related problem) might be the issue. 

We're getting some PCB's made in a couple of weeks, if this is still an issue with those, I'll reopen the issue, otherwise I think this might just be a problem with our setup.

closed 2023-08-21 by Flamabalistic

kostaond, 2023-08-22:
@Flamabalistic thank you. If you observe the issue even with the new PCB don't hesitate to reopen.