espressif/esp-idf #12059: wifi_provisioning配网成功后，再次进入配网失败

lintertion, 2023-08-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

芯片：ESP32S3
IDF版本：release/v5.1 和tags的v5.1
具体表现：第一次配网正常，再次进入配网的时候，在esp_bt_controller_init这个初始化，会有调用semphr_delete_wrapper导致panic'ed (LoadProhibited)。打印了semphr_delete_wrapper里面semphr指针地址是0x7274632d。
崩溃日志：
MAGIC fadebead VERSION 00010006
I (15:22:34.815) BLE_INIT: BT controller compile version [85b425c]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40381c6f  PS      : 0x00060730  A0      : 0x82078f70  A1      : 0x3fcbea50
0x40381c6f: vQueueDelete at /Users/linter/esp/0808/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:2319

A2      : 0xf96c3fce  A3      : 0x00000000  A4      : 0x3fcf7fb8  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80381898  A9      : 0x3fcbea30
A10     : 0x3fca0c30  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x6001cc48  A15     : 0x00060923  SAR     : 0x00000009  EXCCAUSE: 0x0000001c
EXCVADDR: 0xf96c4012  LBEG    : 0x4037e60c  LEND    : 0x4037e618  LCOUNT  : 0x00000000
0x4037e60c: ram_wifi_tx_dig_gain at ??:?

0x4037e618: ram_wifi_tx_dig_gain at ??:?

Backtrace: 0x40381c6c:0x3fcbea50 0x42078f6d:0x3fcbea70 0x420b638d:0x3fcbea90 0x420b7178:0x3fcbeab0 0x42079519:0x3fcbeb00 0x42078c7a:0x3fcbeb40 0x420787e9:0x3fcbebc0 0x420750a8:0x3fcbec00 0x4207464e:0x3fcbec20 0x42035df9:0x3fcbede0 0x4200e4ae:0x3fcbef10 0x4201edb3:0x3fcbef30 0x42037301:0x3fcbef50 0x4203739d:0x3fcbef70 0x42036d9d:0x3fcbefb0 0x420498c8:0x3fcbefd0 0x4204bcbc:0x3fcbeff0 0x42017e42:0x3fcbf020 0x403838ca:0x3fcbf070
0x40381c6c: vQueueDelete at /Users/linter/esp/0808/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:2297

0x42078f6d: semphr_delete_wrapper at /Users/linter/esp/0808/esp-idf/components/bt/controller/esp32c3/bt.c:529

0x420b638d: BTM_BleGetEnergyInfo at ??:?

0x420b7178: btdm_controller_init at ??:?

0x42079519: esp_bt_controller_init at /Users/linter/esp/0808/esp-idf/components/bt/controller/esp32c3/bt.c:1305

0x42078c7a: simple_ble_start at /Users/linter/esp/0808/esp-idf/components/protocomm/src/simple_ble/simple_ble.c:217

0x420787e9: protocomm_ble_start at /Users/linter/esp/0808/esp-idf/components/protocomm/src/transports/protocomm_ble.c:596

0x420750a8: prov_start at /Users/linter/esp/0808/esp-idf/components/wifi_provisioning/src/scheme_ble.c:53

0x4207464e: wifi_prov_mgr_start_service at /Users/linter/esp/0808/esp-idf/components/wifi_provisioning/src/manager.c:300
 (inlined by) wifi_prov_mgr_start_provisioning at /Users/linter/esp/0808/esp-idf/components/wifi_provisioning/src/manager.c:1649

rahult-github, 2023-08-14:
Hi @lintertion  , by default the wifi provision application would do provisioning over BLE , share the wifi credentials to get netwrok information and once the provisioning is over, then the BLE stack will be teared down. 

For your use case , can you please share more information , basically:
1. Any application change done to redo provisioning ?
2. Any configuration modified apart from default configuration ? You can share your sdkconfig.
3. Any steps or way to use the existing application to reproduce the issue you observed above.

lintertion, 2023-08-14:
yes, It seams it is the BLE stack teared down problem. after provisioning i do the udp/tcp communication through wifi ,and no BLE action at all. then enterting provision again, BLE stack init again,  but in esp_bt_controller_init ,it call semphr_delete_wrapper ,this is strange action,  it seems there is some pointer in the BLE stack release after provisioning and not set to NULL. and when we call  esp_bt_controller_init again, it release again and abort ?

if the log above not enough, i can use the provisiong example  in idf to reproduce it .

lintertion, 2023-08-18:
Hi, i use BLE for wifi_provisioning,  and i want to release memory for the other use, and if user want to change wifi again, init BLE again for new provisioning.  And there is a IDF version v5.1.0, i check out last year from master branch at that time. it  works ok, i can reinit BLE stack again for a new provision. But for production.i want to change IDF to release version V5.1, then this problem appeared.

rahult-github, 2023-08-18:
Hi @lintertion ,

 Actually the change to release memory after stack is turned OFF is present since the first version of wifi provisioning and if memory is released, then it is not possible to reinit BLE.  But, as you have said it was working for you , can you please share any commit id that was working and the actual changes done at your end for this usecase ?  I can try the same at my end and should be able to debug more.

rahult-github, 2023-09-04:
Hi @lintertion , any updates ?

rahult-github, 2024-02-14:
Closing issue. Please feel free to reopen with further details

closed 2024-02-14 by rahult-github