espressif/esp-idf #12061: BLE GATT CLIENT indicate  read write have different characteristics

aswathreddy, 2023-08-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have two different characteristics uuid for write and notify I need to change the characteristics handle in between I am new to this can someone help with code snippet

zhp0406, 2023-08-15:
@aswathreddy 
You can take a look at the following two examples:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server_service_table
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_client
If you have any further questions, please provide a more detailed description.

zhp0406, 2023-10-16:
@aswathreddy 
Hi,have your problem solved?

closed 2024-03-10 by Alvin1Zhang