espressif/esp-idf #12066: mbedtls_net_connect returned -0x52

shmelkin, 2023-08-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1 stable

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S2 WROVER

### Power Supply used.

USB

### What is the expected behavior?

Sending an Email via my working SMTP server

### What is the actual behavior?

E (8505) smtp_example: mbedtls_net_connect returned -0x52

### Steps to reproduce.

1. Create smtp_client example project
3. Fill in SMTP configuration
4. Fill in WiFI configuration
5. Flash on ESP32 S2 WROVER Dev-Kit

### Debug Logs.

```plain
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x17c8
load:0x4004b000,len:0x4
load:0x4004b004,len:0xab4
load:0x4004f000,len:0x3160
entry 0x4004b1c4
I (26) boot: ESP-IDF v5.1-dirty 2nd stage bootloader
I (26) boot: compile time Aug 12 2023 17:13:42
I (26) boot: chip revision: v0.0
I (30) boot.esp32s2: SPI Speed      : 80MHz
I (35) boot.esp32s2: SPI Mode       : DIO
I (39) boot.esp32s2: SPI Flash Size : 2MB
I (44) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=2eca4h (191652) map
I (134) esp_image: segment 1: paddr=0003eccc vaddr=3ffc7c20 size=0134ch (  4940) load
I (135) esp_image: segment 2: paddr=00040020 vaddr=40080020 size=8b2b0h (570032) map
I (253) esp_image: segment 3: paddr=000cb2d8 vaddr=3ffc8f6c size=01e98h (  7832) load
I (255) esp_image: segment 4: paddr=000cd178 vaddr=40024000 size=13c18h ( 80920) load
I (289) boot: Loaded app from partition at offset 0x10000
I (289) boot: Disabling RNG early entropy source...
I (301) cpu_start: Unicore app
I (301) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (303) cache: Data cache               : size 8KB, 4Ways, cache line size 32Byte
I (311) cpu_start: Pro cpu up.
I (326) cpu_start: Pro cpu start user code
I (326) cpu_start: cpu freq: 160000000 Hz
I (326) cpu_start: Application information:
I (331) cpu_start: Project name:     smtp_client
I (336) cpu_start: App version:      1
I (340) cpu_start: Compile time:     Aug 12 2023 17:27:47
I (346) cpu_start: ELF file SHA256:  15c0b29e3352de0e...
I (352) cpu_start: ESP-IDF:          v5.1-dirty
I (358) cpu_start: Min chip rev:     v0.0
I (362) cpu_start: Max chip rev:     v1.99 
I (367) cpu_start: Chip rev:         v0.0
I (372) heap_init: Initializing. RAM available for dynamic allocation:
I (379) heap_init: At 3FFCF3C0 len 0002CC40 (179 KiB): DRAM
I (385) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (392) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (399) spi_flash: detected chip: generic
I (402) spi_flash: flash io: dio
W (406) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (424) app_start: Starting scheduler on CPU0
I (425) main_task: Started on CPU0
I (425) main_task: Calling app_main()
I (435) example_connect: Start example_connect.
I (445) wifi:wifi driver task: 3ffd758c, prio:23, stack:6656, core=0
I (445) wifi:wifi firmware version: b2f1f86
I (445) wifi:wifi certification version: v7.0
I (445) wifi:config NVS flash: enabled
I (455) wifi:config nano formating: disabled
I (455) wifi:Init data frame dynamic rx buffer num: 32
I (465) wifi:Init management frame dynamic rx buffer num: 32
I (465) wifi:Init management short buffer num: 32
I (465) wifi:Init dynamic tx buffer num: 32
I (475) wifi:Init static rx buffer size: 1600
I (475) wifi:Init static rx buffer num: 10
I (485) wifi:Init dynamic rx buffer num: 32
I (485) wifi_init: rx ba win: 6
I (485) wifi_init: tcpip mbox: 32
I (495) wifi_init: udp mbox: 6
I (495) wifi_init: tcp mbox: 6
I (505) wifi_init: tcp tx win: 5744
I (505) wifi_init: tcp rx win: 5744
I (505) wifi_init: tcp mss: 1440
I (515) wifi_init: WiFi IRAM OP enabled
I (515) wifi_init: WiFi RX IRAM OP enabled
I (525) phy_init: phy_version 2300,d67cf06,Feb 10 2022,10:03:07
I (565) wifi:mode : sta (7c:df:a1:06:99:f6)
I (565) wifi:enable tsf
I (565) example_connect: Connecting to In einem LAN vor unserer Zeit2.4652fluffyflamingo...
I (575) example_connect: Waiting for IP(s)
I (2995) wifi:new:<7,1>, old:<1,0>, ap:<255,255>, sta:<7,1>, prof:1
I (2995) wifi:state: init -> auth (b0)
I (4615) wifi:state: auth -> assoc (0)
E (4625) wifi:Association refused temporarily, comeback time 1024 mSec
I (5645) wifi:state: assoc -> assoc (0)
I (5655) wifi:state: assoc -> run (10)
I (5695) wifi:connected with In einem LAN vor unserer Zeit2.4, aid = 129, channel 7, 40U, bssid = 1c:ed:6f:42:d1:a5
I (5695) wifi:security: WPA3-SAE, phy: bgn, rssi: -65
I (5705) wifi:pm start, type: 1

I (5715) wifi:<ba-add>idx:0 (ifx:0, 1c:ed:6f:42:d1:a5), tid:0, ssn:0, winSize:64
I (5725) wifi:<ba-add>idx:1 (ifx:0, 1c:ed:6f:42:d1:a5), tid:1, ssn:0, winSize:64
I (5785) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (6235) wifi:<ba-del>idx:1, tid:1
I (6235) wifi:<ba-del>idx:0, tid:0
I (6235) wifi:<ba-add>idx:1 (ifx:0, 1c:ed:6f:42:d1:a5), tid:1, ssn:0, winSize:64
I (6245) wifi:<ba-add>idx:0 (ifx:0, 1c:ed:6f:42:d1:a5), tid:0, ssn:2, winSize:64
I (6585) wifi:<ba-del>idx:0, tid:0
I (6585) wifi:<ba-add>idx:0 (ifx:0, 1c:ed:6f:42:d1:a5), tid:0, ssn:2, winSize:64
I (6725) esp_netif_handlers: example_netif_sta ip: 192.168.188.53, mask: 255.255.255.0, gw: 192.168.188.1
I (6725) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.188.53
I (7435) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:7edf:a1ff:fe06:99f6, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (7445) example_common: Connected to example_netif_sta
I (7455) example_common: - IPv4 address: 192.168.188.53,
I (7455) example_common: - IPv6 address: fe80:0000:0000:0000:7edf:a1ff:fe06:99f6, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (7475) smtp_example: Seeding the random number generator
I (7475) smtp_example: Loading the CA root certificate...
I (7495) smtp_example: Setting hostname for TLS session...
I (7495) smtp_example: Setting up the SSL/TLS structure...
I (7495) smtp_example: Connecting to mail@shmelkin.net:465...
I (7505) main_task: Returned from app_main()
E (8505) smtp_example: mbedtls_net_connect returned -0x52
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4001ad62  PS      : 0x00060830  A0      : 0x800a00ba  A1      : 0x3ffdf850  
0x4001ad62: memset in ROM

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000064  A5      : 0x00000000  
A6      : 0x00000060  A7      : 0x00000006  A8      : 0x800a17d2  A9      : 0x3ffdf830  
A10     : 0x3ffdf9fc  A11     : 0x00000000  A12     : 0x00000190  A13     : 0x3ffdfb8c  
A14     : 0x3ffdfb8c  A15     : 0x00000019  SAR     : 0x00000019  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x00000190  LEND    : 0x3ffdfb8c  LCOUNT  : 0x40026348  
0x40026348: _xt_handle_exc at /home/iljashmelkin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:739

Backtrace: 0x4001ad5f:0x3ffdf850 0x400a00b7:0x3ffdf860 0x40087bf7:0x3ffdf880 0x4002deb9:0x3ffdffe0
0x4001ad5f: memset in ROM

0x400a00b7: mbedtls_strerror at /home/iljashmelkin/esp/esp-idf/components/mbedtls/mbedtls/library/error.c:800

0x40087bf7: smtp_client_task at /home/iljashmelkin/Documents/esp32/smtp_client/smtp_client/main/smtp_client_example_main.c:483

0x4002deb9: vPortTaskWrapper at /home/iljashmelkin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 15c0b29e3352de0e

Rebooting...
```

MaazSk, 2023-08-13:
Hi shmelkin,

I wanted to understand if you have updated the certificate for your smtp client in the code?
The reason I want to know this is because the error `-0x52` corresponds to `MBEDTLS_ERR_X509_CERT_VERIFY_FAILED` 
You can get certificate from `openssl s_client -connect [smtp.gmail.com](http://smtp.gmail.com/):587 -starttls smtp -showcerts`

shmelkin, 2023-08-13:
Hi,
I did not do that. I was trying around now but I cant get it to work.

First off, the command listed here doesnt work for me

```
/**
 * Root cert for smtp.googlemail.com, taken from server_root_cert.pem
 *
 * The PEM file was extracted from the output of this command:
 * openssl s_client -showcerts -connect smtp.googlemail.com:587 -starttls smtp
 *
 * The CA root cert is the last cert given in the chain of certs.
 *
 * To embed it in the app binary, the PEM file is named
 * in the component.mk COMPONENT_EMBED_TXTFILES variable.
 */
 ```
 For me it would be this command `openssl s_client -showcerts -connect mail.shmelkin.net:465 -starttls smtp`
 
 The output it generates, however, is not a certificate but just `CONNECTED(00000003)`
 
 The other command that you provided seems to work, however, it provides a whole lot of certificates and I tried the first and the last one.
 
`openssl s_client -connect mail.shmelkin.net:465 -showcerts`

Output: 

```
depth=2 C = US, O = Internet Security Research Group, CN = ISRG Root X1
verify return:1
depth=1 C = US, O = Let's Encrypt, CN = R3
verify return:1
depth=0 CN = mail.shmelkin.net
verify return:1
---
Certificate chain
 0 s:CN = mail.shmelkin.net
   i:C = US, O = Let's Encrypt, CN = R3
   a:PKEY: rsaEncryption, 4096 (bit); sigalg: RSA-SHA256
   v:NotBefore: Jul 25 23:01:51 2023 GMT; NotAfter: Oct 23 23:01:50 2023 GMT
-----BEGIN CERTIFICATE-----
MIIGVTCCBT2gAwIBAgISA6Ay6K5ylqjMU2FHd65eQpJNMA0GCSqGSIb3DQEBCwUA
MDIxCzAJBgNVBAYTAlVTMRYwFAYDVQQKEw1MZXQncyBFbmNyeXB0MQswCQYDVQQD
EwJSMzAeFw0yMzA3MjUyMzAxNTFaFw0yMzEwMjMyMzAxNTBaMBwxGjAYBgNVBAMT
EW1haWwuc2htZWxraW4ubmV0MIICIjANBgkqhkiG9w0BAQEFAAOCAg8AMIICCgKC
AgEAw0QWkNjF+fACprDN1YnPZre7qMrLv/FxdXYShvLz+B8lqBh584jhd0uTH33d
ygvjxc5qFo+4/A6tb0Ogt7blYW1Ed+8AVHDCkYHa00QGK/ElyhL91ERwhNyfIQuQ
zdJW/hNr9jX8kwpluUiph8hPhVS4yQPf1i53RtGRuI6gHcne1IEerErF3eLm7i+0
HB4WjtUhMpvtkgE+H1rzFsZ86PgtlcsnCdilIGqieB+M4ApdWkeWJWkWchC/8jB2
a16kvhqiH5hBbJBq9SjcF2m6l9H8++r3r2B03+zqZPoPMXBEAR71WXbjJxGJHZU5
EYSTtNJCkahGnMvn8LsSsydQ6wZSlxisq9kfF1IoSz33+86k/jDKGK5fZ3JFbMTY
XoCSmZI5tf72HQdsTY8DhlA1BJXam2dz5WYtWSGHABX6NhHsjauXpR/oeerQYcL2
t63ksMbGe971gBMCRdxwivAywroOiWnfoDnSb7I4qRcjQlRzHz6Fi4YlJZR9GOYy
g3ZlykZ7eFdkm0NLnwxh18wKz6F5+qIWjgAIRUWo7zZzlS1kLqdAIT9Zmg6Zhu8E
LWgzLzJ5y7eY7dDDG1Nyvj2GINUA6RRLjWkOIty8pZjZ0OnvTOoaTfXGuKEfBO6R
uPuqro0jvyNCfYAqUVbOCzi1F8HiOMQZL1ArexoKqq2FN/MCAwEAAaOCAnkwggJ1
MA4GA1UdDwEB/wQEAwIFoDAdBgNVHSUEFjAUBggrBgEFBQcDAQYIKwYBBQUHAwIw
DAYDVR0TAQH/BAIwADAdBgNVHQ4EFgQUc9su0dYxeys8CRR5+ukdOeQtHVYwHwYD
VR0jBBgwFoAUFC6zF7dYVsuuUAlA5h+vnYsUwsYwVQYIKwYBBQUHAQEESTBHMCEG
CCsGAQUFBzABhhVodHRwOi8vcjMuby5sZW5jci5vcmcwIgYIKwYBBQUHMAKGFmh0
dHA6Ly9yMy5pLmxlbmNyLm9yZy8wgYAGA1UdEQR5MHeCFWF1dG9jb25maWcuZmFk
b2dpYS5kZYIXYXV0b2NvbmZpZy5zaG1lbGtpbi5uZXSCF2F1dG9kaXNjb3Zlci5m
YWRvZ2lhLmRlghlhdXRvZGlzY292ZXIuc2htZWxraW4ubmV0ghFtYWlsLnNobWVs
a2luLm5ldDATBgNVHSAEDDAKMAgGBmeBDAECATCCAQUGCisGAQQB1nkCBAIEgfYE
gfMA8QB3AHoyjFTYty22IOo44FIe6YQWcDIThU070ivBOlejUutSAAABiY+AggwA
AAQDAEgwRgIhAKZO+THI9QKUV/RcHFAdR+IPzz9/QRFsOD18JbOVAFOaAiEAwc96
3cUCkmahcFtHqVF0xP+Mkz0jTZy6pUrH+gIc1WgAdgC3Pvsk35xNunXyOcW6WPRs
XfxCz3qfNcSeHQmBJe20mQAAAYmPgIIPAAAEAwBHMEUCIBUpPjFVclcABNkCQci6
RyLqirt6Amy28TA3qiDs87KTAiEA1YYodRaN2jwVSChv4d5PFp0K6b7xkcxVHnfU
[... 118 lines cut ...]
    0070 - 8b 08 d1 2c af dc cc d8-88 a3 1e ee 55 1c 42 4c   ...,........U.BL
    0080 - 8c 1d 00 54 8d 05 ed 0f-00 12 d5 e4 9c 40 01 dc   ...T.........@..
    0090 - a8 e3 3c 6b ec a1 10 7e-24 cf 55 9e c1 7c 5f 76   ..<k...~$.U..|_v
    00a0 - bc de 4b d9 81 89 f1 4f-e9 ea d1 56 2a 4a 65 64   ..K....O...V*Jed
    00b0 - 38 72 20 dc 37 9f 27 48-cb 39 ed dc 67 41 31 a8   8r .7.'H.9..gA1.
    00c0 - e6 59 cd 7e be 5d b0 87-a2 7b 01 66 bc 73 12 60   .Y.~.]...{.f.s.`
    00d0 - 7f 93 4c 9b 0b e5 46 09-3e a2 f4 5e d6 fc ce ae   ..L...F.>..^....

    Start Time: 1691925426
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: no
    Max Early Data: 0
---
read R BLOCK
220 mail.shmelkin.net ESMTP Postcow
```

I copied the first and the last certificate in the file smtp_client/main/server_root_cert.pem

built the project and flashed it. Unfortunately, I get the same error.

What am I doing wrong?

Best!

MaazSk, 2023-08-13:
`-----BEGIN CERTIFICATE-----
MIIFYDCCBEigAwIBAgIQQAF3ITfU6UK47naqPGQKtzANBgkqhkiG9w0BAQsFADA/
MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT
DkRTVCBSb290IENBIFgzMB4XDTIxMDEyMDE5MTQwM1oXDTI0MDkzMDE4MTQwM1ow
TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh
cmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwggIiMA0GCSqGSIb3DQEB
AQUAA4ICDwAwggIKAoICAQCt6CRz9BQ385ueK1coHIe+3LffOJCMbjzmV6B493XC
ov71am72AE8o295ohmxEk7axY/0UEmu/H9LqMZshftEzPLpI9d1537O4/xLxIZpL
wYqGcWlKZmZsj348cL+tKSIG8+TA5oCu4kuPt5l+lAOf00eXfJlII1PoOK5PCm+D
LtFJV4yAdLbaL9A4jXsDcCEbdfIwPPqPrt3aY6vrFk/CjhFLfs8L6P+1dy70sntK
4EwSJQxwjQMpoOFTJOwT2e4ZvxCzSow/iaNhUd6shweU9GNx7C7ib1uYgeGJXDR5
bHbvO5BieebbpJovJsXQEOEO3tkQjhb7t/eo98flAgeYjzYIlefiN5YNNnWe+w5y
sR2bvAP5SQXYgd0FtCrWQemsAXaVCg/Y39W9Eh81LygXbNKYwagJZHduRze6zqxZ
Xmidf3LWicUGQSk+WT7dJvUkyRGnWqNMQB9GoZm1pzpRboY7nn1ypxIFeFntPlF4
FQsDj43QLwWyPntKHEtzBRL8xurgUBN8Q5N0s8p0544fAQjQMNRbcTa0B7rBMDBc
SLeCO5imfWCKoqMpgsy6vYMEG6KDA0Gh1gXxG8K28Kh8hjtGqEgqiNx2mna/H2ql
PRmP6zjzZN7IKw0KKP/32+IVQtQi0Cdd4Xn+GOdwiK1O5tmLOsbdJ1Fu/7xk9TND
TwIDAQABo4IBRjCCAUIwDwYDVR0TAQH/BAUwAwEB/zAOBgNVHQ8BAf8EBAMCAQYw
SwYIKwYBBQUHAQEEPzA9MDsGCCsGAQUFBzAChi9odHRwOi8vYXBwcy5pZGVudHJ1
c3QuY29tL3Jvb3RzL2RzdHJvb3RjYXgzLnA3YzAfBgNVHSMEGDAWgBTEp7Gkeyxx
+tvhS5B1/8QVYIWJEDBUBgNVHSAETTBLMAgGBmeBDAECATA/BgsrBgEEAYLfEwEB
ATAwMC4GCCsGAQUFBwIBFiJodHRwOi8vY3BzLnJvb3QteDEubGV0c2VuY3J5cHQu
b3JnMDwGA1UdHwQ1MDMwMaAvoC2GK2h0dHA6Ly9jcmwuaWRlbnRydXN0LmNvbS9E
U1RST09UQ0FYM0NSTC5jcmwwHQYDVR0OBBYEFHm0WeZ7tuXkAXOACIjIGlj26Ztu
MA0GCSqGSIb3DQEBCwUAA4IBAQAKcwBslm7/DlLQrt2M51oGrS+o44+/yQoDFVDC
5WxCu2+b9LRPwkSICHXM6webFGJueN7sJ7o5XPWioW5WlHAQU7G75K/QosMrAdSW
9MUgNTP52GE24HGNtLi1qoJFlcDyqSMo59ahy2cI2qBDLKobkx/J3vWraV0T9VuG
WCLKTVXkcGdtwlfFRjlBz4pYg1htmf5X6DYO8A4jqv2Il9DjXA6USbW1FzXSLr9O
he8Y4IWS6wY7bCkjCWDcRQJMEhg76fsO3txE+FiYruq9RUWhiF1myv4Q6W+CyBFC
Dfvp7OOGAN6dEOM4+qR9sdjoSYKEBpsr6GtPAQw4dy753ec5
-----END CERTIFICATE-----
`
Just wanted to know if adding only this certificate is also not working?

If no please give me a moment I'll also try and will let you know

shmelkin, 2023-08-13:
I can confirm this does still not work. 

Copied the respective certificate from the command into the file and rebuilt the project, flashed. Resulted in the same behavior.

Maybe the file in the binary is not updated???

MaazSk, 2023-08-13:
Just pulled the updated code. Will try and update

MaazSk, 2023-08-13:
I think you should check your configuration 

> I (7495) smtp_example: Setting up the SSL/TLS structure...
> I (7495) smtp_example: Connecting to mail@shmelkin.net:465...
> I (7505) main_task: Returned from app_main()
> E (8505) smtp_example: [...]

Shouldn't it be connecting to `mail.shmelkin.net`

shmelkin, 2023-08-13:
I cant believe that. Yes I had this typo. Yes the program is now able to connect to my SMTP server. I actually had to look three times until I noticed what you meant in your last message. 
Its still not sending out that mail but that is for sure a different problem. Thank you, have a nice evening sir!!!

closed 2023-08-13 by shmelkin