espressif/esp-idf #12067: WiFi Disconnect Issue using ESP Now

naakhtar, 2023-08-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP V5.0

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

1. With application customer connect module and send data to display but to display data current lot module is taking more time to display data on screen compared to last lot  and after display ESP also getting disconnected ( which should not happen)
2. OTA portion in Module is getting deleted .

Note:- Customer is using ESPNOW protocol.

### What is the actual behavior?

While implementing ESP Now wifi gets disconnect.

### Steps to reproduce.

1. Step - ESP32-WROOM-32 E module connects with via WiFi with mobile. By mobile message fired via WiFi into module to change to reflect price.

2. Step - ESP32-WROOM-32E is connects with others via ESP NOW protocol.

3. Step: ESP NOW protocol is not reflecting and WiFi gets disconnects.
...

Xiehanxin, 2023-08-14:
hi @naakhtar  could you provide some more informations, like detail log and sdkconifg?

zhangyanjiaoesp, 2024-03-26:
@naakhtar do you still have this problem?

Sherry616, 2024-04-23:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.