espressif/esp-idf #12068: How to connect custom ESP32 device to Alexa for RGB lightning control

krupis, 2023-08-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. I have designed my own ESP32 based device to control RGB lightning. I use touch display ST7789 to control the colors and everything works fine!

I would love to somehow make my device compatible with Alexa but not too sure where to start. Perhaps someone could put me on the right track? i have tried searching but could not find any relevant information. Is there any esp-idf examples on how to get things going?

Thanks in advance.

MaazSk, 2023-08-13:
Hi Krupis,

To connect ESP32 with Alexa you can follow this [tutorial of randomnerds](https://randomnerdtutorials.com/alexa-echo-with-esp32-and-esp8266/) it is based on Arduino but it will give you an idea of how it can be achieved

krupis, 2023-08-13:
@MaazSk 
I have actually seen this but was hoping to find something similar that is intended for use with esp-idf and not Arduino framework. I am sure someone has done this on esp-idf

MaazSk, 2023-08-13:
@krupis 

Have a look at this [Alexa Connect Kit](https://www.espressif.com/en/solutions/device-connectivity/ack-solution). 
But if you want to make it on your own the following steps would be required
1.  Connect esp32 to AWS through MQTT or you can use any other MQTT broker
2. Create an Alexa skill to send the data to your device
3. Write a lambda function to connect the skill and the esp32

Or you can make use of [Home assistant](https://www.home-assistant.io/)  which can help you

I hope this helps you. I was thinking to make a repo and create it as an hobby project please let me know if you are interested we can set it up

krupis, 2023-08-13:
@MaazSk 
I had a look at Alexa Conenct Kit link that you have provided. The description sounds nice. Apparently they provide SDK where you can write an alexa compatible application. Unfortunately, it does not seem to be easily available. You have to manually request the access to the SDK:
[image]

Clicking on Request ACK SDK access opens up error 404 page:
[image]

I think it is definately a good idea to make a public repo for esp-idf that would help developers get their alexa compatible projects up and running.