espressif/esp-idf #12069: Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Talha-Babar, 2023-08-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf 5

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32

### Power Supply used.

USB

### What is the expected behavior?

I expected it to give the data in the using i2c.

when i call this function in main.c file it works fine but when i call it .c file it give the error

int i2c_slave_write(uint8_t bus, uint8_t addr, const uint8_t *reg, uint8_t *data, uint32_t len)
{

    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, addr << 1 | I2C_MASTER_WRITE, true);
    i2c_master_write_byte(cmd, *reg, true);
    i2c_master_write(cmd, data, len, true);
    i2c_master_stop(cmd);
    esp_err_t err = i2c_master_cmd_begin(bus, cmd, 1000 / portTICK_PERIOD_MS);
    i2c_cmd_link_delete(cmd);

    return err;
}

### What is the actual behavior?

it give the error  Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled. in log.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
I (181) cpu_start: App cpu up.
I (209) cpu_start: Pro cpu start user code
I (209) cpu_start: cpu freq: 160000000 Hz
I (209) cpu_start: Application information:
I (214) cpu_start: Project name:     main
I (218) cpu_start: App version:      1
I (223) cpu_start: Compile time:     Aug 13 2023 14:52:20
I (229) cpu_start: ELF file SHA256:  166ee6473534c6f0...
I (235) cpu_start: ESP-IDF:          v5.0.1-dirty
I (240) cpu_start: Min chip rev:     v0.0
I (245) cpu_start: Max chip rev:     v3.99 
I (250) cpu_start: Chip rev:         v3.0
I (255) heap_init: Initializing. RAM available for dynamic allocation:
I (262) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (268) heap_init: At 3FFB2818 len 0002D7E8 (181 KiB): DRAM
I (274) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (280) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (287) heap_init: At 4008CF80 len 00013080 (76 KiB): IRAM
I (295) spi_flash: detected chip: generic
I (298) spi_flash: flash io: dio
W (302) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (316) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
- i2c controller configured
- i2c driver installed

- before i2c_slave_write
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d5137  PS      : 0x00060c30  A0      : 0x800d4f73  A1      : 0x3ffb4a80  
0x400d5137: i2c_slave_write at C:/Users/Talha/Downloads/Ultrasonic_HallEffect-main/Ultrasonic_HallEffect-main/components/temp_hum/i2c.c:38

A2      : 0x00000000  A3      : 0x00000086  A4      : 0x00000021  A5      : 0x3ffb4ab0
A6      : 0x00000001  A7      : 0x3ffb59e8  A8      : 0x800d5135  A9      : 0x3ffb4a40
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x3ffaff70  
A14     : 0x3ffaff78  A15     : 0x0000cdcd  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000021  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d5134:0x3ffb4a80 0x400d4f70:0x3ffb4aa0 0x400d4f1f:0x3ffb4af0 0x400e80ee:0x3ffb4b10 0x400895bd:0x3ffb4b40
0x400d5134: i2c_slave_write at C:/Users/Talha/Downloads/Ultrasonic_HallEffect-main/Ultrasonic_HallEffect-main/components/temp_hum/i2c.c:37

0x400d4f70: temp_hum at C:/Users/Talha/Downloads/Ultrasonic_HallEffect-main/Ultrasonic_HallEffect-main/components/temp_hum/temp_hum.c:24

0x400d4f1f: app_main at C:/Users/Talha/Downloads/Ultrasonic_HallEffect-main/Ultrasonic_HallEffect-main/main/main.c:15 (discriminator 1)

0x400e80ee: main_task at C:/Users/Talha/esp/esp-idf-v5.0.1/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

0x400895bd: vPortTaskWrapper at C:/Users/Talha/esp/esp-idf-v5.0.1/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

ESP-Marius, 2023-08-14:
`LoadProhibited` implies you are trying to read from an invalid address. `EXCVADDR` tells us what this address was: `0x21`

Looking at the registers we see that `0x21` is in `A4`, which also happens to be the registers used to pass the 3rd function parameter `const uint8_t *reg`. 

So my guess is that you passed the value `0x21` as the `reg` parameter, but what you actually should pass is a reference to this value.

Talha-Babar, 2023-08-14:
Thank you very much for your help.  I adjusted the code as you suggested, and the issue is now resolved.

Could you kindly elaborate the issue for better understanding. Why calling the address without the reference causing the issue?

ESP-Marius, 2023-08-14:
> Why calling the address without the reference causing the issue? 

Because the functions expects a pointer, not a value. `i2c_master_write_byte(cmd, *reg, true);` will try to read what is at the address `reg`, and `0x21` is not a valid address.

> Why it was working in the main.c without any reference to address and it need reference in other .c files? 

Not sure what was happening, but in general if a function expects a pointer, you cannot pass a value.

Talha-Babar, 2023-08-14:
Thank you very much for your help.

closed 2023-08-14 by Talha-Babar

barfiii, 2024-02-27:
> `LoadProhibited` implies you are trying to read from an invalid address. `EXCVADDR` tells us what this address was: `0x21`
>
> Looking at the registers we see that `0x21` is in `A4`, which also happens [...]

Hi, I'm getting similar issue. Could you please help me with this ?
Here is my log : 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400833fb  PS      : 0x00060031  A0      : 0x8008376d  A1      : 0x3ffb0dd0
0x400833fb: EspSoftwareSerial::UARTBase::rxBitISR(EspSoftwareSerial::UARTBase*) at C:/Users/Admin/Desktop/SoftSerial/current/SoftSerial/main/SoftwareSerial.cpp:721

A2      : 0x00000012  A3      : 0x00000000  A4      : 0x800886a2  A5      : 0x3ffb6a50
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x00060020  A9      : 0x3ffb71b4
A10     : 0x3ffb71b4  A11     : 0xb33fffff  A12     : 0x800da455  A13     : 0x3ffb6a20
A14     : 0x3ff5f000  A15     : 0x00060023  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c
EXCVADDR: **0x0000001a**  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x400833f8:0x3ffb0dd0 0x4008376a:0x3ffb0e00 0x400837fa:0x3ffb0e30 0x40082b7d:0x3ffb0e60 0x40084d2b:0x3ffb6ac0 0x400d2db7:0x3ffb6ae0 0x40086e21:0x3ffb6b00 0x400883b9:0x3ffb6b20
0x400833f8: EspSoftwareSerial::UARTBase::rxBitISR(EspSoftwareSerial::UARTBase*) at C:/Users/Admin/Desktop/SoftSerial/current/SoftSerial/main/SoftwareSerial.cpp:720

And the function is :
```
void IRAM_ATTR UARTBase::rxBitISR(UARTBase* self) {
		const bool level = *self->m_rxReg & self->m_rxBitMask;
		const uint32_t curTick = ticks();
		const bool empty = !self->m_isrBuffer->available();

		// Store level and tick in the buffer unless we have an overflow
		// tick's LSB is repurposed for the level bit
		if (!self->m_isrBuffer->push((curTick | 1U) ^ !level)) self->m_isrOverflow.store(true);
		// Trigger rx callback only when receiver is starved
		if (empty) self->m_rxHandler();
}
```

ESP-Marius, 2024-02-28:
Hard to tell from just that snippet, but my guess would be that either `self` or `self->m_rxReg` has invalid values. You can print them and check.

barfiii, 2024-02-28:
Thanks for the reply. The problem is with the `self->m_rxReg` variable. But I didn't face any these kinds of run time errors at this function while using arduino compiler. Could you please explain why it is having this problem now ?

ESP-Marius, 2024-02-28:
> Could you please explain why it is having this problem now? 

Without knowing the difference between the two projects I cannot really say. Could be due config differences, you missing to do something that the ardunio code did for you automatically etc.

barfiii, 2024-02-29:
Do you have any references to the static function that has been accessing it's class members via pointer like `(UARTBase* self)` in esp-idf examples?