espressif/esp-idf #12070: Configurable TX retry count for Wi-Fi data frames

nickvv-tpw, 2023-08-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My company has a product that does realtime video streaming over Wi-Fi, where we value low latency above reliable transmissions. We have implemented a forward error correction system which allows us to recover from some level of packet loss due to wireless interference. We have implemented a protocol on top of UDP for transmitting the realtime video data, exploiting on the fact that the data is only flowing in a single direction and that we do not need any acknowledgement that UDP datagrams are received at the other end. When packets end up getting dropped due to radio interference, in most cases we're able to recover by using the accompanying forward error correction data.

This works well with our current design, but we're now looking into porting this over to the ESP32-C6 in order to take advantage of the Wi-Fi 6 support. We now have a functional prototype based on the ESP32-C6, and we have run into an issue which greatly affects the viability of the project.

With our existing design, achieving smooth low latency transmission of our realtime data relies on us being able to configure the TX retry limit at the Wi-Fi MAC level. Because we already have a forward error correction layer built into the protocol, in the case where radio interference is causing failed transmissions, it is better to more readily drop outgoing packets rather than to keep retrying until a successful transmission.

Unfortunately it appears that with ESP32-C6, the TX retry number is set to 31 and hardcoded in the driver (as noted here: https://docs.espressif.com/projects/esp-faq/en/latest/software-framework/wifi.html#what-is-the-retransmission-time-for-esp32-s-data-frame-and-management-frame-can-this-be-configured). This causes a big problem for our attempts to port to ESP32-C6, because when we encounter a congested radio environment, our outgoing video data packets will get tied up in the Wi-Fi TX buffers while the driver is cycling through the 31 retransmission attempts for each packet. This ends up creating considerable jitter in the transmission times of the outgoing packets, and in addition it can cause the TX buffers to quickly become depleted and subsequent attempts to transmit returning "out of memory" error from the lwip stack (it should be noted that we've already maxed out the TX buffers as much as the SRAM size will allow).

Is there any way, officially supported or not, to configure the TX retry number for Wi-Fi data frames? Without this seemingly trivial feature, it's unlikely that we will be able to continue with the ESP32-C6.

nickvv-tpw, 2023-08-28:
Hi, can I please get a response on this? Simply a yes or no would be better than nothing.

xuxiao111, 2023-08-28:
Hi, sorry for the late response, we can provide a temporary configuration item to modify the number of TX retransmissions.

nickvv-tpw, 2023-08-28:
That would be great, thank you. Is this something that can be configured at runtime or it needs to be hardcoded into the build?

xuxiao111, 2023-08-28:
Hi @nickvv-tpw, you can use this internal API "int esp_wifi_internal_set_retry_counter(int src, int lrc)" to set the TX retry number at runtime. And the parameter "src" and "lrc" means short-frame retry and long-frame retry. The maximum value should not exceed 32. You can have a try, If you have any questions, please let me know.

nickvv-tpw, 2023-08-29:
Thank you, I'll give this a try and report back if any issues :)

Sherry616, 2024-03-04:
Hi @nickvv-tpw, do you still have this problem? Could you please share the latest update? Thanks.

nickvv-tpw, 2024-03-13:
Hi Sherry,

The issue was resolved thanks to the feedback from your colleague.

Regards,

Nick Van Veen | Director
+852 6296 6343

The Product Workshop Ltd.<https://www.tpwhk.com/> | Kobotix<https://www.kobotix.com/>

Follow Us: Facebook<https://www.facebook.com/kobotix/> | Instagram<https://www.instagram.com/kobotix/> | Twitter<https://www.twitter.com/kobotixtoys> | YouTube<https://www.youtube.com/channel/UCI6E7BrcZqafHkfq9cVhWUw>

Alvin1Zhang, 2024-03-14:
Thanks for sharing the updates, glad to hear the issue is resolved, feel free to reopen.

closed 2024-03-14 by Alvin1Zhang