espressif/esp-idf #12071: Cannot create virtual environment

mrezzaaa, 2023-08-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I want to run ```./install.sh``` and expecting installation success & continue to ```. ./export.sh```

### What is the actual behavior?

Conflicting happen when upgrading to v5.0.3 (stable) with environment miniconda3 installed.
With previous version i have v5.0.1 it's run flawlessly with miniconda3

### Steps to reproduce.

1. Do upgrade like explained on esp-idf documentation page

### Build or installation Logs.

```plain
the Python interpreter
Checking "python3" ...
Python 3.11.3
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Current system platform: macos-arm64
Updating /Users/icewalker/esp-idf/components/esptool_py/esptool/idf-env.json
Selected targets are: esp32s2, esp32h2, esp32s3, esp32c2, esp32c3, esp32
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@11.2_20220823 (already installed)
Skipping riscv32-esp-elf-gdb@11.2_20220823 (already installed)
Skipping xtensa-esp32-elf@esp-2022r1-11.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2022r1-11.2.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2022r1-11.2.0 (already installed)
Skipping riscv32-esp-elf@esp-2022r1-11.2.0 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20230419 (already installed)
Skipping esp-rom-elfs@20220823 (already installed)
Installing Python environment and packages
Creating a new Python environment in /Users/icewalker/opt/miniconda3/bin/python
Error: [Errno 20] Not a directory: '/Users/icewalker/opt/miniconda3/bin/python'
Traceback (most recent call last):
  File "/Users/icewalker/esp-idf/esp-idf/tools/idf_tools.py", line 2637, in <module>
    main(sys.argv[1:])
  File "/Users/icewalker/esp-idf/esp-idf/tools/idf_tools.py", line 2629, in main
    action_func(args)
  File "/Users/icewalker/esp-idf/esp-idf/tools/idf_tools.py", line 2049, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'venv',
  File "/opt/homebrew/Cellar/python@3.11/3.11.3/Frameworks/Python.framework/Versions/3.11/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/Users/icewalker/.espressif/python_env/idf5.0_py3.11_env/bin/python3', '-m', 'venv', '--clear', '--upgrade-deps', '/Users/icewalker/opt/miniconda3/bin/python']' returned non-zero exit status 1.
```

mrezzaaa, 2023-08-14:
which should i use ``IDF_PYTHON_ENV_PATH`` to ?

dobairoland, 2023-08-14:
That shouldn't be set at all. It is used for very specific and advanced purposes.

closed 2023-08-14 by mrezzaaa