espressif/esp-idf #12073: Entire SD card corrupted on any VFS write

i-e-b, 2023-08-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-2164-g3befd5fff7

### Operating System used.

Linux

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROVER-E (Lilygo T-SIM A7670E)

### Power Supply used.

External 3.3V

### What is the expected behavior?

Reading and writing to FAT32 filesystem on an SD card should work.
Writing to a pre-existing file should not change partition table

### What is the actual behavior?

Reading from existing FAT32 filesystem works ok.

Writes (using cstd methods: `fopen`, `fprintf`, `fflush`, `fclose`...) report success

**ANY** write to the filesystem results in complete corruption of SD card:
- Partition *table* damaged (usually results in junk LABELs and FLAGs, sometimes renders the card unusable)
- Partitions damaged (can result in appearance of phantom files, usually turns all files into gibberish)
 

### Steps to reproduce.

Test are done with the steps below.

- 1, 2, 3, 4 -- can be repeated any number of times without any problems
- 1, 2, 5 -- immediately corrupts SD card
- 1, 2, 6 -- immediately corrupts SD card
- 1, 5 -- immediately corrupts SD card
- 1, 6 -- immediately corrupts SD card

1. Mount SD card (no corruption caused during this phase):

```cpp
    // Enable bus
    spi_bus_config_t spiBusConfig = {
        .mosi_io_num = GPIO_NUM_15,
        .miso_io_num = GPIO_NUM_2,
        .sclk_io_num = GPIO_NUM_14
    };
    auto err = spi_bus_initialize(SD_CARD_SPI_HOST, &spiBusConfig , SPI_DMA_CH_AUTO);
    // error checks excluded in bug report for brevity

    // Configure
    sdspi_device_config_t spiConfig = SDSPI_DEVICE_CONFIG_DEFAULT();
    spiConfig.gpio_cs = GPIO_NUM_13;
    spiConfig.host_id = SD_CARD_SPI_HOST;

    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    host.slot = SD_CARD_SPI_HOST;

    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
            .format_if_mount_failed = false,
            .max_files = 4,
            .disk_status_check_enable = false
    };

    err = esp_vfs_fat_sdspi_mount("/sdcard", &host, &spiConfig, &mount_config, &sdCardConfigurationSet);
``` 

2. List directory contents on existing file system (no corruption caused during this phase):

```cpp
    DIR *dp;
    struct dirent *ep;
    dp = opendir ("/sdcard/");
    if (dp != nullptr)
    {
        printf ("Directory content:\n");
        while ((ep = readdir (dp)) != nullptr) {
            printf("    %s\n,", ep->d_name);
        }
        (void) closedir (dp);
        return 0;
    }
    else
    {
        printf ("#### Couldn't open the directory\n");
        return -1;
    }
```

3. Read contents of a pre-existing file (no corruption caused during this phase):

```cpp
    FILE* file = fopen("/sdcard/CONFIG.TXT", "rb");
    if (file != nullptr && !ferror(file)) {
        char str[128];
        auto result = fgets(str, 127, file); // read line
        if (result == nullptr) printf("Failed to read from file\n");
        else printf("Read from file: '%s'\n", result);
        sdCardCloseFile(&file); // also does flush
    } else {
        printf("File was faulty?\n");
    }
    fflush(*file); fclose(*file);
```

4. Open existing file to allow writing, but do not write (no corruption caused during this phase):

```cpp

    FILE* file = fopen("/sdcard/CONFIG.TXT", "rb+");
    fflush(*file); fclose(*file);
```

5. Open existing file to allow writing, write a single byte (this will corrupt the entire SD card):

```cpp

    FILE* file = fopen("/sdcard/CONFIG.TXT", "rb+");
    fputc('\n', file);
    fflush(*file); fclose(*file);
```

6. Create a new file, write NO data, then close (this will corrupt the entire SD card):

```cpp
    file = fopen("/sdcard/MYTEST.TXT", "wb+");
    fflush(*file); fclose(*file);
```

### Debug Logs.

```plain
Before

Directory content:
    CONFIG.TXT
,Trying to read via file system
Opened existing file '/sdcard/CONFIG.TXT'
Read from file: '9600,26,3,0,1,1,0'
```

After
```
Directory content:
    O�����.]�B
,    �o'v�E7.�]�
,    �2���y��.a��
,    0�bfi���.)�}
,Trying to read via file system
File was faulty?
```
```

### More Information.

Tested on 20 different SD cards of different capacities and brands, all of which work on a normal PC (Windows or Linux)

All SD cards have a small text file `CONFIG.TXT` written to them from PC, for this testing (content is `9600,26,3,0,1,1,0`)

I have tested many different settings in the initial setup (step 1 above). Shown is the minimal set that shows the behaviour, but all settings are either invalid (can't mound SD card) or cause the same behaviour.

Most of the time, SD card can be recovered by deleting the partition map and rebuilding, but I have 2 cards that seem totally dead now.

RathiSonika, 2023-08-15:
Hello  @i-e-b ,
Thank you for reporting the issue here, however I tried the example on my setup (ESP32 WROVER) and could not reproduce the issue. I have modified the existing IDF example as per your configuration settings and it worked fine for me. Please find test example and output log for your reference. Can you please reverify if the same example is working on your setup or not?
[sd_card_example_main.txt](https://github.com/espressif/esp-idf/files/12342828/sd_card_example_main.txt)
[output_log.txt](https://github.com/espressif/esp-idf/files/12342846/output_log.txt)

i-e-b, 2023-08-15:
Thank you for your reply.

The example provided fails to mount the SD card.
This was tested with 3 different SD cards from 2 different manufacturers.

Output:
```plain
I (1299) main_task: Calling app_main()
I (1299) example: Initializing SD card
I (1299) example: Using SPI peripheral
I (1309) example: Mounting filesystem
I (1309) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1559) sdspi_transaction: cmd=5, R1 response: command not supported
E (2849) sdmmc_common: sdmmc_init_cid: send_cid returned 0x107
E (2849) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
I (2849) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (2859) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
I (2869) main_task: Returned from app_main()
```

I tried again with completely new hardware (new dev board with new ESP32, new SD card), and powering via USB with a powered hub.

The first run resulted in
```plain
I (1320) main_task: Calling app_main()
I (1320) example: Initializing SD card
I (1320) example: Using SPI peripheral
I (1330) example: Mounting filesystem
I (1330) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1380) sdspi_transaction: cmd=52, R1 response: command not supported
I (1580) sdspi_transaction: cmd=5, R1 response: command not supported
I (2410) example: Filesystem mounted
Name: APPSD
Type: SDSC
Speed: 20.00 MHz (limit: 20.00 MHz)
Size: 240MB
CSD: ver=1, sector_size=512, capacity=491520 read_bl_len=10
SSR: bus_width=1
Directory content:
    CONFIG.TXT
I (2450) example: Reading file /sdcard/nihao.txt
E (2450) example: Failed to open file for reading
I (2460) main_task: Returned from app_main()
```

A second run resulted in:
```plain
I (1320) main_task: Calling app_main()
I (1320) example: Initializing SD card
I (1320) example: Using SPI peripheral
I (1330) example: Mounting filesystem
I (1330) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1380) sdspi_transaction: cmd=52, R1 response: command not supported
I (1580) sdspi_transaction: cmd=5, R1 response: command not supported
W (2410) vfs_fat_sdmmc: failed to mount card (13)
E (2410) vfs_fat_sdmmc: mount_to_vfs failed (0xffffffff).
I (2410) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (2420) example: Failed to mount filesystem. If you want the card to be formatted, set the CONFIG_EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.
I (2430) main_task: Returned from app_main()
```

Inspecting the resulting SD card shows corruption (the FAT file system is no longer recognised. The SD card cannot be mounted):
[image]

adokitkat, 2023-08-15:
Hello. Does this also happen on IDF version `release/v5.1` or `release/v5.0`? Could you try to git checkout to these versions, run `./install.sh` and `. ./export.sh` and then build and flash again please?

i-e-b, 2023-08-15:
Just been to the local shop to get more SD cards, and they all work, no corruption.

I have no idea what would cause all the others to fail (even the Samsung and SanDisk ones), given that they work when used in a PC:
[image]

So at present:
- I have a solution for my immediate problem
- There are some SD cards that work correctly with a PC, but not with the ESP32

I don't have the resources to determine the root cause. Happy to mail you a packet of SD cards if you wish to investigate.

Thank you very much for your time.

closed 2023-08-17 by i-e-b

ahemberger, 2024-02-04:
Hi, I'm experiencing this problem (currently using esp-idf 5.1.2, stable), and trying to get help from google and chatGPT led me to this thread. What ended up being your solution for this? I'm $100 deep in buying random sd cards and sd card breakout boards and am totally flummoxed as to what might be going wrong here. 

What's even more odd is that I've had this working fine for me for a few months - all of a sudden one day a Lexar card I was working with got corrupted when trying to write a file to it, and I've never gotten my project back on track - even the simple sd_card example does not work for me (and it definitely did on this equipment once before).

i-e-b, 2024-02-05:
As far as I can tell for my issue -- there are at least 2 protocols that SD cards might use, and the cheap cards only supported one of them, the ESP32 only supported the other.

For your issue, it sounds like maybe a hardware fault -- damaged or corroded contacts, a blown capacitor?
Do you have a second board to test with, or can you test the faulting card in a PC?

adokitkat, 2024-02-06:
Hi @ahemberger. 

Unfortunately there are various possible factors in SD cards not working properly... 

For example: 
1) Connect your SD card to a PC, reformat it and see if it still works and is not faulty
1) Sometimes the SD adapter can be faulty as well (I had some external ones which didn't work at all for some reason)
1) If you specified a custom clock frequency, the ESP32 is limited by its clock divider and the specified frequency is rounded to nearest obtainable frequency, which is calculated like this: `160MHz / (n + 1), 0 < n < 16` -> not all SD cards are compatible with all frequencies, so sometimes lowering the clock helps. You can try to set probing frequency (400kHz) or 1Mhz first and then slowly raise the frequency.
1) Signal integrity plays a large role in a successful communication - as @i-e-b said, check the cables (if they are not damaged or too long), PCB, components, check also if GPIO pins and cables/routes connected to the SD card are connected to correct pull-up capacitors as well as check if any GPIO used to connect to SD card has a "dual purpose" -> for example if MISO (or any other) pin is connected to a transistor which controls a RGB LED and even if the LED is turned off, the transistor behaves like a capacitor and slows rising and falling edge of the signal, resulting in some SD cards having problem connecting to ESP32 (which can be solved again by lowering the clock speed, using another SD card or removing the component causing this)
1) Cheap cards without a proper branding may not fully comply with SD specification and therefore not work in certain modes and speeds with ESP32s
1) Maybe there is a regression in the IDF version you're using, try to use a newer release branch like release/v5.2 or master branch

Try to get the sd_card example working first.

maranhao, 2024-02-07:
I pursue this issue for a long time, and am already tired of the workaround of changing sd cards.
There seem to be a thin limit between it working and not working, since I've experienced all combinations possible:
- Card works with some ESP32 but not the others;
- ESP32 works with some cards but not others;
- Cards that work on all ESP32;
- Cards that work on PC but not on any ESP32.

Some information:
1. I am in what once was a release version of idf v4.2;
2. Custom HW working on 1-bit SD protocol, 10k external pull-ups on top of internal pull-ups; 
3. When tested, results were the same with custom code or examples;
4. All the cards I test are 16GB SDHC, some from shady sources, but that does not seem to determine if it works or not;
5. The results seem to be the same with max frequency of 20MHz or 400 kHz;
6. When gone wrong, the initialization does not get the name of the card and says it's 121MB in size;
7. And in the cards that do not work the card_ocr is 0x80ff8000 instead of 0xc0ff8000, making it a SDSC instead of SDHC/XC.

Here's an excerpt from a card that does not work:
W (80982) sdmmc_init: sdmmc_card_init: sdmmc_fix_host_flags returned 0x0
W (80982) sdmmc_init: sdmmc_card_init: sdmmc_io_reset returned 0x0
W (81002) sdmmc_init: sdmmc_card_init: sdmmc_send_cmd_go_idle_state returned 0x0
W (81002) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_if_cond returned 0x0
W (81002) sdmmc_init: sdmmc_card_init: sdmmc_init_io returned 0x0
W (81032) sdmmc_common: host_ocr=0x40ff8000 card_ocr=0x80ff8000
W (81032) sdmmc_common: sdmmc_card_init: host_ocr=40ff8000, card_ocr=80ff8000
W (81032) sdmmc_init: sdmmc_card_init: sdmmc_init_ocr returned 0x0
W (81042) sdmmc_init: sdmmc_card_init: sdmmc_init_cid returned 0x0
W (81042) sdmmc_init: sdmmc_card_init: sdmmc_init_rca returned 0x0
W (81052) sdmmc_init: sdmmc_card_init: sdmmc_init_csd returned 0x0
W (81062) sdmmc_init: sdmmc_card_init: sdmmc_init_select_card returned 0x0
W (81062) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_blocklen returned 0x0
W (81072) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_scr returned 0x0
W (81082) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_wait_data_ready returned 0x0
W (81092) sdmmc_init: sdmmc_card_init: sdmmc_init_card_hs_mode returned 0x0
W (81102) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_bus_width returned 0x0
W (81102) sdmmc_common: sdmmc_init_host_bus_width: using 1-bit bus
W (81112) sdmmc_init: sdmmc_card_init: sdmmc_init_host_bus_width returned 0x0
W (81122) sdmmc_common: sdmmc_init_host_frequency: using 400 kHz bus frequency
W (81122) sdmmc_init: sdmmc_card_init: sdmmc_init_host_frequency returned 0x0
W (81132) sdmmc_init: sdmmc_card_init: sdmmc_check_scr returned 0x0
Name:
Type: SDSC
Speed: 400 kHz
Size: 121MB

And another from one that works:

W (59372) sdmmc_init: sdmmc_card_init: sdmmc_fix_host_flags returned 0x0
W (59372) sdmmc_init: sdmmc_card_init: sdmmc_io_reset returned 0x0
W (59392) sdmmc_init: sdmmc_card_init: sdmmc_send_cmd_go_idle_state returned 0x0
W (59392) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_if_cond returned 0x0
W (59392) sdmmc_init: sdmmc_card_init: sdmmc_init_io returned 0x0
W (59512) sdmmc_common: host_ocr=0x40ff8000 card_ocr=0xc0ff8000
W (59512) sdmmc_common: sdmmc_card_init: host_ocr=40ff8000, card_ocr=c0ff8000
W (59512) sdmmc_init: sdmmc_card_init: sdmmc_init_ocr returned 0x0
W (59522) sdmmc_init: sdmmc_card_init: sdmmc_init_cid returned 0x0
W (59522) sdmmc_init: sdmmc_card_init: sdmmc_init_rca returned 0x0
W (59532) sdmmc_init: sdmmc_card_init: sdmmc_init_csd returned 0x0
W (59542) sdmmc_init: sdmmc_card_init: sdmmc_init_select_card returned 0x0
W (59542) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_blocklen returned 0x0
W (59552) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_scr returned 0x0
W (59562) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_wait_data_ready returned 0x0
W (59572) sdmmc_init: sdmmc_card_init: sdmmc_init_card_hs_mode returned 0x0
W (59582) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_bus_width returned 0x0
W (59582) sdmmc_common: sdmmc_init_host_bus_width: using 1-bit bus
W (59592) sdmmc_init: sdmmc_card_init: sdmmc_init_host_bus_width returned 0x0
W (59602) sdmmc_common: sdmmc_init_host_frequency: using 400 kHz bus frequency
W (59602) sdmmc_init: sdmmc_card_init: sdmmc_init_host_frequency returned 0x0
W (59612) sdmmc_init: sdmmc_card_init: sdmmc_check_scr returned 0x0
Name: SMI
Type: SDHC/SDXC
Speed: 400 kHz
Size: 15200MB

adokitkat, 2024-02-07:
Hello @maranhao.

I really advise you to update your ESP-IDF version to at least v4.4 (already in a maintenance period, the support ends in July this year), even better v5.1+, the newer the better (current stable version is v5.2). You IDF version is no longer supported by us and there were a lot of changes in storage components (SDSPI, SDMMC, FATFS, etc.) which may not be backported to your version. 

Also what board are you using? If it's custom, do you have schematics?