espressif/esp-idf #12074: ESP32-S3 ESP-IDF 5 use to flash a new device from it?

filzek, 2023-08-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We are designing a new JIG Fixture to test and prepare a new way to fast testing our devices and one inquire that has come from the sw fw designer engineers are to use the ESP32S3 besides a Raspberry Pi as main Jig Fixture control and test unit, as it can run smooth with a 7" touchscreen, but the main problem is how to FLASH a new ESP32 device from a ESP32 S3?

So, we will have a direct connection among the two ESP32, all parts for the testing.

Question 1:
How could an ESP32S3 directr flash an ESP32 chip?

Question 2:
How could an ESP32S3 burn efuse to the ESP32 chip?

Queston 3:
Is there any real usage scenario where testing machine are being used developed using an ESP32 as main unit control?

dobairoland, 2023-08-15:
> How could an ESP32S3 directr flash an ESP32 chip?

It should be possible with https://github.com/espressif/esp-serial-flasher.

> How could an ESP32S3 burn efuse to the ESP32 chip?

To the best of my knowledge, this is not supported.

> Is there any real usage scenario where testing machine are being used developed using an ESP32 as main unit control?

I'm sorry but I don't get your question exactly. Perhaps you could search the Internet for scenarios where ESP32 are used.

dobairoland, 2023-09-11:
I'm closing this. esp-serial-flasher eFuse support is tracked here: https://github.com/espressif/esp-serial-flasher/issues/44

closed 2023-09-11 by dobairoland