espressif/esp-idf #12077: The program accidentally crashes. Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

01hello, 2023-08-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.0 96119acc

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The program should work properly and should not crash

### What is the actual behavior?

1. The mqtt packets are sent after aes encryption, but occasionally crash occurs during aes encryption

Major error message:

-  Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

-  error: Cannot access memory at address 0x3c25a51e

### Steps to reproduce.

1. The problematic code snippet is as follows:
```c
send_buff = heap_caps_malloc(pack_len + sizeof(int16_t) + 16, MALLOC_CAP_SPIRAM);
    if (send_buff == NULL)
    {
        ret_val = -7;
        goto free_memory;
    }
    send_buff[0] = kind_id >> 8;
    send_buff[1] = (uint8_t)kind_id;

    // AES 加密
    int encrypt_size = 0;
    uint8_t iv[16];
    for (int i = 0; i < 16; i++)
    {
        iv[i] = default_key[i];     /* 重置iv=key */
    }
    ret = net_aes_encrypt(pack_buff, pack_len, &send_buff[2], pack_len + sizeof(int16_t) + 14, &encrypt_size, default_key, iv);
``` 
```c
int net_aes_encrypt(uint8_t *in_buf, int in_size, uint8_t *out_buf, int out_buf_size, int *out_size, const uint8_t *key, uint8_t *iv)
{
    int ret;
    mbedtls_aes_context aes_ctx;
    if (in_buf == NULL || out_buf == NULL)
    {
        return -1;
    }
    /* 刚满16个，那就要在补充称16个字节 */
    int padding_size = in_size + (16 - in_size % 16);   /* 16字节补全后的长度 */
    int padding_data = 16 - (in_size % 16);             /* 补全的字节内容 */
[... 145 lines cut ...]
  15   process 1070393868 xQueueSemaphoreTake (xQueue=0x3fcd088c, xTicksToWait=<optimized out>) at /home/esp/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1570
  16   process 1070511228 0x403838a1 in xQueueReceive (xQueue=0x3c1e3ec4, pvBuffer=0x3fceb2b0, xTicksToWait=<optimized out>) at /home/esp/esp/esp-idf/components/xtensa/include/xt_utils.h:40
  17   process 1070537000 xQueueSemaphoreTake (xQueue=0x3fca2e68 <s_ipc_sem_buffer+92>, xTicksToWait=<optimized out>) at /home/esp/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1570
  18   process 1070541460 0x400559e0 in ?? ()
  19   process 1070535100 0x40383a20 in xQueueSemaphoreTake (xQueue=0x3fca2e0c <s_ipc_sem_buffer>, xTicksToWait=<optimized out>) at /home/esp/esp/esp-idf/components/xtensa/include/xt_utils.h:40
  20   process 1070306980 0x403838a1 in xQueueReceive (xQueue=0x3fcc5b64, pvBuffer=0x3fcc7ad0, xTicksToWait=<optimized out>) at /home/esp/esp/esp-idf/components/xtensa/include/xt_utils.h:40

==================== THREAD 1 (TCB: 0x3fcee8c4, name: 'tiT') =====================
#0  ethernet_input (p=0x3c25a0e0, netif=0x3fceec00) at /home/esp/esp/esp-idf/components/lwip/lwip/src/netif/ethernet.c:113
#1  0x420d3274 in tcpip_thread_handle_msg (msg=0x3c25a100) at /home/esp/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:174
#2  0x420d32ef in tcpip_thread (arg=0x0) at /home/esp/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:148
#3  0x40386400 in vPortTaskWrapper (pxCode=0x420d32bc <tcpip_thread>, pvParameters=0x0) at /home/esp/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

==================== THREAD 2 (TCB: 0x3fcee240, name: 'pub_task_work') =====================
#0  lldesc_get_required_num_constrained (max_desc_size=4092, data_size=96) at /home/esp/esp/esp-idf/components/soc/include/soc/lldesc.h:67
#1  esp_aes_process_dma (ctx=0x3fcc40f0, input=0x3c25a58c <error: Cannot access memory at address 0x3c25a58c>, output=0x3fcd782c <error: Cannot access memory at address 0x3fcd782c>, len=96, stream_out=0x0) at /home/esp/esp/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:370
#2  0x420f4e8e in esp_aes_process_dma_ext_ram (ctx=0x3fcc40f0, input=0x3c25a58c <error: Cannot access memory at address 0x3c25a58c>, output=0x3c25a51e <error: Cannot access memory at address 0x3c25a51e>, len=96, stream_out=0x0, realloc_input=false, realloc_output=realloc_output@entry=true) at /home/esp/esp/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:287
#3  0x420f4c10 in esp_aes_process_dma (ctx=0x3fcc40f0, input=0x3c25a58c <error: Cannot access memory at address 0x3c25a58c>, output=0x3c25a51e <error: Cannot access memory at address 0x3c25a51e>, len=96, stream_out=0x0) at /home/esp/esp/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:366
#4  0x420f4fbd in esp_aes_crypt_cbc (ctx=0x3fcc40f0, mode=1, length=96, iv=0x3fcc41c4 \"PsGC9MSijtRAMbYD\\360A\\314?\\001\", input=0x3c25a58c <error: Cannot access memory at address 0x3c25a58c>, output=0x3c25a51e <error: Cannot access memory at address 0x3c25a51e>) at /home/esp/esp/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:733
#5  0x42022621 in net_aes_encrypt (in_buf=0x3c259f78 <error: Cannot access memory at address 0x3c259f78>, in_size=89, out_buf=0x3c25a51e <error: Cannot access memory at address 0x3c25a51e>, out_buf_size=<optimized out>, out_size=0x3fcc41c0, key=<optimized out>, iv=iv@entry=0x3fcc41c4 \"PsGC9MSijtRAMbYD\\360A\\314?\\001\") at ../components/Module/Network/net_encrypt.c:47
#6  0x42020b97 in mt_report_task_info (para=<optimized out>) at ../components/Module/Network/net_mqtt_trans.c:1259
#7  0x4201924b in pub_report_task_info (state=TASK_BAKING, errcode=NET_OK) at ../components/Module/Public/pub_package.c:1001
#8  0x42019c86 in pub_task_work (p_arg=<optimized out>) at ../components/Module/Public/pub_task.c:245
#9  0x40386400 in vPortTaskWrapper (pxCode=0x42019960 <pub_task_work>, pvParameters=0x0) at /home/esp/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

mahavirj, 2023-08-17:
@01hello 

Is it possible to decouple the MQTT part from this sample application and see if the issue could still be reproduced? Please share minimal sample app that can help to recreate this problem.

Additionally, I would suggest to enable [Stack Overflow](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#stack-overflow) related measures, as they could help to detect potential memory corruption related issue because of stack overflow.

01hello, 2023-08-22:
`/*
 * SPDX-FileCopyrightText: 2010-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>
#include <unistd.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "freertos/semphr.h"
#include "esp_chip_info.h"
#include "esp_flash.h"
#include "esp_log.h"

#include "mbedtls/aes.h"
#include "mbedtls/base64.h"
#include "mbedtls/cipher.h"

static const char *TAG = "main";

TaskHandle_t test_task1_handle;
TaskHandle_t test_task2_handle;
TaskHandle_t test_task3_handle;
TaskHandle_t test_task4_handle;

void test_task1(void *para);
void test_task2(void *para);
void test_task3(void *para);
void test_task4(void *para);

int task1_cnt = 0;
int task2_cnt = 0;
int task3_cnt = 0;
int task4_cnt = 0;

void app_main(void)
{
    int ret;
    printf("Hello world!\n");

    /* Print chip information */
    esp_chip_info_t chip_info;
    uint32_t flash_size;
    esp_chip_info(&chip_info);
    printf("This is %s chip with %d CPU core(s), %s%s%s%s, ",
           CONFIG_IDF_TARGET,
           chip_info.cores,
           (chip_info.features & CHIP_FEATURE_WIFI_BGN) ? "WiFi/" : "",
           (chip_info.features & CHIP_FEATURE_BT) ? "BT" : "",
           (chip_info.features & CHIP_FEATURE_BLE) ? "BLE" : "",
           (chip_info.features & CHIP_FEATURE_IEEE802154) ? ", 802.15.4 (Zigbee/Thread)" : "");

    unsigned major_rev = chip_info.revision / 100;
    unsigned minor_rev = chip_info.revision % 100;
    printf("silicon revision v%d.%d, ", major_rev, minor_rev);
    if(esp_flash_get_size(NULL, &flash_size) != ESP_OK) {
        printf("Get flash size failed");
        return;
    }

    printf("%" PRIu32 "MB %s flash\n", flash_size / (uint32_t)(1024 * 1024),
           (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    printf("Minimum free heap size: %" PRIu32 " bytes\n", esp_get_minimum_free_heap_size());
[... 233 lines cut ...]
        mbedtls_aes_context aes_ctx;
        mbedtls_aes_init(&aes_ctx);
        ret = mbedtls_aes_setkey_enc(&aes_ctx, key, 128);
        if (ret < 0)
        {
            abort();
        }
        ret = mbedtls_aes_crypt_cbc(&aes_ctx, MBEDTLS_AES_ENCRYPT, ENCRYPT_SIZE, iv, (unsigned char *)in_buff, out_buff);
        if (ret < 0)
        {
            abort();
        }
        free(in_buff);
        free(out_buff);
        //
        task4_cnt++;
        vTaskDelay(1 / portTICK_PERIOD_MS);
    }
}
`
thanks for your reply
The above program is the minimum repetition of broken code to do the test.
The above code crashes when using a branch release/v5.0 96119ac. 
Leaving the comments free and using internal ram to allocate memory to in_buff the above program will not crash.
We suspect that decrypted arrays cannot use spiram.

mahavirj, 2023-08-24:
@01hello 

- Please confirm if you are using ESP32-S3 silicon here
- Please provide your `sdkconfig` here
- Did you try running your code against latest `release/v5.0` branch?
- With this test code, what sort of crash do you get? Also it happens in how many iterations? Can you please attach some log here?

Harshal5, 2023-08-25:
Hi @01hello,

I tried running the above test code on the target ESP32-S3 against the commit mentioned above release/v5.0 [96119ac](https://github.com/espressif/esp-idf/commit/96119acc8b0832f95941f37291d206fce86f922d) for almost an hour but was not able to come across any crash. Attached are the application and the sdkconfig that I tried out for the [above](https://github.com/espressif/esp-idf/issues/12077#issuecomment-1687272442) code. ([hello_world.zip](https://github.com/espressif/esp-idf/files/12438576/hello_world.zip))

Could you share some more information regarding the crash and the sdkconfig?

01hello, 2023-08-28:
@Harshal5 @mahavirj 

Thank you for your reply. I'm sorry for the slow reply. I will reply as soon as possible.

- Attached is the project used, using the branch v5.0.2-378-g96119acc8b, I used this project to run immediately crash, serial port output information and coredump analysis are in the compress.(include sdkconfig)
[Test_code.zip](https://github.com/espressif/esp-idf/files/12449886/Test_code.zip)

- I have tried using release/v5.1 25876e8f4946b0fefc7840eaa6cd05fb65f71e15 branch without any problems
- If you also crash after use, You can try changing the relevant part of the routine to the following and it will not crash.
`//uint8_t *in_buff = heap_caps_malloc(ENCRYPT_SIZE, MALLOC_CAP_SPIRAM);
        uint8_t *out_buff = heap_caps_malloc(ENCRYPT_SIZE, MALLOC_CAP_SPIRAM);
        uint8_t *in_buff = heap_caps_malloc(ENCRYPT_SIZE, MALLOC_CAP_8BIT|MALLOC_CAP_INTERNAL);`

- I can't quite understand why adjusting the memory allocated from the spiram to internal ram won't crash. I searched a lot of information but couldn't find the exact explanation.

01hello, 2023-08-30:
@Harshal5 

- I also tested with the 8MB octal psram esp32-s3 devkit and did not crash, but I did crash with the 2MB quad psram esp32-s3 test under the same project. The sdkconfig file is the same as you sent except for CONFIG_SPIRAM_MODE_OCT, which should have no impact on the test.

- I guess it is the chip problem, because three esp32s3 of the same model have crashed, and the model used is ESP32-S3-WROOM-1U M0N8R2       FCC ID:2AC7Z-ESPS3WROOM1U     IC:21098-ESPS3WROOM1U

(8MB octal psram esp32-s3 output log: 
[uart_output_8M.txt](https://github.com/espressif/esp-idf/files/12471403/uart_output_8M.txt)
sdkconfig file: I use the same sdkconfig file as you do

2MB quad psram esp32-s3 output log: 
[uart_output_2M.txt](https://github.com/espressif/esp-idf/files/12471412/uart_output_2M.txt)
sdkconfig file:  
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/12471427/sdkconfig.zip))

Harshal5, 2023-09-04:
@01hello 

We even tried using the ESP32-S3-WROOM-1 M0N8R2, the 2MB PSRAM devkit, but were still not able to reproduce the issue on our side. Following are the logs for the same:

> I (0) cpu_start: Starting scheduler on APP CPU.
I (1088) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
Hello world!
This is esp32s3 chip with 2 CPU core(s), WiFi/BLE, silicon revision v0.1, 8MB external flash
Minimum free heap size: 2426256 bytes
W (2115) main: heap[2397080],CNT[1005][502][502][500]
W (3121) main: heap[2397080],CNT[2011][1005][1005][1003]
W (4125) main: heap[2397080],CNT[3015][1507][1507][1505]
W (5129) main: heap[2397080],CNT[4019][2009][2009][2007]
W (6131) main: heap[2397080],CNT[5021][2510][2510][2508]
W (7135) main: heap[2397080],CNT[6025][3012][3012][3010]
W (8137) main: heap[2397080],CNT[7027][3513][3513][3511]
W (9141) main: heap[2397080],CNT[8031][4015][4015][4013]

Harshal5, 2023-10-17:
Hi @01hello ,

Any updates on the observations? If not, let us close the issue and if needed you can re-open for further discussions.

Harshal5, 2023-11-03:
Closing the issue as of now, please feel free to re-open if needed.

closed 2023-11-03 by Harshal5