espressif/esp-idf #12080: EspTouch with RSA encryption

offgrid88, 2023-08-15:
### Is your feature request related to a problem?

To securely broadcast wifi credentials I use smartconfigv2 but hard-coding the encryption key is not that secure and can be compromised at all time

### Describe the solution you'd like.

Is there a possibility this would be added as smartconfigv3 for example. I can make the implementation but I fear I messed it up as I am not experienced with mbedtls. If there is no chance that this feature will be available, It'll be helpful to have some guidelines on how to implement it the right way.

### Describe alternatives you've considered.

I considered BluFi but I am trying to make multiple devices connect to the same wifi

Xiehanxin, 2023-08-17:
hi @offgrid88 , I would like to clarify your use case. Are you looking to have multiple devices connect to the internet simultaneously？

offgrid88, 2023-08-17:
Yes I am trying to connected a big number of devices to the network simultaneously and I want to encrypt the SSID and PASSWORD of the AP using Asymmetric Encryption .

closed 2023-12-22 by offgrid88