espressif/esp-idf #12084: esp32c6 use idf5.1 rmt two channel send

Shaco-Ma, 2023-08-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32c6-wroom-1U

### Power Supply used.

USB

### What is the expected behavior?

rmt_tx_channel_config_t tx_chan_config = {
        .clk_src = RMT_CLK_SRC_DEFAULT, // select source clock
        .gpio_num = RMT_LED1_STRIP_GPIO_NUM,
        .mem_block_symbols = 64, // increase the block size can make the LED less flickering
        .resolution_hz = RMT_LED_STRIP_RESOLUTION_HZ,
        .trans_queue_depth = 1, // set the number of transactions that can be pending in the background
        .flags.with_dma = false,
};
ESP_ERROR_CHECK(rmt_new_tx_channel(&tx_chan_config, &led_chan[0]));
rmt_tx_channel_config_t tx_chan_config1 = {
        .clk_src = RMT_CLK_SRC_DEFAULT, // select source clock
        .gpio_num = RMT_LED2_STRIP_GPIO_NUM,
        .mem_block_symbols = 128, // increase the block size can make the LED less flickering
        .resolution_hz = RMT_LED_STRIP_RESOLUTION_HZ,
        .trans_queue_depth = 1, // set the number of transactions that can be pending in the background
        .flags.with_dma = false,
};
ESP_ERROR_CHECK(rmt_new_tx_channel(&tx_chan_config1, &led_chan[1]));
// 安装同步管理器
rmt_sync_manager_handle_t synchro = NULL;
rmt_sync_manager_config_t synchro_config = {
        .tx_channel_array = led_chan,
        .array_size = sizeof(led_chan) / sizeof(led_chan[0]),
};
ESP_ERROR_CHECK(rmt_new_sync_manager(&synchro_config, &synchro));
led_strip_encoder_config_t encoder_config = {
        .resolution = RMT_LED_STRIP_RESOLUTION_HZ,
};
ESP_ERROR_CHECK(rmt_transmit(led_chan[0], led_encoder[0], led_strip_pixels[0], sizeof(led_strip_pixels[0]), &tx_config[0]));
ESP_ERROR_CHECK(rmt_transmit(led_chan[1], led_encoder[1], led_strip_pixels[1], sizeof(led_strip_pixels[1]), &tx_config[1]));

use this code open two rmt channel,it's wrong in this log:
E (1283) rmt: rmt_tx_register_to_group(131): no free tx channels
E (1283) rmt: rmt_new_tx_channel(232): register channel failed

### What is the actual behavior?

init two channel wrong log:
E (1283) rmt: rmt_tx_register_to_group(131): no free tx channels
E (1283) rmt: rmt_new_tx_channel(232): register channel failed

### Steps to reproduce.

1. power on
2. wong in open two rmt channel

### Debug Logs.

```plain
E (1283) rmt: rmt_tx_register_to_group(131): no free tx channels
E (1283) rmt: rmt_new_tx_channel(232): register channel failed
```

suda-morris, 2023-08-16:
@Shaco-Ma RMT only has two TX channels on esp32-c6. By average, one TX channel has `48` words of clock memory. If you set a value bigger than that, the previous channel will occupy the next channel.

```diff
--- .mem_block_symbols = 64, // increase the block size can make the LED less flickering
+++ .mem_block_symbols = 48,
```

closed 2023-08-16 by suda-morris

Shaco-Ma, 2023-08-16:
> @Shaco-Ma RMT only has two TX channels on esp32-c6. By average, one TX channel has `48` words of clock memory. If you set a value bigger than that, the previous channel will occupy the next [...]

OK,thanks,but the docs in https://docs.espressif.com/projects/esp-idf/zh_CN/release-v5.1/esp32c6/api-reference/peripherals/rmt.html is Wrong.......
thanks very much!