espressif/esp-idf #12086: 注册的PROFILE_A_APP_ID与esp_gattc_cb回调中不一致

ForgetCSX, 2023-08-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.5-286-g48e28cf3cf

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

任意的esp32开发板

### Power Supply used.

USB

### What is the expected behavior?

在使用 esp_ble_gattc_app_register() 注册后，我认为在static void esp_gattc_cb回调函数中的参数中可以获取到注册的app_id,当运行起来后，确实可以在回调函数中获取到id，但是我发现这个id有时候会变化，不是注册的id，我在回调函数中打印了这个参数，有几次和预期不一样。

### What is the actual behavior?

获取到了不同的app_id

### Steps to reproduce.

1. Step
复制一份gatt_client例程出来
2. Step
在回调函数开头加入打印，以显示每次进入的app_id
[image]
3. Step
get_idf
idf.py set-target esp32
idf.py build
idf.py -p /dev/ttyUSB0 flash monitor
4. Step
使用另一块板子运行gatt_server例子，无需修改
5. Step
此时我们可以看到日志，（日志我放下面了）
可以看到，关于app_id的打印有以下几个不同的值，但我认为他不应该出现0之外的值，我只注册和链接到一个设备。
I (1029) GATTC_DEMO: EVT 0, gattc if 3, app_id 0
I (1279) GATTC_DEMO: EVT 40, gattc if 3, app_id 5560
I (2609) GATTC_DEMO: EVT 7, gattc if 3, app_id 43
I (2649) GATTC_DEMO: EVT 38, gattc if 3, app_id 42

为什么会出现这个问题，因为我想通过这个app_id来区分链接的设备，这个我就不用为每一个设备单独的写一个gattc_profile_event_handler函数了，只需要在函数里面根据id去区分作不同的处理即可。

### Debug Logs.

```plain
Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /home/xiang/esp/test/gatt_client
Executing "/home/xiang/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/xiang/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /home/xiang/esp/test/gatt_client/build/gatt_client_demo.elf -m '/home/xiang/.espressif/python_env/idf4.4_py3.8_env/bin/python' '/home/xiang/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyUSB0'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
����[0m
I (71) boot:  1 phy_init         RF�ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6624
load:0x40078000,len:15060
ho 0 tail 12 room 4
load:0x40080400,len:3816
0x40080400: _init at ??:?

entry 0x40080698
I (29) boot: ESP-IDF v4.4.5-286-g48e28cf3cf 2nd stage bootloader
I (29) boot: compile time 18:44:15
I (29) boot: chip revision: v1.0
I (33) boot.esp32: SPI Speed      : 40MHz
I (38) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1e178h (123256) map
I (143) esp_image: segment 1: paddr=0002e1a0 vaddr=3ffbdb60 size=01e78h (  7800) load
I (146) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=7e570h (517488) map
I (336) esp_image: segment 3: paddr=000ae598 vaddr=3ffbf9d8 size=02044h (  8260) load
I (339) esp_image: segment 4: paddr=000b05e4 vaddr=40080000 size=177f4h ( 96244) load
I (393) boot: Loaded app from partition at offset 0x10000
I (393) boot: Disabling RNG early entropy source...
I (404) cpu_start: Pro cpu up.
I (405) cpu_start: Starting app cpu, entry point is 0x40081198
0x40081198: call_start_cpu1 at /home/xiang/esp/esp-idf/components/esp_system/port/cpu_start.c:147

I (0) cpu_start: App cpu up.
I (421) cpu_start: Pro cpu start user code
I (421) cpu_start: cpu freq: 160000000
I (421) cpu_start: Application information:
I (425) cpu_start: Project name:     gatt_client_demo
I (431) cpu_start: App version:      1
I (436) cpu_start: Compile time:     Aug 16 2023 18:44:10
I (442) cpu_start: ELF file SHA256:  edbaf5d2d28529ad...
I (448) cpu_start: ESP-IDF:          v4.4.5-286-g48e28cf3cf
I (454) cpu_start: Min chip rev:     v0.0
I (459) cpu_start: Max chip rev:     v3.99 
I (463) cpu_start: Chip rev:         v1.0
I (468) heap_init: Initializing. RAM available for dynamic allocation:
I (476) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (481) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (488) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (494) heap_init: At 3FFC6260 len 00019DA0 (103 KiB): DRAM
I (500) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (506) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (513) heap_init: At 400977F4 len 0000880C (34 KiB): IRAM
I (520) spi_flash: detected chip: gd
I (523) spi_flash: flash io: dio
W (527) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (541) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (589) BTDM_INIT: BT controller compile version [7c67f29]
I (589) system_api: Base MAC address is not set
I (589) system_api: read default base MAC address from EFUSE
I (599) BTDM_INIT: Bluetooth MAC: 30:ae:a4:21:d5:62
I (599) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1029) GATTC_DEMO: EVT 0, gattc if 3, app_id 0
I (1029) GATTC_DEMO: REG_EVT
I (1039) GATTC_DEMO: scan start success
I (1049) GATTC_DEMO: 57 4c 62 c0 30 15 
I (1049) GATTC_DEMO: searched Adv Data Len 15, Scan Response Len 0
I (1049) GATTC_DEMO: searched Device Name Len 0
I (1049) GATTC_DEMO: 

I (1089) GATTC_DEMO: 10 08 93 fc 49 f5 
I (1089) GATTC_DEMO: searched Adv Data Len 18, Scan Response Len 0
I (1099) GATTC_DEMO: searched Device Name Len 0
I (1099) GATTC_DEMO: 

I (1099) GATTC_DEMO: 0c b8 15 81 34 72 
I (1109) GATTC_DEMO: searched Adv Data Len 31, Scan Response Len 28
I (1109) GATTC_DEMO: searched Device Name Len 14
I (1119) GATTC_DEMO: ESP_GATTS_DEMO
I (1119) GATTC_DEMO: 

I (1129) GATTC_DEMO: searched device ESP_GATTS_DEMO

I (1129) GATTC_DEMO: connect to the remote device.
I (1139) GATTC_DEMO: stop scan successfully
I (1279) GATTC_DEMO: EVT 40, gattc if 3, app_id 5560
I (1279) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (1279) GATTC_DEMO: REMOTE BDA:
I (1289) GATTC_DEMO: 0c b8 15 81 34 72 
I (1289) GATTC_DEMO: EVT 2, gattc if 3, app_id 0
I (1289) GATTC_DEMO: open success
I (1479) GATTC_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
I (2609) GATTC_DEMO: EVT 46, gattc if 3, app_id 0
I (2609) GATTC_DEMO: discover service complete conn_id 0
I (2609) GATTC_DEMO: EVT 7, gattc if 3, app_id 43
I (2609) GATTC_DEMO: SEARCH RES: conn_id = 0 is primary service 1
I (2619) GATTC_DEMO: start handle 40 end handle 43 current handle value 40
I (2629) GATTC_DEMO: service found
I (2629) GATTC_DEMO: UUID16: ff
I (2629) GATTC_DEMO: EVT 6, gattc if 3, app_id 0
I (2639) GATTC_DEMO: Get service information from remote device
I (2649) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT
I (2649) GATTC_DEMO: EVT 38, gattc if 3, app_id 42
I (2659) GATTC_DEMO: ESP_GATTC_REG_FOR_NOTIFY_EVT
I (2679) GATTC_DEMO: EVT 18, gattc if 3, app_id 0
I (2689) GATTC_DEMO: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 500, conn_id 0
I (2849) GATTC_DEMO: EVT 10, gattc if 3, app_id 33045
I (2849) GATTC_DEMO: ESP_GATTC_NOTIFY_EVT, receive notify value:
I (2849) GATTC_DEMO: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 
I (2849) GATTC_DEMO: EVT 9, gattc if 3, app_id 0
I (2859) GATTC_DEMO: write descr success 
I (2919) GATTC_DEMO: EVT 4, gattc if 3, app_id 0
I (2919) GATTC_DEMO: write char success
```

zhp0406, 2023-08-18:
@ForgetCSX 
你在这里打印并不合理，esp_ble_gattc_cb_param_t 是联合体，
typedef union {
...
} esp_ble_gattc_cb_param_t;             /*!< GATT client callback parameter union type */

app_id 只存在于 reg 中（ESP_GATTC_REG_EVT）

如果此时上来的事件不是 ESP_GATTC_REG_EVT，这里打印出的 app_id 就不正确。

zhp0406, 2023-08-18:
@ForgetCSX 
如果client 连接了多个设备，想区分每一个连接，可以考虑使用： 
uint16_t conn_id;              /*!< Connection id */

ForgetCSX, 2023-08-21:
感谢您的回复，我这里还有一个疑问，按您所说使用uint16_t conn_id; /*!< Connection id */，那么需要关注那一个EVT的conn_id呢，我看esp_ble_gattc_cb_param_t里面是有许多个EVT的，并且，conn_id这个在链接的时候不确定的,根据链接的先后顺序递增，如果根据这个还得再去匹配一下mac地址。

zhp0406, 2023-08-21:
@ForgetCSX 
你想做什么效果呢？

ForgetCSX, 2023-08-21:
一个蓝牙控制器，可以链接多个蓝牙设备，并且蓝牙不是同一类型的设备，在蓝牙服务里面通过mac地址完成链接，完成链接后需要能对这些链接的设备进行读写操作，相当于这个设备是一个蓝牙HUB

zhp0406, 2023-08-22:
@ForgetCSX 
根据你的描述，应该是多连接的情况，下面的 example 是否能满足你的要求？
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gattc_multi_connect

ForgetCSX, 2023-08-25:
@zhp0406 这个例子可以满足我的要求，我已经修改好了，但是我有一个设备，他的uuid不是基本的uuid，所以我配置了128bit的uuid，获取uuid服务都是正常的，但是当我调用esp_ble_gattc_write_char发送数据的时候会显示Write char failed, error status = 87错误，如下图所示：
[image]
我的蓝牙设备信息如下，通过手机蓝牙调试助手获取：
[image]

ForgetCSX, 2023-08-25:
@zhp0406 我发现是没有notify这个properties的但是在蓝牙调试助手上是可以看到的，这个是如何设置的啊，我强行将ESP_GATT_CHAR_PROP_BIT_NOTIFY改为ESP_GATT_CHAR_PROP_BIT_WRITE_NR去通知，直接就报错了。
[image]

zhp0406, 2023-08-25:
@ForgetCSX 
[0001-fix-bt-bluedroid-Fix-uuid-length-error-when-uuid-is-.patch](https://github.com/espressif/esp-idf/files/12435865/0001-fix-bt-bluedroid-Fix-uuid-length-error-when-uuid-is-.patch)
你使用上面的 patch 进行一下测试，另外，如果是另一个问题，建议你关闭已经解决的问题，并重新提一个 issue。谢谢

ForgetCSX, 2023-08-25:
好的，感谢

zhp0406, 2023-08-25:
@ForgetCSX 
麻烦你新开一个 issue 吧，并反馈下我的 patch 是否生效。如果没有生效我就需要再检查问题。

ForgetCSX, 2023-08-25:
@zhp0406 好的,以打开新问题https://github.com/espressif/esp-idf/issues/12139

closed 2023-08-25 by ForgetCSX