espressif/esp-idf #12087: esp_console_set_vprintf feature request

joelguittet, 2023-08-16:
### Is your feature request related to a problem?

Today the esp_console is properly writing the output to stdout (generally the UART interface).
There is no simple solution to redirect the esp_console output to another handler

### Describe the solution you'd like.

Setting the output handler should be very easy. Just as `esp_log_set_vprintf` for the logs, I suggest a `esp_console_set_vprintf` to set a vprintf like function to handle the esp_console output.

### Describe alternatives you've considered.

I'm trying to make an alternative redirecting stdout but it's not working in any situation in my project for the moment.

igrr, 2023-08-16:
@joelguittet Could you describe where are you trying to redirect console I/O?

Since unlike logs, interactive line editing features of console require more than just "vprintf", implementing a "vprintf" hook wouldn't be sufficient.

If you are redirecting a socket, you can wrap the existing socket with a `FILE*` stream using `fdopen`:
```c++
stdout = fdopen(sock_fd);
stdin = fdopen(sock_fd);  // probably stdin as well, if the console is an interactive one
```
(in the task which runs the console; before launching the console.)

If this is something other than a socket (e.g. an LCD for output and a USB keyboard for input) then you probably need to implement a [VFS driver](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/vfs.html). VFS driver should implement functions like `read`, `write`, `fcntl`, `flush`, mapping them to the underlying hardware. Then you can replace stdout and stdin in a similar fasion, e.g. `stdin = freopen("/myvfs/0", "r");`

If you have doubts about either of the approaches, please describe your use case in more detail.

joelguittet, 2023-08-17:
@igrr thanks for this detailed answer.
My use case is a remote terminal. Data are transmitted over websocket using a specific protocol.
So I understand I need a VFS driver and I need to implement the wanted functions (at least read and write).
I don't see so much examples using this, any good starting point you may advice ?

igrr, 2023-08-18:
With a remote terminal, are you doing line editing remotely, or on the ESP? I.e. do you want to transfer individual keystrokes from remote to the ESP, or just send the whole command line when the user presses "Enter"?

If you are doing line editing remotely, and then calling `esp_console_run(cmdline, &ret);` to execute the command then you might not actually need the complexity of a VFS. It would be enough to simply replace stdout using the [`fwopen`](https://man.freebsd.org/cgi/man.cgi?query=fwopen&sektion=3&manpath=freebsd-release-ports) function — the output sent to the stdout would be passed on to your callback.

If you want to transfer every keystroke over a websocket and do line editing on ESP side, then you do need to implement a VFS. Indeed, we don't have such an example! The closest thing we have in IDF is probably the Bluetooth serial port profile (SPP) VFS driver in components/bt/host/bluedroid/btc/profile/std/spp/btc_spp.c. It's quite complex (because Bluetooth!), but might be still useful as a starting point.

* `open` should return a file descriptor number. If you only have one websocket connection and are only using it in one place in the application, then `return 1;` might be enough. Otherwise you have to track the context of each connection and associate it with a file descriptor.
* `read` should read the incoming data from the websocket, either in a blocking or a non-blocking way (see below). The simplest thing to do is probably to create a [ringbuffer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_additions.html#ring-buffers), push data into the ringbuffer from websocket thread, and take data from the ringbuffer in VFS `read` implementation.
* `write` should write the outgoing data to the websocket. Again, you can use a ringbuffer, or maybe even call websocket client functions directly.
* If you want `linenoiseProbe` to work, `fcntl` should be toggle between blocking and non-blocking reads. Note, `linenoiseProbe` sends an escape sequence to the terminal and expects an escape sequence back, to distinguish a VT100 capable terminal from a "dumb" one. If it doesn't get the escape sequence back within 500ms, it will assume that the terminal doesn't handle cursor positioning and other similar VT100 escape codes. Not sure if this is necessary in your application, at all. If you know that your terminal doesn't support cursor positioning VT100 codes, you can simply call `linenoiseSetDumbMode` and ignore `linenoiseProbe`. In this case, `fcntl` is not necessary, and your `read` function should be blocking.
* Optional: implement `fstat`, setting `st->st_mode = S_IFCHR` so that the FILE* stream is created without buffering. (See components/vfs/vfs_usb_serial_jtag.c for an example.)

chipweinberger, 2023-08-20:
The `esp_lineedit` component I suggested exposes  `esp_linenoise_set_stdin(...)` & `esp_linenoise_set_stdout(...)` =)

https://github.com/espressif/esp-idf/pull/10526

joelguittet, 2023-09-07:
@igrr thanks for the answer, Im back on this topic after a small pause.

Line editing is a nice to have feature and I'm interested to get it working. So I need to write a VFS according to your answer.
I understood properly the interface and it does not seems to be very complicated to write open/close/write/read function and instantiate the right VFS. I understand a path is required, let say `/dev/my-path`. Great.

Then once this is written, something I don't understand is how to do the link with the esp_console.
- How the esp_console can be used to redirect stdin and stdout with `/dev/my-path` ?
- Do I need to do something like in the semihost vfs example: `FILE *fout = freopen("/dev/my-path", "w", stdout);` ?
- What becomes the `esp_console_run` stuff ?

In the `btc_spp.c` file I don't see a link with the console...

igrr, 2023-09-18:
Hi @joelguittet, sorry for the late reply. I made an example of a console over websocket client here: https://github.com/igrr/esp-protocols/pull/1. For the VFS driver, please take a look at `websocket_client_vfs.c`, feel free to ask questions if the idea is not clear.

closed 2023-10-31 by igrr

chipweinberger, 2023-10-31:
@igrr 

that would be a good example to add to the esp-idf repo imo

joelguittet, 2023-11-02:
@igrr yes no worries to close this at the moment I have not work again on this topic yet.
Thanks again for your answers and the example, should be good 👍