espressif/esp-idf #12091: Wifi throughput issue (https file download) with huge waits from stream->available()

0x0fe, 2023-08-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32D0WDR2

### Power Supply used.

External 5V

### What is the expected behavior?

A consistent (and decent) throughput when downloading files from HTTP and HTTPS.

### What is the actual behavior?

A very inconsistent throughpout  during file download, ranging from 20KB/s to 980KB/s, totally randomly.

```
0	16KB	2ms
1	0KB	1ms
1	16KB	22ms
1	0KB	0ms
1	1KB	3ms
28087	16KB	19ms
1	1KB	1ms
1	1KB	2ms
283	16KB	24ms
1	2KB	0ms
1	1KB	2ms
161	16KB	28ms
1	3KB	2ms
1	1KB	2ms
151	16KB	25ms
1	4KB	2ms
1	1KB	3ms
113	16KB	29ms
1	5KB	2ms
1	1KB	4ms
95	16KB	26ms
1	6KB	1ms
1	1KB	2ms
17	16KB	29ms
1	7KB	2ms
1	1KB	5ms
1	16KB	34ms
1	8KB	2ms
1	1KB	5ms
1	16KB	37ms
1	9KB	3ms
[... 3300 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=64
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT is not set
# CONFIG_SUPPORT_TERMIOS is not set
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
```
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/12371474/sdkconfig.zip)

igrr, 2023-08-17:
Hi @0x0fe,
Just to exclude hardware or environment issue, could you try the attached IDF example on your hardware? It downloads a file and writes it to an SD card. It has consistent throughput of about 1.5 MBytes/s.
[https-download-demo.zip](https://github.com/espressif/esp-idf/files/12372061/https-download-demo.zip)

Note, i've edited your issue to put the very long debug logs into [collapsible sections](https://gist.github.com/pierrejoubert73/902cc94d79424356a8d20be2b382e1ab). Kindly do the same in the future, or otherwise attach long logs as files!

0x0fe, 2023-08-17:
@igrr sure, i am going to run this, on the test board i dont have the SD_MMC installed, so maybe i'll have to comment that part. 
Thank you for editing the logs, i'll remind to use these collapsible sections.

0x0fe, 2023-08-17:
with IDF 5.0, something wrong with wifi init.

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7296
load:0x40078000,len:16484
ho 0 tail 12 room 4
load:0x40080400,len:4260
entry 0x40080668
[0;32mI (29) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 19:32:40[0m
[0;32mI (30) boot: chip revision: v3.1[0m
[0;32mI (33) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (39) boot.esp32: SPI Speed      : 80MHz[0m
[0;32mI (43) boot.esp32: SPI Mode       : QIO[0m
[0;32mI (48) boot.esp32: SPI Flash Size : 2MB[0m
[0;32mI (52) boot: Enabling RNG early entropy source...[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (84) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (91) boot: End of partition table[0m
[0;32mI (95) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=344ach (214188) map[0m
[0;32mI (163) esp_image: segment 1: paddr=000444d4 vaddr=3ffb0000 size=033bch ( 13244) load[0m
[0;32mI (167) esp_image: segment 2: paddr=00047898 vaddr=40080000 size=08780h ( 34688) load[0m
[0;32mI (180) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=8fc84h (588932) map[0m
[0;32mI (342) esp_image: segment 4: paddr=000dfcac vaddr=40088780 size=0fd1ch ( 64796) load[0m
[0;32mI (375) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (375) boot: Disabling RNG early entropy source...[0m
[0;32mI (387) cpu_start: Pro cpu up.[0m
[0;32mI (387) cpu_start: Starting app cpu, entry point is 0x40081658[0m
[0;32mI (372) cpu_start: App cpu up.[0m
[0;32mI (401) cpu_start: Pro cpu start user code[0m
[0;32mI (401) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (401) cpu_start: Application information:[0m
[0;32mI (406) cpu_start: Project name:     https-download-demo[0m
[0;32mI (412) cpu_start: App version:      1[0m
[0;32mI (416) cpu_start: Compile time:     Aug 17 2023 19:31:57[0m
[0;32mI (422) cpu_start: ELF file SHA256:  d8dffe1b3def59d5...[0m
[0;32mI (428) cpu_start: ESP-IDF:          v5.0.1-dirty[0m
[0;32mI (434) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (439) cpu_start: Max chip rev:     v3.99 [0m
[0;32mI (443) cpu_start: Chip rev:         v3.1[0m
[0;32mI (448) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (455) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (461) heap_init: At 3FFB7760 len 000288A0 (162 KiB): DRAM[0m
[0;32mI (468) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (474) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (480) heap_init: At 4009849C len 00007B64 (30 KiB): IRAM[0m
[0;32mI (487) spi_flash: detected chip: generic[0m
[0;32mI (491) spi_flash: flash io: qio[0m
[0;33mW (495) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (509) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (524) example_connect: Start example_connect.[0m
W (529) wifi:wifi osi_nvs_open fail ret=4353

[0;32mI (533) wifi_init: rx ba win: 6[0m
[0;32mI (537) wifi_init: tcpip mbox: 64[0m
[0;32mI (541) wifi_init: udp mbox: 64[0m
[0;32mI (545) wifi_init: tcp mbox: 64[0m
[0;32mI (548) wifi_init: tcp tx win: 65534[0m
[0;32mI (553) wifi_init: tcp rx win: 65534[0m
[0;32mI (557) wifi_init: tcp mss: 1440[0m
[0;32mI (561) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (565) wifi_init: WiFi RX IRAM OP enabled[0m
[0;32mI (570) wifi_init: LWIP IRAM OP enabled[0m
ESP_ERROR_CHECK failed: esp_err_t 0x1101 (ESP_ERR_NVS_NOT_INITIALIZED) at 0x400d7cdb
file: "/IDF/examples/common_components/protocol_examples_common/wifi_connect.c" line 138
func: example_wifi_start
expression: esp_wifi_init(&cfg)

abort() was called at PC 0x40089517 on core 0

Backtrace: 0x40082126:0x3ffb96c0 0x40089521:0x3ffb96e0 0x40092c96:0x3ffb9700 0x40089517:0x3ffb9770 0x400d7cdb:0x3ffb97a0 0x400d7ffa:0x3ffb98b0 0x400d7933:0x3ffb99f0 0x400d7213:0x3ffb9a10 0x400d76fb:0x3ffb9a30 0x40158eeb:0x3ffb9a60 0x4008c6b1:0x3ffb9a90
```

0x0fe, 2023-08-17:
so, i added my own wifi_init, and  disabled the writing to SD card. Indeed the dowwnload speed is 1.8MB/s.

```
(1898) wifi:AP's beacon interval = 102400 us, DTIM period = 3
(2880) esp_netif_handlers: sta ip: 192.168.1.81, mask: 255.255.255.0, gw: 192.168.1.1
(2882) example: got ip:192.168.1.81
(2884) example: connected to ap SSID:xxxxx password:xxxxxxx
(2890) example: Downloading file from http://cachefly.cachefly.net/10mb.test with buffer size 49152
(3033) example: HTTP_EVENT_ON_CONNECTED
(3037) example: HTTP_EVENT_HEADER_SENT
(3109) example: Content-length: 10485760
(8516) example: HTTP_EVENT_ON_FINISH
(8516) example: HTTP Status = 200, content_length = 10485760
(8517) example: Time taken: 5614 ms Speed: 1823.81 kB/sec
(8522) example: Download task spent 0 ms blocked on writing to ringbuffer
(8530) example: File write task spent 0 ms blocked on writing to SD car[0m
(13514) example: Failed to read from ringbuffer
```

0x0fe, 2023-08-22:
hello, any update on this? any clue as for why the underlaying C http server presents normal performance while the higher level CPP wrapper has very poor performance?

0x0fe, 2023-08-23:
@igrr 

> HTTP client library isn't a wrapper around an IDF one

Ok, i thought it was, anyway bottom line is : the Arduino HTTP client performance is poor, compared to the IDF C client which reaches 2MB on good days, versus 60KB with wind in the back for the Arduino HTTP client. 
Is there an alternative (async client or something else)? i feel that SDKconfig parameters are already pushed far. There must be some fundamental différence in the implémentation impeding performance.

0x0fe, 2023-09-29:
@igrr 
it turned out i have an issue with this, the download speed indicated is not real, because it actually fails early. I added the writing to SD card, and i cannot get this test to work properly.
https://github.com/espressif/esp-idf/issues/12328

closed 2024-08-09 by Sherry616