espressif/esp-idf #12092: cargo generate esp-rs/esp-idf-template cargo not building default project

olaoni, 2023-08-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.5

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3-DEV KIT

### Power Supply used.

USB

### What is the expected behavior?

I expect the project to build.

### What is the actual behavior?

It fails 

### Steps to reproduce.

Installed git and python3
Step 1:
rustup toolchain install nightly --component rust-src

Step 2:
rustup target add riscv32imc-unknown-none-elf 

Step 3:
cargo install espup

Step 4:
espup install

Step 5:
cat $HOME/export-esp.sh >> ~/.bashrc
source ~/.bashrc

Step 6
cargo install ldproxy

### Debug Logs.

```plain
onio@ecclesiastes:~/esp$ cargo build
error: could not find `Cargo.toml` in `/home/onio/esp` or any parent directory
onio@ecclesiastes:~/esp$ cd hello
onio@ecclesiastes:~/esp/hello$ 
onio@ecclesiastes:~/esp/hello$ 
onio@ecclesiastes:~/esp/hello$ cargo build
    Updating crates.io index
[... 348 lines cut ...]
                       at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panicking.rs:487:40
                std::panicking::try
                       at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panicking.rs:451:19
                std::panic::catch_unwind
                       at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panic.rs:140:14
                std::rt::lang_start_internal::{{closure}}
                       at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/rt.rs:148:48
                std::panicking::try::do_call
                       at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panicking.rs:487:40
                std::panicking::try
                       at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panicking.rs:451:19
                std::panic::catch_unwind
                       at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panic.rs:140:14
                std::rt::lang_start_internal
                       at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/rt.rs:148:20
             4: main
             5: __libc_start_call_main
                       at ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
             6: __libc_start_main_impl
                       at ./csu/../csu/libc-start.c:392:3
             7: _start
          

error: could not compile `hello` (bin "hello") due to previous error
```

igrr, 2023-08-17:
I'm not sure this is a bug specifically in IDF. The error indicates that the linker was given object files with incompatible version of RISC-V integer ISA.

esp-2021r2-patch5-8.4.0 GCC based toolchain used in IDF v4.4.x uses binutils 2.35.1, which produces this arch tag: rv32i**2p0**_m2p0_a2p0_f2p0_d2p0_c2p0.

Newer binutils releases produce an arch tag with a different "I" version (2.1 instead of 2.0): rv32i**2p1**_m2p0_c2p0_zicsr2p0_zifencei2p0.

My guess is that Rust compiler you are using also generates code with something like rv32i**2p1**_m2p0_c2p0, which is incompatible with object files from IDF which are built with rv32i**2p0**_m2p0_a2p0_f2p0_d2p0_c2p0.

Just my understanding of the issue so far. I don't yet have an idea where/how it should be fixed. As far as IDF is concerned, release/v4.4 has always been generating code with rv32i2p0, no changes have occurred.

olaoni, 2023-08-17:
Hi igrr,
Thank you for your speedy reply. 

Any idea on a work around? 

Regards
onio

olaoni, 2023-08-19:
Hi All,

I thought I should share this information just in case someone might have the same issue and stumble on to this page.

After many days of trying to get the default esp-idf-template project to build. I have manage to get a simple blinky program working. 

Thanks to [Hutscape](https://hutscape.com/tutorials/blinky-rust-esp32c3) page that showed a work around on how to create a project that builds.

Just to re-iterate when the project is created using the recommended esp-idf-template command as below

`cargo generate esp-rs/esp-template` 

The project is created successfully, but the problem occurs when you issue the command to build for example

`cargo build`

This command fails as indicated in my first message above. 
In order to create a project that will build, I found Hutscape page which used the following command

`cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template cargo`

Creating the project with this form allowed the project to build. 

Below is the source code and Cargo.toml for the blinky program.

**Cargo.toml**
`[package]
name = "blinky"
version = "0.1.0"
authors = ["Ola Oni <oniola@gmail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.66"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]

default = ["std", "hal", "esp-idf-sys/native"]

pio = ["esp-idf-sys/pio"]
all = ["std", "nightly", "experimental", "embassy"]
hal = ["esp-idf-hal", "embedded-svc", "esp-idf-svc"]
std = ["alloc", "esp-idf-sys/std", "esp-idf-sys/binstart", "embedded-svc?/std", "esp-idf-hal?/std", "esp-idf-svc?/std"]
alloc = ["embedded-svc?/alloc", "esp-idf-hal?/alloc", "esp-idf-svc?/alloc"]
nightly = ["embedded-svc?/nightly", "esp-idf-svc?/nightly"] # Future: "esp-idf-hal?/nightly"
experimental = ["embedded-svc?/experimental", "esp-idf-svc?/experimental"]
embassy = ["esp-idf-hal?/embassy-sync", "esp-idf-hal?/critical-section", "esp-idf-hal?/edge-executor", "esp-idf-svc?/embassy-time-driver", "esp-idf-svc?/embassy-time-isr-queue"]

[dependencies]
#anyhow = "=1.0.71"
log = { version = "0.4.17", default-features = false }
esp-idf-sys = { version = "0.33", default-features = false }
esp-idf-hal = { version = "0.41", optional = true, default-features = false }
esp-idf-svc = { version = "0.46", optional = true, default-features = false }
embedded-svc = { version = "0.25", optional = true, default-features = false }

[build-dependencies]
embuild = "0.31.2"
`
**src/main.rs**
`use esp_idf_hal::gpio::*;
use esp_idf_hal::peripherals::Peripherals;

use esp_idf_sys as _;

fn main() {
    esp_idf_sys::link_patches();

    let peripherals = Peripherals::take().unwrap();

    let mut led = PinDriver::output(peripherals.pins.gpio8).unwrap();

    println!("led blinky!");

    loop {
        led.set_high().unwrap();
        std::thread::sleep(std::time::Duration::from_secs(1));
        led.set_low().unwrap();
        std::thread::sleep(std::time::Duration::from_secs(1));
    }
}
`

closed 2023-08-19 by olaoni