espressif/esp-idf #12094: [ESP-IDF MQTT] repeated disconnection of the device. by "No PING_RESP, disconnected"

eric-Kiaofarming, 2023-08-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

framework-espidf @ 3.50002.230601

### Operating System used.

macOS

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

toolchain-xtensa-esp32 @ 11.2.0+2022r1

### Power Supply used.

External 3.3V

### What is the expected behavior?

The device is always online,
And can continue to send and receive data.

### What is the actual behavior?

After several times of sending and receiving Ping actions, there is a possibility that the device may not receive a response and cause the device to be disconnected and reconnected. Reducing the keepalive time can effectively increase the probability of occurrence.

```
        sprintf(mqtt_borker_uri,"mqtt://%s:%d",host,port);

        esp_mqtt_client_config_t mqtt_cfg = {
                .session.protocol_ver = MQTT_PROTOCOL_V_3_1_1,
                .session = {
                        .keepalive = 3,
                        .disable_keepalive = false,
                        .last_will = {
                                .topic = _INIT,
                                .retain = true,
                                .msg = "",
                                .msg_len = 0,
                                .qos = 1,
                        },
                },
        };

        while(ESP_WIFI_CONNECTED == false) {
                 vTaskDelay(50 / portTICK_RATE_MS);
        }

        esp_mqtt_client_handle_t _client = esp_mqtt_client_init(&mqtt_cfg);

        esp_mqtt_client_set_uri(_client, mqtt_borker_uri);
        esp_mqtt_client_register_event(_client, ESP_EVENT_ANY_ID,mqtt_event_handler,NULL);

        esp_mqtt_client_start(_client);
```

### Steps to reproduce.
[... 136 lines cut ...]
I (41) boot.esp32: SPI Flash Size : 8MB
I (46) boot: Enabling RNG early entropy source...
I (516) boot: Loaded app from partition at offset 0x10000
I (516) boot: Disabling RNG early entropy source...
I (528) cpu_start: Pro cpu up.
I (528) cpu_start: Starting app cpu, entry point is 0x40081be0
I (0) cpu_start: App cpu up.
D (536) clk: RTC_SLOW_CLK calibration value: 3321498
I (549) cpu_start: Pro cpu start user code
I (549) cpu_start: cpu freq: 160000000 Hz
I (550) cpu_start: Application information:
I (554) cpu_start: Project name:     controller
I (559) cpu_start: App version:      18b9876-dirty
I (565) cpu_start: Compile time:     Aug 17 2023 13:56:24
I (571) cpu_start: ELF file SHA256:  81f5676df903c66e...
I (577) cpu_start: ESP-IDF:          5.0.2
I (582) cpu_start: Min chip rev:     v0.0
I (586) cpu_start: Max chip rev:     v3.99 
I (591) cpu_start: Chip rev:         v3.0
```
```

### More Information.

My mqtt broker is mosquitto built on the intranet

euripedesrocha, 2023-08-28:
@eric-Kiaofarming the behavior you are describing is how the keep alive mechanism is supposed to work, refer to [mqtt specification](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc385349238). 
Reducing the time, you likely increase the disconnection probability, depending on your network situation.

eric-Kiaofarming, 2023-08-29:
@euripedesrocha  I tested 13 to 30 seconds and had the same problem. I use 3 seconds to speed up the frequency of occurrences.

euripedesrocha, 2023-09-05:
@eric-Kiaofarming the behavior you described is the expected behavior for MQTT keep alive. You can [disable the keep alive](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html#_CPPv4N24esp_mqtt_client_config_t9session_t17disable_keepaliveE) for the MQTT client If you want to, but the disconnection may happen from the broker side. I suggest that you investigate the adequate value for your system, considering the possible network scenarios. For reference, we set by default the keep alive to 120s, which will make an attempt every 60s.

closed 2023-09-05 by euripedesrocha

eric-Kiaofarming, 2023-09-05:
@euripedesrocha 
Thanks for your reply, I've tried setting 120s and still have this problem.
I have tried disabling keep alive. It works but cannot automatically set offline on mqtt server. I need to create a process to detect it, when the device is offline.

From the network packets I captured, the keepalive problem mostly occurs when "ESP32 inserts other send out (Data or TCP Ack) packets while waiting for ping response", is there any way to prohibit other work intervene while waiting for ping response?