espressif/esp-idf #12095: BluFi配网严重bug

jiafeng-1, 2023-08-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-c3-Mimi-1

### Power Supply used.

USB

### What is the expected behavior?

**配网正常使用的版本为v4.4.2**
BlUFI Version：
I (612) BLUFI_EXAMPLE: BLUFI VERSION 0102

代码为BluFi 示例
EspBluFi app版本1.3.1、app BluFi库版本为2.1.0

## IOS配网
IOS使用EspBluFi应该能够正常配网，ios页面应该显示
```bash
Connected device
Bulufi Connection has prepared
Post configure params complete
Reveive device status:
OpMode:Station
station connect wifi_now
connect wifi bssid：xxxx
connect wifi ssdi：xxxx
```
设备端应该显示
```bash
I (504092) BLUFI_EXAMPLE: BLUFI Set WIFI opmode 1

I (504182) BLUFI_EXAMPLE: Recv STA SSID radar

I (504252) BLUFI_EXAMPLE: Recv STA PASSWORD racobit123

I (504332) BLUFI_EXAMPLE: BLUFI requset wifi connect to AP

I (504342) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (504342) wifi:state: init -> auth (b0)
I (504342) wifi:state: auth -> assoc (0)
I (504352) wifi:state: assoc -> run (10)
I (504462) wifi:connected with radar, aid = 4, channel 1, 40U, bssid = 68:77:24:19:6a:90
I (504462) wifi:security: WPA2-PSK, phy: bgn, rssi: -18
I (504472) wifi:pm start, type: 1

I (504472) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
[... 685 lines cut ...]

D (54760) BT_HCI: btu_hcif_command_complete_evt
```

### Steps to reproduce.

只需运行V4.4.4版本BluFi示例即可发现问题
怀疑是ESP32运行的BluFi库版本问题，0102运行正常，0103运行不正常

### More Information.

我遇到了相同的问题，Android配网没有问题，但是IOS设备不可以。尝试了大概两三台iPhone手机都是不可以。Android都可以
使用的手机型号：iPhone 8
系统版本：16.6

另一台手机型号：iPhone 12

系统版本：16.2

EspBlufi 软件版本：1.3.1

ESP32 idf版本： ESP-IDF v4.4.4-dirty
使用的官方提供的Blufi例程，没有修改menuconfig

@methodmissing @svenfuchs @paulreimer @kumekay @

zhp0406, 2023-08-18:
@jiafeng-1 
master 分支能够正常配网吗？

jiafeng-1, 2023-08-18:
> @jiafeng-1 master 分支能够正常配网吗？

master分支并没有尝试，只测试了4.4.2和4.4.4.

lancechan00, 2023-08-18:
在上面文档中，我找到了以下内容：

标题：ESP32-C3-WROOM-02 can't connect to Wi-Fi

我可以看到设备的名字是 ESP32-C3-WROOM-02。

这个名字共有 17 个字符。

我一直以来使用blufi，发现当设备名称设置成16个字符的时候，我ios设备搜索不到，安卓没问题。当我把设备名称设置成12个字符，这个问题就没有了。

zhp0406, 2023-08-18:
@lancechan00 

广播数据的最大长度为31字节。如果广播数据过长，底层会进行截断，可能会导致意料之外的错误。在blufi示例中，广播数据包含以下内容：

AD类型：标志（占用3字节）
AD类型：发射功率水平（占用3字节）
AD类型：完整的16位UUID列表（占用4字节）
AD类型：外围连接间隔范围（占用6字节）
AD类型：完整的本地名称（剩余15字节可供使用）
其中，"完整的本地名称"指设备名称，其它数据类型共使用16字节。考虑到AD类型占用一个字节，可以得出结论：
设备的名称不应超过14字节，即不能超过14个字符。

如果有必要使用更长的设备名称，可以考虑将设备名称放入扫描响应数据中。如果不清除怎么用扫描响应数据完成长设备名设置，请继续@我。

@jiafeng-1 
你的设备名字设置的合理吗？

jiafeng-1, 2023-08-18:
用的官方demo，什么都没修改的情况下，就有这个问题。
设备名称显示的都是正常的，用的`esp_ble_gap_set_device_name()`设置的名字，字符在16个左右。
我这个不是名字问题，能被扫描到，单纯的连接的时候ios自动断链。目前的解决办法是降版本为4.4.2 @zhp0406

closed 2023-11-09 by Alvin1Zhang