espressif/esp-idf #12096: Upgraded esptool.py to 4.6.2 but IDF requested 4.5.1

eme-bennytang, 2023-08-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

idf.py should have set-target and rebuild sdk.config for me

### What is the actual behavior?

Prompt error saying the required esptool.py version should 4.5.0, but I have to upgrade it to fix a build issue with 
- ESP32S3-WROOMN16R8
- Secure boot enabled
So far the esptool.py upgrade did fix the problem for me, I am able to flash application now by either VS CODE IDE/esptool.py write_flash directly

But I can't use idf.py anymore

### Steps to reproduce.

Upgraded esptool.py to 4.6.2 to fix this issue
https://github.com/espressif/esp-idf/issues/10788

### Build or installation Logs.

```plain
idf.py set-target esp32s3
/home/benny/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
The following Python requirements are not satisfied:
esptool~=4.5.0 - (esptool 4.6.2 (/home/benny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages), Requirement.parse('esptool~=4.5.0'))
To install the missing packages, please run "install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/benny/.espressif/python_env/idf5.0_py3.10_env
    Python interpreter used: /home/benny/.espressif/python_env/idf5.0_py3.10_env/bin/python
Constraint file: /home/benny/.espressif/espidf.constraints.v5.0.txt
Requirement files:
 - /home/benny/esp/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/benny/.espressif/python_env/idf5.0_py3.10_env/bin/python

ESP-IDF v5.0.2
```

dobairoland, 2023-08-18:
Hi @eme-bennytang. `esptool~=4.5.0` means that the version should be at least 4.5.0 but could not be 5.0 or newer. That means 4.6.2 should be an acceptable version.

Could you please use the Doctor command to generate a status report about your environment?

eme-bennytang, 2023-08-18:
Hi Roland, 

Thanks for the reply, but I have upgraded my IDF version to release/v5.1 to cope with this problem.
Just for future reference, should I just execute ./install.sh to upgrade my tools (e.g. esptool.py) or should I "pip install esptool==4.6.2"? I am guessing this might be the problem

dobairoland, 2023-08-18:
> Just for future reference, should I just execute ./install.sh to upgrade my tools

Yes, this is the recommended way for ESP-IDF. In case this is not working with the VScode plugin then please report it in https://github.com/espressif/vscode-esp-idf-extension.

> or should I "pip install esptool==4.6.2"? I am guessing this might be the problem

Yes, this can be a workaround if the above is not working.

So I guess we can close this for now.

dobairoland, 2023-08-18:
Oh, correction: `esptool~=4.5.0` means  `4.5.0 <= x < 4.6`. I reopen this issue and correct it.

Thank you @fhrbata for pointing out my mistake!

dobairoland, 2023-08-18:
No, I was confused. I thought that the newest esptool is included with ESP-IDF v5.0.x as well. I'll correct this mistake and after that I close this issue. I'm sorry for the confusion.

eme-bennytang, 2023-08-18:
so you do not expect the IDF 5.0.2 using only esptool 4.5.0 - 4.6.0 and will fix and include latest esptool version in 5.0.x, so that others can just upgrade the esptool by calling ./install.sh?

Thanks anyway, I should have no issue using IDF release/v5.1

dobairoland, 2023-08-18:
> so you do not expect the IDF 5.0.2 using only esptool 4.5.0 - 4.6.0 and will fix and include latest esptool version in 5.0.x, so that others can just upgrade the esptool by calling ./install.sh?

Yes, that is correct.

closed 2023-08-18 by dobairoland