espressif/esp-idf #12098: Are the ESP-IDF peripherals API for SPI and I2C compatible with interrupt reading?

herculanodavi, 2023-08-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is it possible to have a performant interrupt read with I2C and SPI? 

By reading the ESP32/ESP32S3 technical references, I see that this could be done by handling registers, but I hoped that there would be a way to do that with the driver/HAL/LL libraries from ESP-IDF. My expectation and the reality (as I currently understand) are as follows:

**EXPECTATION:** When recording data from a sensor, I'd like to do this interrupt read pattern:

1. mcu waits for sensor
2. sensor sets interrupt line high 
3. mcu gpio interrupt triggers start of I2C/SPI transaction
4. end of transaction interrupt triggers data handling by a task

**REALITY:** I can't do that with the peripherals API. What I can do is:

1. have a dedicated task, t1, to perform transactions when notified (waiting with `ulTaskNotifyTake`)
1. mcu waits for sensor
2. sensor sets interrupt line high 
3. mcu gpio interrupt notifies t1 (`vTaskNotifyGiveFromISR`)
4. t1 runs blocking and triggers data handling when it's done

which is not as performant as the expectation.

**With I2C,** the only way to start a transaction is with `i2c_master_cmd_begin(...)`, but it is always blocking:
> Send all the queued commands on the I2C bus, in master mode. The task will be blocked until all the commands have been sent out. The I2C port is protected by mutex, so this function is thread-safe. This function shall only be called in I2C master mode.

**With SPI,** I tried calling `spi_device_queue_trans` from the GPIO interrupt while calling `spi_device_get_trans_result` from the data handling task, but calling `spi_device_get_trans_result` before the first `spi_device_queue_trans` gives out an error ([example code here](https://github.com/herculanodavi/interrupt-read-esp32.git)).

Is there anything I can do to make this work?

o-marshmallow, 2023-08-22:
Hello @herculanodavi ,

If I understand correctly, you want the minimum latency between the moment the external sensor pulls the GPIO high and the master (ESP32/S3) initiates a transfer.

The way you did is the proper way to do: have a Task that waits on a notification, have a GPIO interrupt handler that notifies that Task, which is scheduled right after and will initiate the I2C transfer. As you know, the I2C transfers on ESP-IDF are blocking, but this would not help since that would not make the transfer happen any earlier.

One thing you can make sure of is that the I2C/SPI Task has a higher priority than the other tasks.

If you want something more efficient, with less latency, I can think of the following:
* Use the ULP (low-power core), as a co-processor that will handle the sensor GPIO and initiate the I2C transfers, the received data will ened to be saved in RTC RAM, so that it can be accessed by the main core too. Since it doesn't have any RTOS on it, it's just bare-metal, you will have no latency.
* Use the I2C HAL/LL layer directly within your GPIO interrupt handler so that you can schedule and initiate a transfer from the interrupt handler. If you choose this route, do not use the default ESP-IDF I2C driver, else you may encounter some incompatible behaviors (because the driver may expect the I2C hardware registers to be in a certain state when making a transfer)
* Implement a high priority interrupt in assembly. You may not have access to the whole IDF API here, so you might want to notify your task with something simple like a shared variable: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/hlinterrupts.html
* Implement a high priority interrupt in assembly which will prepare the I2C hardware registers to initiate a transfer

herculanodavi, 2023-08-24:
Hi @o-marshmallow,

What I really want is something similar to what you said -- I'd like to offload as much work from the MCU as possible to the peripherals. Maybe I'm wrong, but the blocking API calls spend a lot of CPU cycles constantly checking if the operation is done. 

There's are no big issues with the current latency, but I want to perform calculations (like doing DSP with a ping-pong buffer) while performing the transactions and this polling behaviour generates a bottleneck.

I've seen that for I2C there is this new driver API, [I2C Master](https://github.com/espressif/esp-idf/blob/master/components/driver/i2c/include/driver/i2c_master.h). This would be more in the path of what I wanted to achieve. I see it was not released in version 5.1.1, so would it be released in 5.1.2 or only in 5.2?

As for SPI, would you have any suggestions? I tried doing this but it fails since I'm calling `spi_device_get_trans_result` first. I hoped that I could use this API in a similar way than FreeRTOS' Queue Receive/Send.

```c
typedef struct transaction_args {
  spi_device_handle_t spi;
  spi_transaction_t *trans;
} transaction_args_t;

static void IRAM_ATTR gpio_isr_handler(void *arg) {
  transaction_args_t *trans_args = (transaction_args_t *)arg;
  ESP_ERROR_CHECK(
      spi_device_queue_trans(trans_args->spi, trans_args->trans, 0));
}

void data_handler_task(void *args) {
  transaction_args_t *trans_args = (transaction_args_t *)args;
  spi_transaction_t *received_trans_args = NULL;

  for (;;) {
    spi_device_get_trans_result(trans_args->spi, &received_trans_args,
                                portMAX_DELAY);
  }
}
```

ammaree, 2023-08-26:
@o-marshmallow 
@igrr 

I have a very similar situation as the above, except mine is (currently) I2C specific.
We have 16x GPIO inputs on 2x PCF8574 I2C devices, with the PCF8574 IRQ pins tied to GPIO's configured for interrupts on the falling edge only. But due to the nature of the PCF8574 it generates interrupt for both 1->0 as well as 0->1 changes. Unfortunately. determine which of the 2 cases the interrupt is for, we have to read the device for both interrupts.

In our GPIO interrupt handler we add an entry into a queue that feeds a dedicated I2C task. Majority of the time the I2C task can cope with the incoming requests, but when the water meters are busy we run into problems with the I2C task not being able to handle the requests, we suspect due to the blocking nature.

Very recently we noticed the message suggesting we change to the new I2C driver mentioned above but we have not been able to find any documentation on this new driver anywhere. We had a look at the source code and the API's are very different, far simpler than the current/old driver and APIs.

Before we start the reasonably complex process of rewriting our I2C support subsystem it will be of great value if we can have a detailed writeup on the new driver and API, including....
- How do the 2 drivers architecturally compare?
- What problems in the old/current driver is fixed in the new driver?
- How do the 2 drivers compare form a performance/blocking point of view?
- Are the I2C peripherals in the ESP32[S2/S3] exactly the same that the drivers will behave exactly the same or is there a recommended rule for choosing the driver based on the ESP32 version?

Your soonest assistance with information addressing the above would be highly appreciated.

Thanks

o-marshmallow, 2023-08-28:
Hello @herculanodavi ,

> the blocking API calls spend a lot of CPU cycles constantly checking if the operation is done.

This isn't true, the CPU will not poll the I2C peripheral until it is finished. It will prepare and start the I2C transfer and then wait for a a completion interrupt from the I2C peripheral to occur to resume the Task's code. You can see this here: https://github.com/espressif/esp-idf/blob/master/components/driver/i2c/i2c.c#L1542

While this Task is waiting for the I2C to be completed, FreeRTOS will schedule other Tasks of your system, if none is ready, Idle task will run. In all cases, there will be no polling and so your CPU will never "waste" time or cycles waiting for the I2C transfer to finish.

The new I2C driver will offer the possibility to have asynchronous I2C transfers, which means that the I2C start transaction will not block and will let you continue executing the code of the **current Task**. I insist on "the current Task": if you currently have a Task that only performs I2C transfers and another one that does calculations unrelated to I2C, having the possibility to have asynchronous I2C transfers won't be more useful in your case since FreeRTOS can already schedule the secodn task while the I2C one is waiting (as I said above).

Regarding SPI, `spi_device_queue_trans` cannot be called from an ISR. The solutions are the same as the one listed above for the I2C. For example, you can have a Task of higher priority than your main Task that will wait for a notification from your SPI interrupt handler, as soon as it is schedule by FreeRTOS, you can start a new transfer.

o-marshmallow, 2023-08-28:
Hello @ammaree,

Do you know at which point your program starts having problem? Does your queue become full at one point? Does the Task that initiates the I2C transfer take too much time to process the queue? Does the Task that initiates the I2C transfer get seldom shceduled by FreeRTOS?

In your case, it seems to me that you don't need a queue, but simply a notification. If you get an interrupt on your GPIO, it means that the state of a pin has changed, so you notify the I2C Task that a transfer is required. It will be scheduled by FreeRTOS as soon as you call the `portYIELD_FROM_ISR()` macro after `xTaskNotifyFromISR`  (check the last parameter).
If you still want to keep track of the number of notifications to your ISR Task (e.g. three GPIO interrupts must trigger three I2C transfer), you can use a counting semaphore.

The new I2C driver was written from scratch, it is more efficient than the former one:
* Simpler API
* Supports synchronous and **asynchronous** transfers (user callback called upon transfer completion)
* No more dynamic memory allocations for synchronous transfers
* Takes advantage of the hardware command list to have the least amount of interrupts

Overall, it is more performant, requires less CPU power to initiate and perform transfers and gets interrupt less frequently by filling the hardware commands beforehand.

herculanodavi, 2023-08-28:
Thanks @o-marshmallow for all the info. The issue can be closed from my side, but I'm not sure about @ammaree.

ammaree, 2023-08-28:
@o-marshmallow
We have over the weekend burned the midnight oil and converted our I2C task and supporting routines to the new I2C driver. So far testing is not positive. We can initialise the bus and send a General Reset, but probing addresses 0x8 to 0x78 ALL return ESP_OK on a platform that has a single device at 0x20.

Regarding the proposal to use an event flag, not sure how workable this is. ON most of our hardware platforms we have multiple I2C devices. The majority of those are NOT IRQ driven but polled and hence use the mechanism to add entries into the queue, each sensor or actuator on its own predetermined schedule.

Adding event flag functionality and getting the task to wait on either of:
a) an item from the queue (current scenario); and/or
b) an event from the PCF8574 
seems like complicating since, not even sure if FreeRTOS can be configured to simultaneously wait for either of those?

If we can resolve the i2c_master_probe() problem that will help a lot towards testing.

o-marshmallow, 2023-08-30:
@ammaree 

> but probing addresses 0x8 to 0x78 ALL return ESP_OK on a platform that has a single device at 0x20.

That sounds like a bug from the new driver, it must be reported. Would you mind creating a new issue for that?

If I understand your situation, you have several I2C devices that msut be polled at their own pace, so you have a FIFO where you enqueue the devices to read. Then the PCF8574 is not related to these devices, right? It's just yet another I2C device that would generate an interrupt where you enqueue its I2C device/address.

FreeRTOS doesn't have the possibility to wait on several "objects" at the same time, you still need two Tasks for that.

ammaree, 2023-08-30:
@o-marshmallow 
@igrr 
I have created a new issue with basic example code.

If somebody can please ASAP have a look and fix, we can proceed with testing the new driver.

Right now we are at a standstill....

ammaree, 2023-08-30:
This Issue can be closed from our side, provided new issue gets some attention please

closed 2023-08-30 by herculanodavi