espressif/esp-idf #12099: LP GPIO IRQ don't work when LP_IO_LP_GPIO0_INT_TYPE enabled

DexiangYang, 2023-08-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1449-gd2471b11e7-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

when LP_IO_LP_GPIO0_INT_TYPE be set to 3, LP_IO_LP_GPIO_STATUS_INTERRUPT will be (1 << 0), after GPIO 0 has been connected to GND then pull up.

### What is the actual behavior?

when LP_IO_LP_GPIO0_INT_TYPE enabled, LP_IO_LP_GPIO_STATUS_INTERRUPT still is 0, after GPIO 0 has been connected to GND, How to use LP GPIO IRQ.

### Steps to reproduce.

CONNECT GPIO 0 to GND then pull up

songruo, 2023-08-21:
Hi @DexiangYang,

There is no IRQ for LP_IO peripheral. May I ask what you are trying to achieve? Why can't you use the GPIO interrupt for the IO0? The LP_IO interrupt is implemented for light sleep wakeup trigger.

DexiangYang, 2023-08-21:
> Hi @DexiangYang,
> 
> There is no IRQ for LP_IO peripheral. May I ask what you are trying to achieve? Why can't you use the GPIO interrupt for the IO0? The LP_IO interrupt is implemented for light sleep wakeup trigger.

我想捕获一下边沿的信息，并不需要进入中断，只要能够置位中断标志位其实就行

DexiangYang, 2023-08-21:
> Hi @DexiangYang,
> 
> There is no IRQ for LP_IO peripheral. May I ask what you are trying to achieve? Why can't you use the GPIO interrupt for the IO0? The LP_IO interrupt is implemented for light sleep wakeup trigger.

我这个是使用ESP32 C6的LP Core查询终端信息，所以没用GPIO0的中断

songruo, 2023-08-23:
@DexiangYang 除了要配置 `LP_IO_LP_GPIO0_INT_TYPE`，试试将 `LP_IO_LP_GPIO0_WAKEUP_ENABLE`位 也置1，这样应该就能在 `LP_IO_LP_GPIO_STATUS_INTERRUPT` 中反应出来了。

DexiangYang, 2023-08-30:
> 

你好，我尝试了您提供的两个寄存器都配置了，但是还是没有效果
`    // REG_SET_FIELD(LP_IO_PIN0_REG, LP_IO_LP_GPIO0_INT_TYPE, 3);
    lp_core_printf("LP_IO_LP_GPIO0_INT_TYPE : %02x\n", REG_GET_FIELD(LP_IO_PIN0_REG, LP_IO_LP_GPIO0_INT_TYPE));

    // REG_SET_FIELD(LP_IO_PIN0_REG, LP_IO_LP_GPIO0_WAKEUP_ENABLE, 1);
    lp_core_printf("LP_IO_LP_GPIO0_WAKEUP_ENABLE : %02x\n", REG_GET_FIELD(LP_IO_PIN0_REG, LP_IO_LP_GPIO0_WAKEUP_ENABLE));

    LP_IO.pin[0].wakeup_enable = 1;
    LP_IO.pin[0].int_type = 3;

    // REG_SET_FIELD(LP_IO_PIN0_REG, LP_IO_LP_GPIO0_INT_TYPE, 3);
    lp_core_printf("LP_IO_LP_GPIO0_INT_TYPE : %02x\n", REG_GET_FIELD(LP_IO_PIN0_REG, LP_IO_LP_GPIO0_INT_TYPE));

    // REG_SET_FIELD(LP_IO_PIN0_REG, LP_IO_LP_GPIO0_WAKEUP_ENABLE, 1);
    lp_core_printf("LP_IO_LP_GPIO0_WAKEUP_ENABLE : %02x\n", REG_GET_FIELD(LP_IO_PIN0_REG, LP_IO_LP_GPIO0_WAKEUP_ENABLE));
    ulp_lp_core_delay_us(1000 * 1000);`
这是我的配置代码

DexiangYang, 2023-08-30:
我查看ulp_lp_core_run函数中，LP IO Wakeup功能是被禁止的，是不是目前芯片的LP核不支持GPIO中断

songruo, 2023-08-30:
@DexiangYang 首先 LP GPIO 不支持沿中断，只可以用电平中断。需要的全部配置如下：
1. 将 IO0 切到 LP_GPIO 功能 （置位 LP_AON_GPIO_MUX_REG BIT0，并 LP_IO.gpio[0].mcu_sel = 0）
2. 将 LP_IO0 输入使能 （ LP_IO.gpio[0].fun_ie = 1）
3. 按需配置上下拉使能 （LP_IO.gpio[0].fun_wpu，LP_IO.gpio[0].fun_wpd）
4. 使能唤醒并配触发模式 （LP_IO.pin[0].wakeup_enable = 1，LP_IO.pin[0].int_type = 4 / 5）

然后应该就可以在LP核上polling检查 LP_IO_LP_GPIO_STATUS_INTERRUPT了。

DexiangYang, 2023-08-30:
> 

`
extern lp_io_dev_t LP_IO;

int main (void) {

    LP_IO.pin[0].int_type = 4;
    ulp_lp_core_delay_us(1000);
    LP_IO.pin[0].wakeup_enable = 1;

    lp_core_printf("LP_IO_LP_GPIO0_INT_TYPE : %02x\n", REG_GET_FIELD(LP_IO_PIN0_REG, LP_IO_LP_GPIO0_INT_TYPE));
    lp_core_printf("LP_IO_LP_GPIO0_WAKEUP_ENABLE : %02x\n", REG_GET_FIELD(LP_IO_PIN0_REG, LP_IO_LP_GPIO0_WAKEUP_ENABLE));
    lp_core_printf("LP_AON_GPIO_MUX_SEL : %02x\n", REG_GET_FIELD(LP_AON_GPIO_MUX_REG, LP_AON_GPIO_MUX_SEL));
    lp_core_printf("LP_IO.gpio[0].mcu_sel : %d\n", LP_IO.gpio[0].mcu_sel);
    lp_core_printf("LP_IO.gpio[0].fun_wpu : %d\n", LP_IO.gpio[0].fun_wpu);
    lp_core_printf("LP_IO.gpio[0].fun_wpd : %d\n", LP_IO.gpio[0].fun_wpd);
    ulp_lp_core_delay_us(1000 * 1000);

    while(1) {
        lp_core_printf("%02x\n", LP_IO.status.status_interrupt);
        lp_core_printf("LP_IO_LP_GPIO_STATUS_INTERRUPT : %02x\n", REG_GET_FIELD(LP_IO_STATUS_REG, LP_IO_LP_GPIO_STATUS_INTERRUPT));
        lp_core_printf("%02x\n", LP_IO.status.status_interrupt);
        lp_core_printf("LP_IO_LP_GPIO_IN_DATA_NEXT : %02x\n", REG_GET_FIELD(LP_IO_IN_REG, LP_IO_LP_GPIO_IN_DATA_NEXT));
        lp_core_printf("%02x\n", LP_IO.status.status_interrupt);
        lp_core_printf("LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : %02x\n", REG_GET_FIELD(LP_IO_STATUS_INTERRUPT_REG, LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT));
        ulp_lp_core_delay_us(1000 * 1000);
    }

    return 0;
}
`

你好，这是我的完整代码，我测试了发现当我LP GPIO 0接地启动时，有中断标志被置位，但是当我断开GPIO 0和地的链接时启动，中断标志始终不会职位，但是我观察LP_IO_LP_GPIO_IN_DATA_NEXT和LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT都是有相应的变化的

DexiangYang, 2023-08-30:
[15:25:59.826]收←◆LP_IO_LP_GPIO0_INT_TYPE : 04
LP_IO_LP_GPIO0_WAKEUP_ENABLE : 01
LP_AON_GPIO_MUX_SEL : 3d
LP_IO.gpio[0].mcu_sel : 0
LP_IO.gpio[0].fun_wpu : 1
LP_IO.gpio[0].fun_wpd : 0

[15:26:00.904]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

[15:26:01.962]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

[15:26:03.020]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

[15:26:04.078]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

[15:26:05.136]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

[15:26:06.193]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

[15:26:07.251]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

[15:26:08.308]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 10
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 01

[15:26:09.366]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 10
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 01

[15:26:10.424]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 10
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 01

[15:26:11.482]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

[15:26:12.540]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

[15:26:13.598]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

[15:26:14.655]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

[15:26:15.713]收←◆00
LP_IO_LP_GPIO_STATUS_INTERRUPT : 00
00
LP_IO_LP_GPIO_IN_DATA_NEXT : 11
00
LP_IO_LP_GPIO_STATUS_INTERRUPT_NEXT : 00

DexiangYang, 2023-08-30:
上述是LP UART 输出的打印

DexiangYang, 2023-11-02:
@songruo 你好，方便试一下吗

songruo, 2024-05-30:
Sorry for the late reply. It is confirmed that there is a hardware bug on C6. `LP_IO.data.clk_en` bit has to be set, so that `LP_IO.status.status_interrupt `can get updated, and then the LP_IO interrupt can get triggered on the LP core.

closed 2025-01-14 by songruo