espressif/esp-idf #12100: High Memory Usage in ESP IDF Station Wi-Fi  code

Noushadalik, 2023-08-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

External 3.3V

### What is the expected behavior?

For our project, we are using the ESP IDF {Station Wi-Fi example[ code.](https://github.com/espressif/esp-idf/blob/release/v4.4/examples/wifi/getting_started/station/main/station_example_main.c) We expect a reasonable memory consumption that is relatively constant throughout the execution of the application. The available heap size should be within acceptable limits, ensuring that memory fragmentation is minimized

### What is the actual behavior?

We observe significant memory consumption when using the ESP IDF Wi-Fi instance in our project. The log given below describes the memory usage patterns at various stages of application execution

2023-08-18T11:13:47Z--->W (5542) Before wifi connect: free_heap_size = 126284, largest_free_block = 110576
...
2023-08-18T11:13:47Z--->E (7572) After WiFI connect : free_heap_size = 88788, largest_free_block = 86000
..

### Steps to reproduce.

1) Compile and install the ESP IDF Station WiFi example code.
2) View the logs on the computer while the instance is running.
3) Note the memory-related numbers in the logs.
4) Note any changes while starting the wifi module.
5) See what happens when the instance tries to connect to a network.
6) Monitor the memory details after a successful connection.

### Debug Logs.

```plain
`2023-08-18T11:13:47Z--->W (5542) After netif: **free_heap_size = 126284, largest_free_block = 110576**
Task: Pre-Scheduler allocs -> CAP_8BIT: 4192 CAP_32BIT: 0
Task: ipc0 -> CAP_8BIT: 17416 CAP_32BIT: 0
Task: main -> CAP_8BIT: 41184 CAP_32BIT: 0
Task: ipc1 -> CAP_8BIT: 28 CAP_32BIT: 0

2023-08-18T11:13:47Z--->E (5572) WiFi test 1: free_heap_size = 120940, largest_free_block = 110576
2023-08-18T11:13:47Z--->I (5602) wifi:2023-08-18T11:13:47Z--->wifi driver task: 3ffe2930, prio:23, stack:6656, core=02023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (5642) wifi:2023-08-18T11:13:47Z--->wifi firmware version: 63017e02023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (5682) wifi:2023-08-18T11:13:47Z--->wifi certification version: v7.02023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (5722) wifi:2023-08-18T11:13:47Z--->config NVS flash: enabled2023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (5752) wifi:2023-08-18T11:13:47Z--->config nano formating: disabled2023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (5782) wifi:2023-08-18T11:13:47Z--->Init data frame dynamic rx buffer num: 322023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (5822) wifi:2023-08-18T11:13:47Z--->Init management frame dynamic rx buffer num: 322023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (5852) wifi:2023-08-18T11:13:47Z--->Init management short buffer num: 322023-08-18T11:13:47Z--->
[... 48 lines cut ...]
2023-08-18T11:13:47Z--->I (6142) wifi:2023-08-18T11:13:47Z--->mode : sta (10:97:bd:0a:aa:90)2023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (6172) wifi:2023-08-18T11:13:47Z--->enable tsf2023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (6212) WIFI: wifi_init_sta finished.
2023-08-18T11:13:47Z--->I (6222) wifi:2023-08-18T11:13:47Z--->new:<2,0>, old:<1,0>, ap:<255,255>, sta:<2,0>, prof:12023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (6262) wifi:2023-08-18T11:13:47Z--->state: init -> auth (b0)2023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (6302) wifi:2023-08-18T11:13:47Z--->state: auth -> assoc (0)2023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (6362) wifi:2023-08-18T11:13:47Z--->state: assoc -> run (10)2023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (6422) wifi:2023-08-18T11:13:47Z--->connected with SoLo, aid = 4, channel 2, BW20, bssid = 4e:c6:a6:42:cf:912023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (6452) wifi:2023-08-18T11:13:47Z--->security: WPA2-PSK, phy: bgn, rssi: -422023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (6492) wifi:2023-08-18T11:13:47Z--->pm start, type: 1
2023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (6532) wifi:2023-08-18T11:13:47Z--->AP's beacon interval = 102400 us, DTIM period = 22023-08-18T11:13:47Z--->
2023-08-18T11:13:47Z--->I (7542) esp_netif_handlers: sta ip: 192.168.188.168, mask: 255.255.255.0, gw: 192.168.188.250
2023-08-18T11:13:47Z--->I (7562) WIFI: got ip:192.168.188.168
2023-08-18T11:13:47Z--->E (7572) WiFi test 3: free_heap_size = 88788, largest_free_block = 86000
2023-08-18T11:13:47Z--->I (7582) WIFI: connected to--->SoLo
2023-08-18T11:13:47Z--->W (7632) After WiFi: free_heap_size = 88988, largest_free_block = 86000
Task: Pre-Scheduler allocs -> CAP_8BIT: 4192 CAP_32BIT: 0
Task: ipc0 -> CAP_8BIT: 17416 CAP_32BIT: 0
Task: main -> CAP_8BIT: 55892 CAP_32BIT: 0
Task: ipc1 -> CAP_8BIT: 28 CAP_32BIT: 0
Task: tiT -> CAP_8BIT: 440 CAP_32BIT: 0
Task: wifi -> CAP_8BIT: 21992 CAP_32BIT: 0
Task: sys_evt -> CAP_8BIT: 156 CAP_32BIT: 0`
```

Xiehanxin, 2023-08-21:
hi @Noushadalik ,maybe you can refer to this to modify your parameters to reduce the memory occupy https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#how-to-configure-parameters

Sherry616, 2024-03-04:
Hi @Noushadalik, do you still have this problem? Could you please share the latest update? Thanks.

Sherry616, 2024-04-09:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2024-04-09 by Sherry616