espressif/esp-idf #12104: LoadProhibited when initializing BLE GATT

ftab, 2023-08-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-2305-g968ce2efc2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-LyraT v4.3

### Power Supply used.

USB

### What is the expected behavior?

Successfully boot

### What is the actual behavior?

Crashes during GATT initialization

### Steps to reproduce.

1. My fork of esp-adf (includes the latest esp-idf as well as a patch required by esp-adf): `git clone --recursive https://github.com/radiosound-com/esp-adf.git -b btdm-crash-testing`
2. Some environment variables
    ```sh
    export ESPBAUD=2000000
    export ADF_PATH=~/esp/esp-adf # or wherever you cloned it
    ```
3. `~/esp/esp-adf/esp-idf/install.sh` if necessary
4. `. ~/esp/esp-adf/esp-idf/export.sh`
5. In esp-adf/examples/advanced_examples/wifi_bt_ble_coex: [sdkconfig](https://github.com/espressif/esp-idf/files/12383624/sdkconfig.txt)
6. idf.py menuconfig (optional, change to your wifi credentials, but the app currently doesn't get that far)
7. idf.py build flash monitor

### Debug Logs.

```plain
--- idf_monitor on /dev/ttyUSB1 2000000 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (29) boot: ESP-IDF v5.2-dev-2305-g968ce2efc2-dirty 2nd stage bootloader
I (29) boot: Multicore bootloader
I (29) boot: chip revision: v1.0
I (29) qio_mode: Enabling default flash chip QIO
I (30) boot.esp32: SPI Speed      : 80MHz
I (30) boot.esp32: SPI Mode       : QIO
I (30) boot.esp32: SPI Flash Size : 4MB
I (30) boot: Enabling RNG early entropy source...
I (31) boot: Partition Table:
I (31) boot: ## Label            Usage          Type ST Offset   Length
I (31) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (32) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (32) boot:  2 factory          factory app      00 00 00010000 00380000
[... 571 lines cut ...]
D (1476) esp_core_dump_elf: Add segment size=512, start_off=0x568c
D (1477) esp_core_dump_elf: Found 0 bad task out of 14
D (1477) esp_core_dump_elf: ================ Processing coredump info ================
D (1477) esp_core_dump_elf: Application SHA256='d1e7e0fe94f34680', length=16.
D (1478) esp_core_dump_elf: Add note size=104, start_off=0x588c
D (1478) esp_core_dump_elf: Add note size=172, start_off=0x588c
D (1478) esp_core_dump_elf: Add NOTE segment, size=276, start_off=0x588c
D (1478) esp_core_dump_elf: =========== Data written size = 22968 bytes ==========
D (1479) esp_core_dump_checksum: Total length of hashed data: 22976
I (1479) esp_core_dump_flash: Write end offset 0x59c4, check sum length 4
I (1480) esp_core_dump_common: Core dump used 1068 bytes on stack. 3112 bytes left free.
I (1480) esp_core_dump_common: Restoring stack @ 0x3ffddce0
I (1481) esp_core_dump_flash: Core dump has been saved to flash.
CPU halted.
```

### More Information.

In the process of attempting to gather data for #10555 I have encountered a few other issues and am trying to update to the latest esp-adf and esp-idf to rule out any that are already fixed.

Here is a coredump and the corresponding .elf file [coredump.zip](https://github.com/espressif/esp-idf/files/12383702/coredump.zip)

I experienced this with release/v5.1 (with commit 0634bb9 manually applied) as well as on master (both pulled today and then the patch applied that esp-adf requires)

I did not experience this on release/v5.0

zhp0406, 2023-08-23:
@ftab 
thanks for report the bug,could you please test it with the following patch? 
[0001-fix-bt-Fix-bug-while-calculating-block-cipher-using-.patch](https://github.com/espressif/esp-idf/files/12417866/0001-fix-bt-Fix-bug-while-calculating-block-cipher-using-.patch)

ftab, 2023-08-23:
@zhp0406 I have applied the patch, but the issue still occurs. Same backtrace as the above.

zhp0406, 2023-08-24:
@ftab Hi,
I am trying to solve the problem,could you please test it again with the following updated patch?
[0001-fixed-gatt-database-hash-calculate.patch](https://github.com/espressif/esp-idf/files/12427562/0001-fixed-gatt-database-hash-calculate.patch)

thanks.

ftab, 2023-08-24:
That seems to do the trick. The example finishes booting up now. Thanks!

closed 2023-08-25 by Alvin1Zhang

ftab, 2023-08-30:
```c
/******************************************************************************
 *
 *  Copyright (C) 2008-2012 Broadcom Corporation
 *
 *  Licensed under the Apache License, Version 2.0 (the "License");
 *  you may not use this file except in compliance with the License.
 *  You may obtain a copy of the License at:
 *
 *  http://www.apache.org/licenses/LICENSE-2.0
 *
 *  Unless required by applicable law or agreed to in writing, software
 *  distributed under the License is distributed on an "AS IS" BASIS,
 *  WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 *  See the License for the specific language governing permissions and
 *  limitations under the License.
 *
 ******************************************************************************/

/******************************************************************************
 *
 *  This file contains the implementation of the AES128 CMAC algorithm.
 *
 ******************************************************************************/

#include "common/bt_target.h"
#include "osi/allocator.h"

#if SMP_INCLUDED == TRUE
//    #include <stdio.h>
#include <string.h>

#include "stack/btm_ble_api.h"
#include "smp_int.h"
#include "stack/hcimsgs.h"

typedef struct {
    UINT8               *text;
    UINT16              len;
    UINT16              round;
} tCMAC_CB;

tCMAC_CB    cmac_cb;

/* Rb for AES-128 as block cipher, LSB as [0] */
const BT_OCTET16 const_Rb = {
    0x87, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};

void print128(BT_OCTET16 x, const UINT8 *key_name)
{
#if SMP_DEBUG == TRUE && SMP_DEBUG_VERBOSE == TRUE
    UINT8  *p = (UINT8 *)x;
    UINT8  i;

    SMP_TRACE_WARNING("%s(MSB ~ LSB) = ", key_name);

    for (i = 0; i < 4; i ++) {
        SMP_TRACE_WARNING("%02x %02x %02x %02x",
                          p[BT_OCTET16_LEN - i * 4 - 1], p[BT_OCTET16_LEN - i * 4 - 2],
                          p[BT_OCTET16_LEN - i * 4 - 3], p[BT_OCTET16_LEN - i * 4 - 4]);
    }
#endif
}

/*******************************************************************************
**
** Function         padding
**
[... 278 lines cut ...]
    };
    UINT8 i = 0, tmp;
    UINT16 len;

    len = 64;

    for (i = 0; i < len / 2; i ++) {
        tmp = M[i];
        M[i] = M[len - 1 - i];
        M[len - 1 - i] = tmp;
    }

    memset(&cmac_cb, 0, sizeof(tCMAC_CB));

    SMP_TRACE_WARNING("\n Example 1: len = %d\n", len);

    aes_cipher_msg_auth_code(key, M, len, 128, test_cmac_cback, 0);

}
#endif
#endif
```
This is after applying your patches

Started from esp-idf 968ce2efc2, plus [this patch required by esp-adf](https://github.com/espressif/esp-adf/blob/master/idf_patches/idf_v5.1_freertos.patch)