espressif/esp-idf #12106: Wifi init causing usb host vcp client to suddenly disconnect, IDF5.1

colinb2048, 2023-08-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Wemos Lolin S2

### Power Supply used.

External 5V

### What is the expected behavior?

Init wifi and usb, the order of which shouldn't matter, receive UDP packets and write the data to usb vcp client, and vice versa

### What is the actual behavior?

USB vcp client suddenly disconnects after wifi init. Frequency of disconnect is random when no data flowing, but when receiving udp packet and sending via usb, almost always suddenly disconnects. 

### Steps to reproduce.

I've merged the wifi init sta and usb host vcp examples to create bidirectional comms between usb and wifi devices.
With no wifi, usb works as expected.
If usb installed first, wifi fails to connect to access point.
With wifi active, trying to send to usb vcp clients almost always suddenly disconnects

### More Information.

Source:

`#include <stdio.h>
#include <string.h>
#include <iostream>

#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"

#include "usb/cdc_acm_host.h"
#include "usb/vcp_ch34x.hpp"
#include "usb/vcp_cp210x.hpp"
#include "usb/vcp_ftdi.hpp"
#include "usb/vcp.hpp"
#include "usb/usb_host.h"

#include "driver/uart.h"
#include "driver/gpio.h"

#include "freertos/event_groups.h"
#include "esp_system.h"
[... 441 lines cut ...]

    ESP_LOGI(TAG, "Installing CDC-ACM driver");
    ESP_ERROR_CHECK(cdc_acm_host_install(NULL));

    xTaskCreate(usb_connect_task, "usb_con", 4096, NULL, 10, NULL);

    /*
    Now the USB-to-UART converter is configured and receiving data.
    You can use standard CDC-ACM API to interact with it. E.g.

    ESP_ERROR_CHECK(vcp->set_control_line_state(false, true));
    ESP_ERROR_CHECK(vcp->tx_blocking((uint8_t *)"Test string", 12));
    */

    // Send some dummy data
    // ESP_LOGI(TAG, "Sending data through CdcAcmDevice");
    // uint8_t data[] = "test_string";
    // ESP_ERROR_CHECK(vcp->tx_blocking(data, sizeof(data)));
    // ESP_ERROR_CHECK(vcp->set_control_line_state(true, true));

    // We are done. Wait for device disconnection and start over
    // ESP_LOGI(TAG, "Done. You can reconnect the VCP device to run again.");
    // xSemaphoreTake(device_disconnected_sem, portMAX_DELAY);
    // }
}`

DarmorGamz, 2023-08-22:
Can you provide your SDKConfig. Do you have Enable USB when phy init?

colinb2048, 2023-08-22:
Thanks for the suggestion, it wasn't, but after enabling 'Enable USB when phy init' i get a compilation error: undefined reference to `phy_bbpll_en_usb'

I've found references to the same compilation error but doesn't appear those threads reached a solution?

roma-jam, 2023-09-26:
Hey @colinb2048, 

Seems like there could be two problems:  either power consumption spikes during WiFi data transfer, which leads to the usb host device disconnection or some interlacing in driver code, which is not so trivial.
To proceed, I would like to reproduce the problem and verify all theories. 

To be able to do that, may I ask to share the KConfig options as well? (regarding the partitions and other configuration for WiFi). 
Probably, it will be easy to share the whole project (main, sdkconfig and partitions.csv).

Meanwhile, I'll try to find the problem.
Thanks.

roma-jam, 2023-10-09:
Hey @colinb2048, 

I have tried to reproduce the problem on ESP32-S2 devkit with example code you had provided, but no luck. 
I tried to send UDP packets from HOST with several length, all seems to correctly received from socket in `udp_Receive_Task()` in `udp_rx_buffer` and then transmitted via `vcp->tx_blocking()` as well. 

To proceed, I need to clarify some details. 
1. How are you powering the USB device? How the USB-OTG wire is connected to the Wemos Lolin S2 board?
2. May I ask you about your UDP packet length? I saw in code that you have a 7-byte length threshold. Probably, it is a some-command length. Do you use variable length packets? If yes, which size? 
3. I checked out that Wemos Lolin S2 has a me6211c33 LDO on board, which has a 500 mA OC. Mine dev board has a sgm2212-33, which has 800 mA OC. Probably, it is important in your case, but without answer on first question it is hard to say. With a USB-OTG we need to power the attachable device as well. 

Thanks.

mzhboy, 2023-10-15:
Hi , I'm using micropython/circuitpython , and encountered the same issue. 
In fact， it's one of two esp32-s2-lolin boards encountered this issue, while the other one works fine.
I also modified en pin circuit on both boards, replace pull-up reisstor(r3) with 1k and C14 with 4.7uF.
The input capacitor is too small, so another 220uF capacitor added to VBUS.

roma-jam, 2023-10-25:
Hi @mzhboy, 

Thanks for the reporting that!
I have tried to reproduce the problem, but with another board and didn't succeeded.

I have a couple of questions, answering to them could help us to move forward. 

1. From which source do you provide the VBUS to the connected device? 
2. How do you supply the Wemos Lolin S2 board? With some USB port or +5V external power? 
3. Is there any option to oscilloscope the VBUS level during encountering the issue?

Thanks!

mzhboy, 2023-10-25:
@roma-jam 
Powering from usb hub, the hub has external PSU, output is 5.2V 3A.
Others esp32 boards with onboard usbserial bridge have no issue.

[image]

roma-jam, 2023-11-07:
Hi @mzhboy, 

Thanks for sharing additional information! 
But I am not sure that your issue is connected to the topic-starter issue. 
Anyway, let me clear some things to understand it better. 

1. In python-example you had provided, you are using the vcp which is the usb-serial bridge on the boards. Do you use the esp32 as a USB Host to be able to plug external USB Device (with using external USB jack port)? 
2. the original issue was about the device, that has a initialized WiFi and uses the esp32s2 option to be able a USB Host for attaching external USB Device.  It is little bit unclear, but are you using the `usb_host` driver from esp-idf [components](components/usb)? 

In case, that it seems that you are not using the `usb_host` driver (but please, correct me if I am wrong), seems that you issue '_may be_' related but definitely not because you are using the `usb_host` driver. 

Which means, that the problem is not related to `usb_host` driver itself...

Otherwise, for us to be able to proceed, could you please provide the simplest possible example with `usb_host` driver usage to reproduce the failure? 
Thanks.

roma-jam, 2023-11-21:
Hi everyone, 

Considering the information, that micropython code also has the same problem but doesn't use the `usb_host` library, it seems that the problem could be deeper, than the layer `usb_host` library works at. 

There is a chance, that this issue has the same nature as [this one](https://github.com/espressif/esp-idf/issues/12185). 

Regarding the problem with micropython, I could suggest to create a new ticket if you still have problems with it.

In case there will be any new information, regarding the `usb_host` library and working ESP32 as a USB Host, feel free to reopen this ticket. 
Thanks.

closed 2023-11-21 by roma-jam

tore-espressif, 2023-11-27:
Hello everyone, we revisited the topic again, but we could not reproduce it. So, I'd like to ask for more information:

1. Which chip version do you use? (command `esptool chip_id`)
2. Could you please share dump of your eFuses? (command `espefuse.py -p PORT dump`)
3. Could you please try updating to IDF v5.1.2? This https://github.com/espressif/esp-idf/commit/c98aa927f689a23533b4851e1b6acce218bfe4dd commit might fix it for you